nRF24/RF24 #810: rp2040 issue with radio write

matt-walker0, 2021-11-29:
Hi all,

NRF24 is setup, debug print shows registers set as planned. PWR level is set to low. All good so far... however, 
The following code freezes the Pico. I've written a fair bit of C code, so this function is called from C to interact with the C++ rf24 lib.

// Assumes, stop listen prior to run. Return true if ack_rec
bool nrf24SendData(uint8_t buffer[5]) {
    uint8_t temp_buff[5] = {1,2,3,4,5};
    // LED ON statement here works!
    radio.write(temp_buff,5);
    // LED ON statement here doesnt =/
    return(true);
}

Straight after this function an LED is set to turn on, which it never reaches. 

This could well be bad hardware, (I'm learning not to trust the NRF24 module too much :) ) I thought at first it was due to voltage drop without the capacitor, but now with 47uF still same result. Everything is soldered directly onto a PCB.

Any ideas welcome. Thanks!

2bndy5, 2021-11-29:
I would suggest a few things (these are not specific to the rp2040)
```c++
// Assumes, stop listen prior to run. Return true if ack_rec
bool nrf24SendData(uint8_t buffer[], uint8_t len = 5) {
     uint8_t temp_buff[5] = {1,2,3,4,5};
     // LED ON statement here works!
     bool result = radio.write(&temp_buff, 5);
     // LED ON statement here doesnt =/
     return result;
}
```
Notice
1. passing in the buffer as a pointer
2. additionally passing in the buffer len as a separate arg (this is basic C++ convention). This might need to be a `const` depending on how you intend to use it.
3. passing `temp_buf` as a reference (not a pointer) to `RF24::write()` <-- this may be why the pico get frozen.
4. use actual boolean value reported from `RF24::write()` as the returned data. This will help in the event that a payload actually fails to transmit.

matt-walker0, 2021-11-30:
Many thanks for the advice. 
I'd hardset to buffer length of 5, as all msg will be 5 bytes, but guess for the sake of 30s worth adding length.

I've tried to run code again (copying above suggestion) this time with the TX command repeating every 0.5s, unfortunately doesn't get past (or seem to complete) the first TX 

'
^^ DEBUG MSG ABOVE ^^
PA Power	= PA_LOW
ARC		= 0

TXing: 1, 2, 3, 4, 5
*** PANIC ***
`
Not entirely sure where the ***PANIC*** comes from, doesn't seem to be in this lib so assuming Linux thing when USB device malfunctions.

2bndy5, 2021-11-30:
The `PANIC` may be from the Pico itself. Usually, output like that is caused by a segmentation fault or bad memory access/management. Check your use of pointers and c-strings.

matt-walker0, 2021-12-01:
Quick update, I've tracked the bug down to use of IRQ handler, removing IRQ handling allows for sucessful TX. This function was working well for RX data but I suspect during TX it is triggered and may interact (in a bad way!) with TX process. Likely my code error, but will continue debug.

2bndy5, 2021-12-01:
sounds like you should use [`maskIRQ()`](https://rf24.readthedocs.io/en/latest/classRF24.html#_CPPv4N4RF247maskIRQEbbb) to disable the `tx_ok` & `tx_fail` flags triggering the IRQ pin. Are you using ACK payloads (because the `rx_ready` flag will also trigger the IRQ pin during `write()`)?

matt-walker0, 2021-12-01:
Ah I was looking at that exact command in the PicoIRQ example. 
I am using ACK from the receiver, although during this testing no other devices are active, hence should never get ACK-REC.

Using some advanced debugging techniques, moving the LED on statement around :) , I found the following:

`
// Returns enum depending upon what occured
uint8_t nrf24InterruptHandle() {
    bool tx_ok, tx_fail, rx_ready;      
    // LED works here
    radio.whatHappened(tx_ok, tx_fail, rx_ready); // get values for IRQ masks 
   // LED doesnt work here

    if(rx_ready == true) { // new rx data
        return(NRF24_RX_READY);
    }
    else if(tx_ok == true) {
        return(NRF24_TX_OK);
    }
    else if(tx_fail == true) {
        return(NRF24_TX_FAIL);       
    }
    return(NRF24_NONE);
}
`

So with that in mind I masked the tx pins in the radio setup function.
`// Allows for function to be inserted for call on IRQ falling
void nrf24SetupIRQ(uint8_t irq_pin, void (*irq_handler)(uint gpio, uint32_t event) ) {
    radio.maskIRQ(0,0,1);
    gpio_set_irq_enabled_with_callback(irq_pin, GPIO_IRQ_EDGE_FALL, true, irq_handler);
}`
Although still getting **PANIC** . Close now, I keep testing....
Out of interest, you must disable tx_ok and tx_fail to prevent TX from failing? I know that you can pick up on whether a TX was successful from write function return, so these are not needed anyway?

closed 2021-12-01 by matt-walker0

2bndy5, 2021-12-01:
> you must disable tx_ok and tx_fail to prevent TX from failing? 

I only mentioned it because the \*\*PANIC\*\* problem might be related to the IRQ pin triggering a call to you `irq_handler()` while still executing `write()`. This is likely causing a race-condition on "which function gets to use the SPI first?" If that doesn't answer your question, I'm afraid I don't understand the question.

BTW, the `rx_ready` flag **will** be asserted immediately after the `tx_ok` flag is asserted (if an ACK packet contained a payload). This will also lead to a race condition while executing `write()`.

matt-walker0, 2021-12-01:
Yep I think likely was race condition with IRQ triggering while still "writing" message. 
Many thanks and have a good day.