nRF24/RF24 #812: [Request] new `isFifo()` function

2bndy5, 2021-12-02:
Currently, the library does not provide users with a way to examine the FIFO buffers statuses, with the exception of `rxFifoFull()`. In developing the CirPy lib, I have found the ability to examine the FIFO statuses extremely helpful when debugging a problem.

## Proposal
Using 2 overloaded functions, we can expose info from the FIFO_STATUS register.
```cpp
    /**
     * @param about_tx `true` focuses on the TX FIFO, `false` focuses on the RX FIFO
     * @return 
     * - `0` if the specified FIFO is neither full nor empty.
     * - `1` if the specified FIFO is empty.
     * - `2` if the specified FIFO is full.
     */ 
    uint8_t is_fifo(const bool about_tx)
    {
        return static_cast<uint8_t>((read_register(FIFO_STATUS) >> (4 * about_tx)) & 3);
    }

    /** 
     * @param about_tx `true` focuses on the TX FIFO, `false` focuses on the RX FIFO
     * @param check_empty
     * - `true` checks if the specified FIFO is empty
     * - `false` checks is the specified FIFO is full.
     * @return A boolean answer to the question "is the [TX/RX] FIFO [empty/full]?"
     */
    bool is_fifo(const bool about_tx, const bool check_empty)
    {
        return static_cast<bool>(is_fifo(about_tx) & _BV(!check_empty));
    }
```

### Additional context
It is not my intention to make `rxFifoFull()` internally call `isFifo(0, 0)` as that would add to the code-size. Initially, I had planned to add these functions to the newer python wrapper in pyRF24, but I think it could be useful to those trying to debug the FIFO buffers' state in C++.

TMRh20, 2021-12-03:
This might be good but I think we should organize the class docs better somehow because there are so many functions. I’ll have to think about how to categorize them…

2bndy5, 2021-12-03:
Yeah, I started organizing them by radio's feature in the CirPy lib's docs

2bndy5, 2021-12-03:
I could do some experimenting with the docs on my "improve-sphinx-docs" branch...

TMRh20, 2021-12-03:
Sure, I haven’t given it a good look yet but if u want to try some things out, that sounds great.

2bndy5, 2022-07-09:
I want to fit this in before the next release. I'll also be adding the model variant info to `printPrettyDetails()` (per https://github.com/nRF24/RF24/commit/aeb8b427304af7c5d45e03330a013ae98f895040#commitcomment-78046335 obervation)

PR #849 (2022-07-09): Add isFifo() methods
resolves #812 

I also updated the return value of `writeAckPayload()` (from void to bool) in the python wrapper. I used this function to fill the TX FIFO and then used `isFifo()` to verify the returned info.

The added docs are accurate about `isFifo()`'s returned data.
closed 2022-07-09 by TMRh20