nRF24/RF24 #813: undefined symbol:_ZTV3SPI with RPI2 and Bullseye

ErikApption, 2021-12-04:
**Describe the bug**

I cannot get the python wrapper to work with my raspberry pi 2 and bullseye with spidev driver.

The package seems to work with RPi driver but only with root user. When I build it with spidev driver I get the following error:

```
Traceback (most recent call last):
  File "/home/pi/nrf24stuff/./nrf24_receiver.py", line 11, in <module>
    from RF24 import RF24, RF24_PA_LOW
ImportError: /usr/local/lib/python3.9/dist-packages/RF24-1.4.2-py3.9-linux-armv7l.egg/RF24.cpython-39-arm-linux-gnueabihf.so: undefined symbol: _ZTV3SPI
```

ErikApption, 2021-12-04:
well have no clear explanation but a reboot did fix the issue

closed 2021-12-04 by ErikApption

2bndy5, 2021-12-04:
Classic! Turn it off then turn it back on again. 🤪 I'm glad the solution was that simple because I wasn't sure where to start looking for the problem here.