nRF24/RF24 #822: [Question] Communication question between RF-Nano and NodeMCU esp32 + RF24L01

pruge, 2022-01-18:
With two RF-NANOs, the following code works well.
However, the wrong value is transmitted between RF-NANO and NodeMCU esp32 + RF24L01.
I don't know what the problem was.

send
```
0 15 25 3000
```

receive
```
0 15 25 11
0 15 25 11
```

## sender
```cpp
#include <RF24.h>
#include <SPI.h>
#include <nRF24L01.h>
struct Data {
 public:
  Data() {}
  Data(uint8_t _no, uint8_t _temp, uint8_t _humi, uint16_t _co2) {
    no = _no;
    temp = _temp;
    humi = _humi;
    co2 = _co2;
  }
  uint8_t no;  // sensor no
  uint8_t temp;
  uint8_t humi;
  uint16_t co2;
};

RF24 radio(9, 10);  // CE, CSN
const byte address[6] = "00001";
void setup() {
  radio.begin();
  radio.openWritingPipe(address);
  radio.setPALevel(RF24_PA_MIN);
  radio.stopListening();
}
void loop() {
  // const char text[] = "Hello World";
  const Data data(0, 15, 25, 3000);
  radio.write(&data, sizeof(data));
  delay(1000);
}
```

## Receiver
```cpp
#include <RF24.h>
#include <SPI.h>
#include <nRF24L01.h>
struct Data {
 public:
  Data() {}
  Data(uint8_t _no, uint8_t _temp, uint8_t _humi, uint16_t _co2) {
    no = _no;
    temp = _temp;
    humi = _humi;
    co2 = _co2;
  }
  uint8_t no;  // sensor no
  uint8_t temp;
  uint8_t humi;
  uint16_t co2;
};
RF24 radio(9, 10);  // CE, CSN  // RF-NANO
RF24 radio(21, 22);  // CE, CSN  // NodeMCU esp32 + RF24L01
const byte address[6] = "00001";
void setup() {
  Serial.begin(115200);
  radio.begin();
  radio.openReadingPipe(0, address);
  radio.setPALevel(RF24_PA_MIN);
  radio.startListening();
}
void loop() {
  if (radio.available()) {
    Data data;
    radio.read(&data, sizeof(data));
    Serial.print(data.no);
    Serial.print(" ");
    Serial.print(data.temp);
    Serial.print(" ");
    Serial.print(data.humi);
    Serial.print(" ");
    Serial.println(data.co2);
  }
}
```

pruge, 2022-01-18:
@Avamander  Thanks for replying.
When I print the hex, more 00 is coming in from the receiving side.

send
```cpp
  const Data data(0, 15, 25, 3000);
  printHex((unsigned char*)&data, sizeof(data));

  00 0F 19 B8 0B  
```

receive
```cpp
  Data data;
  radio.read(&data, sizeof(data));
  printHex((unsigned char*)&data, sizeof(data));

  00 0F 19 B8 0B 00   
```

```cpp
char* hexChar(char* destBuf, unsigned char srcData) {
  destBuf[0] = hexmap[(srcData & 0xF0) >> 4];
  destBuf[1] = hexmap[srcData & 0x0F];

  return destBuf;
}
void printHex(unsigned char* data,
              signed char len,
              bool reverse = false) {
  char tempByte[3] = {0, 0, 0};

  if (reverse) {
    for (signed char i = len - 1; i > -1; i--) {
      Serial.print((char*)hexChar(tempByte, data[i]));
      Serial.print(F(" "));
    }
  } else {
    for (signed char i = 0; i < len; i++) {
      Serial.print((char*)hexChar(tempByte, data[i]));
      Serial.print(F(" "));
    }
  }
  Serial.println();
}
```

pruge, 2022-01-18:
[(stack overflow)Why isn't sizeof for a struct equal to the sum of sizeof of each member?](https://stackoverflow.com/questions/119123/why-isnt-sizeof-for-a-struct-equal-to-the-sum-of-sizeof-of-each-member)

> This is because of padding added to satisfy alignment constraints. Data structure alignment impacts both performance and correctness of programs:

```cpp
struct Data {
  uint8_t no;       // 1byte
  uint8_t temp;     // 1byte
  uint8_t humi;     // 1byte
  uint16_t co2;     // 2byte
};

struct Data {
  uint8_t no;       // 1byte
  uint8_t temp;     // 1byte
  uint8_t humi;     // 1byte
  uint8_t dummy;    // 1byte, Exactly should be added here.
  uint16_t co2;     // 2byte
};
```

Added dummy to match padding.
It works fine. Thank you @Avamander 

send 
```cpp
0 15 25 3000
```

receive
```cpp
0 15 25 3000
```

Avamander, 2022-01-18:
Good to know you got it solved.

closed 2022-01-18 by Avamander