nRF24/RF24 #823: [Question] Using 3-pin attiny85 setup as a receiver?

Strat00s, 2022-01-19:
Hey everyone. I am currently working on a project, where I am using attiny85 as a temperature probe, which uses the nrf24 to send data to a main station. However, the main station also needs to send some information back to the attiny. And here is the problem.
Using all of the 5 pins to control the nrf, I can get everything to work. I can send and receive data no problem. But using the 3-pin method, I am not even getting ack payloads while sending data.

So, is it even possible to use the 3-pin connection to receive data?

2bndy5, 2022-01-19:
This is just a guess, but you can try lowering the SPI speed.
```c
radio = RF24(3, 3, 4000000); // default is 10MHz
```
The 3 pin solution may have been developed back when the default SPI speed was 4MHz, and the 3 pin solution is very sensitive to timing.

Strat00s, 2022-01-19:
> This is just a guess, but you can try lowering the SPI speed.
> 
> ```c
> #define CE_CSN_PIN 3
> radio = RF24(CE_CSN_PIN, CE_CSN_PIN, 4000000); // default is 10MHz
> ```
> 
> The 3 pin solution may have been developed back when the default SPI speed was 4MHz, and the 3 pin solution is very sensitive to timing.

Also just to be sure. The schematic in the .ino file shows that CSN is connected to SCK via a diode, to VCC via a 22k resistor and to GND via a 10nF cap. EN is directly connected to VCC. With this, the only pins used are pins 0-2, so what's up with the pin 3? I always replaced the pin 3 definition with pin 2, as that one is the closes to controlling something (same pin as SCK).

2bndy5, 2022-01-19:
> so what's up with the pin 3?

It was just a guess, and I didn't check the schematic just to show how to lower the SPI speed. Please continue using whatever pin(s) works for you.

Strat00s, 2022-01-19:
@2bndy5 I meant it in general, as it is also in the example (but it does not make any sense, as the pin is not used for anything)

Strat00s, 2022-01-19:
Update. I changed the speed to your suggested 4MHz and now it seems to work pretty much every time (sending with ack and receiving). Before that, I even had problems with transfers (it worked only sometimes).

closed 2022-01-19 by Strat00s