nRF24/RF24 #827: Hardware SPI, Multiple Devices with nRF24

De-Man, 2022-02-08:
Hi Team, thank you for your work on this library, I have been using it on an Arduino.

Apologies in advance if this is documented somewhere and I haven't seen it.

An issue I have come across with this library when using Arduino's Hardware SPI with an nRF24 device + another device. In my case a [Adafruit MicroSD card breakout board+](https://www.adafruit.com/product/254), I have also noted multiple people having the same issues and never really seeing an adequate resolve, such as:

- [https://forum.arduino.cc/t/nrf24l01-over-software-spi/524284](https://forum.arduino.cc/t/nrf24l01-over-software-spi/524284)
- [https://github.com/nRF24/RF24/issues/692](https://github.com/nRF24/RF24/issues/692)

There are many more examples.

The general resolve is to use SW SPI on the nRF24 and use the HW SPI for everything else. As I need the speed for both, I ideally do not want to use SW SPI.

In this case I am using an Arduino MEGA.

After a couple of hours spent troubleshooting this with an Oscope and various other means, I concluded the following:
- Conducted the same trouble shooting exercises as above links:
  - nRF24 worked with HW SPI, with no other slaves.
  - SD worked with HW SPI, with no other slaves.
  - nRF24 would fail as soon as SD was connected. SD would continue to work. 
- Was not a logic level issue:
  - Master In Slave Out:
    - SD Reader = 3.3v
    - nRF24 = 3.3v
  - Master Out Slave In:
    - Arduino = 5v
- Both Chip Selects / (Slave Selects) worked as expected with both slaves on the same SPI. Even with the nRF24 not working. This was verified with the scope when the nRF tried to MISO data, CS would go low. Same for SD.

This concluded that it was unlikely to be a hardware issue.

Therefore I went through SPI setting alignment routine:
- The Arduino defaults SPI speed to [4MHz](https://www.arduino.cc/en/Reference/SPISetClockDivider)
- This Library's `RF24_config.h` defaults to 10MHz `#define RF24_SPI_SPEED 10000000`

After I changed the RF24_config.h to 4MHz or less, SPI worked with both devices.

I didn't dig too far in to the R24 library and how it interfaces with the HW SPI library. But is there any way to fix this by defaulting to the SPI speed.

Thank you.

2bndy5, 2022-02-08:
> But is there any way to fix this by defaulting to the SPI speed.

Not sure what you mean here. It sounds like you did fix it by setting the RF24 lib's SPI speed to 4MHz. I'm assuming you did as instructed in the [common issues doc](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#payloads-receivedsent-are-inaccurate-or-printdetails-outputs-the-unexpected-values). If you don't want to use the constructor's parameter, then you should be able to pass `-DRF24_SPI_SPEED=4000000` to the compiler (an advanced technique that only works in PlatformIO - not Arduino IDE).

If you mean that you want to use 10MHz for nRF24 and something else for the SD card, then that is hardware dependent. Specifically, if the board's Arduino core's SPI library supports [`beginTransaction()` and `endTransaction()`](https://github.com/nRF24/RF24/blob/1e4c4a37f7ab23542247cedea638d7256aab65cd/RF24.cpp#L114-L142). If it turns out that the board's SPI library does not support using those functions, then the SPI speed used on the bus has to be a value that all slave devices can support.

> This concluded that it was unlikely to be a hardware issue.

Do you know if the SD card library you're using employs the `begin/endTransaction()`? If it doesn't then this may be messing up the RF24 lib when you init the SD card on the bus.

2bndy5, 2022-02-08:
Assuming you mean the Arduino MEGA2560, then that board's [SPI lib does indeed use `*Transction()`](https://github.com/arduino/ArduinoCore-avr/blob/2cebe625833afcdb76cc941ccf90e5f5fefc27a9/libraries/SPI/src/SPI.h#L19-L21)

I don't know what SD card library you're using, so I can't tell if the SD card lib uses those functions...

2bndy5, 2022-02-08:
Looking at the SD card lib that ships with the Arduino IDE, I think you can specify the SPI speed using an overloaded `begin(uint32_t clock, uint8_t csPin)`. It does seem to use `*Transaction()`, but there are so many options in how to use that library... I have see you code's `setup()`

2bndy5, 2022-02-08:
This is interesting:
```cpp
  boolean SDClass::begin(uint32_t clock, uint8_t csPin) {
    if (root.isOpen()) {
      root.close();
    }

    return card.init(SPI_HALF_SPEED, csPin) &&
           card.setSpiClock(clock) &&
           volume.init(card) &&
           root.openRoot(volume);
  }
```
So, basically, the lib
1. inits the SD card at 250000 Hz
2. sets the SPI speed to whatever you actually passed to `SD::begin()`
3. init a virtual obj called volume (related to read/write operation
4. finally does something with the init'd volume obj (I suspect it would be similar to `cd .` in PC terminals)

Have you tried init the SD card first, then init the nRF24?

De-Man, 2022-02-09:
Hi Brendan, seems you can't get enough of nRF24 related support, thankfully for me :)

Correct, using the Arduino SD library:

```
  boolean SDClass::begin(uint8_t csPin) {
    if (root.isOpen()) {
      root.close();
    }

    /*

      Performs the initialisation required by the sdfatlib library.

      Return true if initialization succeeds, false otherwise.

    */
    return card.init(SPI_HALF_SPEED, csPin) &&
           volume.init(card) &&
           root.openRoot(volume);
  }
```
or, overloaded:
```
  boolean SDClass::begin(uint32_t clock, uint8_t csPin) {
    if (root.isOpen()) {
      root.close();
    }

    return card.init(SPI_HALF_SPEED, csPin) &&
           card.setSpiClock(clock) &&
           volume.init(card) &&
           root.openRoot(volume);
  }
```
`card.` is an object of the Sd2Card class, there are a couple of overloaded inits; however, `SPI_HALF_SPEED` and equates to 4MHz:
```
uint8_t Sd2Card::setSckRate(uint8_t sckRateID) {
  if (sckRateID > 6) {
    error(SD_CARD_ERROR_SCK_RATE);
    return false;
  }
  #ifndef USE_SPI_LIB
  // see avr processor datasheet for SPI register bit definitions
  if ((sckRateID & 1) || sckRateID == 6) {
    SPSR &= ~(1 << SPI2X);
  } else {
    SPSR |= (1 << SPI2X);
  }
  SPCR &= ~((1 << SPR1) | (1 << SPR0));
  SPCR |= (sckRateID & 4 ? (1 << SPR1) : 0)
          | (sckRateID & 2 ? (1 << SPR0) : 0);
  #else // USE_SPI_LIB
  switch (sckRateID) {
    case 0:  settings = SPISettings(25000000, MSBFIRST, SPI_MODE0); break;
    case 1:  settings = SPISettings(4000000, MSBFIRST, SPI_MODE0); break;
    case 2:  settings = SPISettings(2000000, MSBFIRST, SPI_MODE0); break;
    case 3:  settings = SPISettings(1000000, MSBFIRST, SPI_MODE0); break;
    case 4:  settings = SPISettings(500000, MSBFIRST, SPI_MODE0); break;
    case 5:  settings = SPISettings(250000, MSBFIRST, SPI_MODE0); break;
    default: settings = SPISettings(125000, MSBFIRST, SPI_MODE0);
  }
  #endif // USE_SPI_LIB
  return true;
}
#ifdef USE_SPI_LIB
```
Nevertheless, from you points:

> Assuming you mean the Arduino MEGA2560

Correct.

> If you don't want to use the constructor's parameter, then you should be able to pass `DRF24_SPI_SPEED=4000000` to the compiler (an advanced technique that only works in PlatformIO - not Arduino IDE)

Yes, I use PlatformIO. and this is what I did.

> Not sure what you mean here. It sounds like you did fix it by setting the RF24 lib's SPI speed to 4MHz. 

I thought I had opened this up as an enhancement request, not a bug? but yes I resolved it by setting the RF24 clock speed lower.

I was suggesting whether it is possible for the RF24 library to some how default to the SD SPI clock speed if the SPI clock speed has been set. As opposed to setting the SPI clock speed in 2 places i.e.:

1. SD.begin; and
2. RF24_config.h or RF constructor.

- however, on reflection it is probably a void request as you may want your SPI devices running on different speeds.

Also as a record for anyone else that comes across this issue.

> Do you know if the SD card library you're using employs the begin/endTransaction()? If it doesn't then this may be messing up the RF24 lib when you init the SD card on the bus.

Yes it does.

Feel free to close this one if you like.

2bndy5, 2022-02-09:
> seems you can't get enough of nRF24 related support

🤣 pretty sure it could be classified as a clinical addiction. Really, I think of it as unforeseen opportunities to learn. And, I ❤️ Open Source.

---
>I thought I had opened this up as an enhancement request, not a bug? but yes I resolved it by setting the RF24 clock speed lower.

Apologies. I didn't fully understand your intentions. Ultimately, we don't want to add anything to the lib that assumes anything about the user's app (including other SPI-based devices/libs and their set SPI speed).

FWIW, the RF24 lib's default SPI speed used to be 4MHz (a default inherited from the Arduino AVR core), but that was changed because the radio can work with 10MHz max. I have been considering changing it back to 4MHz, but there needs to a serious presidence (ie if a nRF24 clone doesn't work with the default 10MHz). We recently discovered a regression concerning the ATTiny85's 3 pin solution (developed when default was 4MHz), but that seems too specific to revert the SPI default speed.

In the end, I think you lucked out here because you didn't buy one of those cheapo SD card modules from amazon, which tend to have many other problems (like an unexposed CS pin that's always active low) when you try to use it on a bus with multiple slave devices.

closed 2022-02-09 by 2bndy5