nRF24/RF24 #828: support esp8266

N1d45, 2022-02-10:
Hello. 
Is it possible to use the nRF24l01+ module with an ESP8266 (D1 Mini)? I have problems with a sketch that works fine on an Arduino Nano but not on an ESP8266.
[code]
#include <SPI.h>
#include "RF24.h"

RF24 radio(D2,D8); // CE, CSN
uint8_t address[][6] = {"1_Chi", "2_Chi", "3_Chi", "4_Chi", "5_Chi", "6_Chi"}; // Adressen der Kanäle, sollten bei den beteiligten Modulen gleich sein

struct PaketStruktur{
  uint8_t Sensor_ID;          // (0..255) - Eindeutige ID des Senders
  int16_t Temp;               // Wert für die Temperatur * 10
  float Humi;                 // Wert für die Luftfeuchtigkeit %
  uint16_t Druck;             // Wert für den Druck hPa
  uint16_t Vcc;               // Wert der Batteriespannung in mV
  uint16_t Wert1;             // Zusätzlicher Wert 1
  uint16_t Wert2;             // Zusätzlicher Wert 2
};

PaketStruktur payload = { 50, 0, 0, 0, 0, 0, 0 };
//_______________________ ID  T  H  P  V  1  2 __

void setup() {
  Serial.begin(9600);
  pinMode(2,OUTPUT);                              // LED an D4
  
  if(radio.begin()){                              // Stellt SPI Verbindung zum nRF24-Modul her
    radio.setChannel(84);                         // Sendekanal (zwischen 0 - 125), untere Kanäle oft von WIFI belegt, 0 - 84 in Europa erlaubt
    radio.setPayloadSize(sizeof(payload));        // Sende nur so viele Daten, wie nötig
    radio.setAutoAck(true);                       // Automatische Bestätigung bei Empfang Daten
    radio.setPALevel(RF24_PA_MAX);                // Sendeleistung definieren (RF24_PA_MAX), (RF24_PA_HIGH), ...
    radio.setDataRate(RF24_250KBPS);              // Übertragungsgeschwindigkeit definieren
    radio.setRetries (3, 8);                      // Abstand von 1ms (X+1 steps of 250μs)  8x wiederholen
    radio.openWritingPipe(address[1]);            // Erstellt Sendekanal mit Addresse
    radio.openReadingPipe(0, address[0]);         // Erstellt Empfangskanal mit Nummer (0-5) mit Addresse
    radio.startListening();                       // Setzt das Modul in den Sender-Modus
    Serial.println("Ich war hier");
  }
  else
  {
    digitalWrite(2,HIGH);
  }
}

void loop() {
  uint8_t pipe;
  if (radio.available(&pipe)) {                   // Wenn Daten Verfügbar
    radio.read(&payload, sizeof(payload));        // Lesen des Payloads
    Serial.print("Pipe: "); Serial.println(pipe);
    Serial.print("ID: "); Serial.println(payload.Sensor_ID);
    Serial.print("Temp: "); Serial.println(payload.Temp);
    Serial.print("Humi: "); Serial.println(payload.Humi);
    Serial.print("Druck: "); Serial.println(payload.Druck);
    Serial.print("mVCC: "); Serial.println(payload.Vcc);
    Serial.print("Reed1: "); Serial.println(payload.Wert1);
    Serial.print("Reed2: "); Serial.println(payload.Wert2);
    Serial.println();
    digitalWrite(2,!digitalRead(2));
  }
}
[/code]
On the ESP8266 radio.avaiable does not become true
On the Nano radio.avaiable becomes true and the data is written to the payload.

What can I do to make it work on the ESP8266?

TMRh20, 2022-02-10:
ESP8266 is compatible with the RF24 library, I would suggest double checking your wiring etc. following the diagram found [here](https://www.mysensors.org/build/connect_radio) for ESP.

Beyond that, you can use the radio.printDetails() function after calling radio.begin() to print out some debugging info and post it here if no luck.

TMRh20, 2022-02-10:
You will also want to re-arrange your data structure so the data aligns on both devices. 
```
struct PaketStruktur{
float Humi; // Wert für die Luftfeuchtigkeit %
uint16_t Druck; // Wert für den Druck hPa
uint16_t Vcc; // Wert der Batteriespannung in mV
uint16_t Wert1; // Zusätzlicher Wert 1
uint16_t Wert2; // Zusätzlicher Wert 2
int16_t Temp; // Wert für die Temperatur * 10
uint8_t Sensor_ID; // (0..255) - Eindeutige ID des Senders
};
```
See https://tmrh20.blogspot.com/2020/09/transferring-data-between-systems-using.html for more info

2bndy5, 2022-02-11:
> What surprises me is that
if (radio.available(&pipe))
does not return true, so the if query is not executed. If it was only the data structure, then data should still come into the payload, even if not readable.

You have to remember that data types are not always allocated to the same amount of memory for all processors. We have seen this before with the ESP8266 Arduino core. Simply put, If `sizeof(payload)` does not equal the same number on both RX and TX nodes, then the radios are congifured to expect/use differently sized payloads. It is vital that `radio.payloadSize()` be set to the same setting for both nodes when not using dynamically sized payloads (a feature specific to the radio, not your code). Otherwise `radio.available*()` will return false because it may be finding the transmitted payload, but it has the wrong size that the radio is configured to expect.

N1d45, 2022-02-11:
Thank you very much, that was the problem.

I am happy that you were able to solve it.

2bndy5, 2022-02-11:
Awesome!
I'm closing this issue now that it is solved. @N1d45 Can you also post this solution on the Arduino forums? That way, if anyone else with a similar problem finds your question, they could try applying your solution.

closed 2022-02-11 by 2bndy5

N1d45, 2022-02-11:
Ja, habe ich soeben getan. Und ich habe ein Link hierher hinterlassen.

Danke

gufranah, 2024-08-26:
> Thank you very much, that was the problem.
> 
> I am happy that you were able to solve it.

hello please I have question, what is the program to do this circiut