nRF24/RF24 #829: [Question] Cubecell with nRF24L01

THPP123, 2022-02-16:
I have been trying to connect a nRF24L01 to a Cubecell AB01 without much luck. Most of the time, the radio.available() function always returns true resulting in a bunch of garbage. I have figured out that the nRF24L01 doesn't want to configure properly. With printPrettyDetails() I can confirm that the settings are not stored.

However, with the example sketch contained within the RF24 library, everything works fine (confirming that the AB01 and nRF24L01 are working fine). But as soon as I change the datatype of the receiver payload to uint8_t, the radio.available() is always true, resulting in a bunch of received messages with garbage. PrintPrettyDetails() also shows the wrong settings.

Does anyone have experience with this? I have also tried the example sketch from Robin2. This results in a lot of garbage as described before.

I am using:

- nRF24L01 with pcb antenna. I am periodicially using an esp32 to test that the nRF24L01 is still functioning okay
- a heltec AB01. The nRF24L01 is connected to the spi interface and CE -> GPIO3 and CSN -> GPIO2
- To program I'm using platformio. But in desperation I am also able to use the Arduino IDE (exactly the same behaviour)

I have tried anything I could think of, without success. Any help would be appreciated!

2bndy5, 2022-02-16:
This scenario is described in the [COMMON_ISSUES.md](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#payloads-receivedsent-are-inaccurate-or-printdetails-outputs-the-unexpected-values). Try lowering the SPI speed.

THPP123, 2022-02-16:
Thanks for your reply!

I have changed the code to the following: 
```
/**
 * See documentation at https://nRF24.github.io/RF24
 * See License information at root directory of this library
 * Author: Brendan Doherty (2bndy5)
 */

/**
 * A simple example of sending data from 1 nRF24L01 transceiver to another.
 *
 * This example was written to be used on 2 devices acting as "nodes".
 * Use the Serial Monitor to change each node's behavior.
 */
#include <SPI.h>
#include "printf.h"
#include "RF24.h"

// instantiate an object for the nRF24L01 transceiver
RF24 radio(GPIO3, GPIO2, 4000000); // using pin 7 for the CE pin, and pin 8 for the CSN pin

// Let these addresses be used for the pair
uint8_t address[][6] = {"1Node", "2Node"};
// It is very helpful to think of an address as a path instead of as
// an identifying device destination

// to use different addresses on a pair of radios, we need a variable to
// uniquely identify which address this radio will use to transmit
bool radioNumber = 1; // 0 uses address[0] to transmit, 1 uses address[1] to transmit

// Used to control whether this node is sending or receiving
bool role = false;  // true = TX role, false = RX role

// For this example, we'll be using a payload containing
// a single float number that will be incremented
// on every successful transmission
float payload = 0;

void setup() {

  Serial.begin(115200);
  while (!Serial) {
    // some boards need to wait to ensure access to serial over USB
  }
  pinMode(Vext, OUTPUT);
  digitalWrite(Vext, LOW);
  delay(10);
  // initialize the transceiver on the SPI bus
  if (!radio.begin()) {
    Serial.println(F("radio hardware is not responding!!"));
    while (1) {} // hold in infinite loop
  }

  // print example's introductory prompt
  Serial.println(F("RF24/examples/GettingStarted"));
  
  radioNumber = 1;
  Serial.print(F("radioNumber = "));
  Serial.println((int)radioNumber);

  // Set the PA Level low to try preventing power supply related problems
  // because these examples are likely run with nodes in close proximity to
  // each other.
  radio.setPALevel(RF24_PA_LOW);  // RF24_PA_MAX is default.

  // save on transmission time by setting the radio to only transmit the
  // number of bytes we need to transmit a float
  radio.setPayloadSize(sizeof(float)); // float datatype occupies 4 bytes

  // set the TX address of the RX node into the TX pipe
  radio.openWritingPipe(address[radioNumber]);     // always uses pipe 0

  // set the RX address of the TX node into a RX pipe
  radio.openReadingPipe(1, address[!radioNumber]); // using pipe 1

  // additional setup specific to the node's role
  if (role) {
    radio.stopListening();  // put radio in TX mode
  } else {
    radio.startListening(); // put radio in RX mode
  }

  // For debugging info
  // printf_begin();             // needed only once for printing details
  // radio.printDetails();       // (smaller) function that prints raw register values
  radio.printPrettyDetails(); // (larger) function that prints human readable data

} // setup

void loop() {

  // This device is a RX node

  uint8_t pipe;
  if (radio.available(&pipe)) {             // is there a payload? get the pipe number that recieved it
    uint8_t bytes = radio.getPayloadSize(); // get the size of the payload
    radio.read(&payload, bytes);            // fetch payload from FIFO
    Serial.print(F("Received "));
    Serial.print(bytes);                    // print the size of the payload
    Serial.print(F(" bytes on pipe "));
    Serial.print(pipe);                     // print the pipe number
    Serial.print(F(": "));
    Serial.println(payload);                // print the payload's value
  }

} // loop
```

Unfortunatly, even with datatype float, I get an infinite amount of "Received 4 bytes on pipe 0: 0.00" on the serial monitor. 
I have completely unplugged the board for more than 30 secs to completely depower the radio.

2bndy5, 2022-02-16:
I still think there's a problem with the SPI connection, but it is wierd that the example worked without modification (unless I read the OP wrong).

The other possibility is the difference in allocation of data types, but this doesn't account for erroneous output in `printDetails()`.

>But as soon as I change the datatype of the receiver payload to uint8_t, the radio.available() is always true

The `payloadSize()` setting must be set to the same value on both nodes. Changing it on only 1 node will likely result in mis-interpretted data from received payloads. Sometimes a `uint8_t` is allocated to more than a single byte on certain boards. 

---
I've never heard of a Cubecell AB01 board. Looking at the Arduino core for that board, I see that they offer a modified `SPI::begin()` function. If they don't support the universal `SPI::begin()` (without args), then the custom Arduino Core might be messing up the RF24 lib. Again, I find it weird that the example reportedly worked without modification though.

2bndy5, 2022-02-16:
SPI lib looks compatible for that Arduino core.

THPP123, 2022-02-16:
The interesting part about this whole situation for me is that the transmitter is not turned on. After some debugging with printDetails(), I'm beginning to think that the nRF24L01 is sending data to itself. Since the tx and pipe0 address are the same 

printDetails() output
```
radioNumber = 1
SPI Frequency           = 10 Mhz
Channel                 = 14 (~ 2414 MHz)
RF Data Rate            = 2 MBPS
RF Power Amplifier      = PA_MAX
RF Low Noise Amplifier  = Disabled
CRC Length              = 16 bits
Address Length          = 4 bytes
Static Payload Length   = 4 bytes
Auto Retry Delay        = 250 microseconds
Auto Retry Attempts     = 14 maximum
Packets lost on
    current channel     = 0
Retry attempts made for
    last transmission   = 14
Multicast               = Disabled
Custom ACK Payload      = Enabled
Dynamic Payloads        = Enabled
Auto Acknowledgment     = 0b001110
Primary Mode            = TX
TX address              = 0x646f4e320e
pipe 0 (closed) bound   = 0x646f4e320e
pipe 1 ( open ) bound   = 0x646f4e310e
pipe 2 ( open ) bound   = 0x0e
pipe 3 ( open ) bound   = 0x0e
pipe 4 (closed) bound   = 0x0e
pipe 5 (closed) bound   = 0x0e
```

2bndy5, 2022-02-16:
pipe 0 is the transmitting pipe. pipe 0 and the TX address are the same because that is a requirement of using auto-ack (which is enabled by default).

2bndy5, 2022-02-16:
It seems as though each byte read from the SPI MISO has `0x0e`. Usually this is the radio's STATUS byte after a fresh boot. Maybe the Arduino core isn't using the right endianess?

2bndy5, 2022-02-16:
I'm not entirely sure about this, but it looks like the SPI lib in that core only defines `SPI_MSBFIRST`  but not `MSBFIRST` (the RF24 lib uses `MSBFIRST`). I don't see a `#define MSBFIRST SPI_MSBFIRST` for that specific board, but it is defined for another board in that core. @THPP123 are you seeing any error/warnings from the compiler?

THPP123, 2022-02-17:
This is the output from the build command (after a clean all)
```
Processing cubecell_board (platform: asrmicro650x; board: cubecell_board; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/asrmicro650x/cubecell_board.html
PLATFORM: ASR Microelectronics ASR650x (1.3.1) > Heltec CubeCell-Board (HTCC-AB01)
HARDWARE: ASR6501 48MHz, 16KB RAM, 128KB Flash
PACKAGES:
 - framework-arduinoasrmicro650x 1.3.0
 - tool-cubecellelftool 0.0.1
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing nrf24/RF24 @ ^1.4.2

Unpacking  [------------------------------------]    0%
Unpacking  [------------------------------------]    1%
Unpacking  [------------------------------------]    2%
Unpacking  [#-----------------------------------]    3%
Unpacking  [#-----------------------------------]    4%
Unpacking  [##----------------------------------]    5%
Unpacking  [##----------------------------------]    6%
Unpacking  [##----------------------------------]    7%
Unpacking  [##----------------------------------]    8%
Unpacking  [###---------------------------------]    8%
Unpacking  [###---------------------------------]    9%
Unpacking  [###---------------------------------]   10%
Unpacking  [####--------------------------------]   11%
Unpacking  [####--------------------------------]   12%
Unpacking  [####--------------------------------]   13%
Unpacking  [#####-------------------------------]   14%
Unpacking  [#####-------------------------------]   15%
Unpacking  [#####-------------------------------]   16%
Unpacking  [######------------------------------]   17%
Unpacking  [######------------------------------]   18%
Unpacking  [#######-----------------------------]   19%
Unpacking  [#######-----------------------------]   20%
Unpacking  [#######-----------------------------]   21%
Unpacking  [########----------------------------]   22%
Unpacking  [########----------------------------]   23%
Unpacking  [########----------------------------]   24%
Unpacking  [#########---------------------------]   25%
Unpacking  [#########---------------------------]   26%
Unpacking  [#########---------------------------]   27%
Unpacking  [##########--------------------------]   28%
Unpacking  [##########--------------------------]   29%
Unpacking  [##########--------------------------]   30%
Unpacking  [###########-------------------------]   30%
Unpacking  [###########-------------------------]   31%
Unpacking  [###########-------------------------]   32%
Unpacking  [############------------------------]   33%
Unpacking  [############------------------------]   34%
Unpacking  [############------------------------]   35%
Unpacking  [#############-----------------------]   36%
Unpacking  [#############-----------------------]   37%
Unpacking  [#############-----------------------]   38%
Unpacking  [##############----------------------]   39%
Unpacking  [##############----------------------]   40%
Unpacking  [##############----------------------]   41%
Unpacking  [###############---------------------]   42%
Unpacking  [###############---------------------]   43%
Unpacking  [################--------------------]   44%
Unpacking  [################--------------------]   45%
Unpacking  [################--------------------]   46%
Unpacking  [################--------------------]   47%
Unpacking  [#################-------------------]   47%
Unpacking  [#################-------------------]   48%
Unpacking  [#################-------------------]   49%
Unpacking  [##################------------------]   50%
Unpacking  [##################------------------]   51%
[... 143 lines cut ...]
Compiling .pio\build\cubecell_board\FrameworkArduino\projects\PSoC4\UART_2_SCB_IRQ.c.o
Compiling .pio\build\cubecell_board\FrameworkArduino\projects\PSoC4\UART_2_SPI_UART.c.o
Compiling .pio\build\cubecell_board\FrameworkArduino\projects\PSoC4\UART_2_SPI_UART_INT.c.o
Compiling .pio\build\cubecell_board\FrameworkArduino\projects\PSoC4\UART_2_UART.c.o
Compiling .pio\build\cubecell_board\FrameworkArduino\projects\PSoC4\cyPm.c.o
Compiling .pio\build\cubecell_board\FrameworkArduino\projects\PSoC4\cy_em_eeprom.c.o
Compiling .pio\build\cubecell_board\FrameworkArduino\projects\PSoC4\cybootloader.c.o
Compiling .pio\build\cubecell_board\FrameworkArduino\projects\PSoC4\cymetadata.c.o
Compiling .pio\build\cubecell_board\FrameworkArduino\projects\PSoC4\cyutils.c.o
Compiling .pio\build\cubecell_board\FrameworkArduino\wiring_shift.cpp.o
Archiving .pio\build\cubecell_board\libFrameworkArduino.a
Indexing .pio\build\cubecell_board\libFrameworkArduino.a
Linking .pio\build\cubecell_board\firmware.elf
Checking size .pio\build\cubecell_board\firmware.elf
Building .pio\build\cubecell_board\firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.6% (used 920 bytes from 16384 bytes)
Flash: [===       ]  26.9% (used 35300 bytes from 131072 bytes)
Building .pio\build\cubecell_board\firmware.cyacd
======================================================== [SUCCESS] Took 23.50 seconds ========================================================

Terminal will be reused by tasks, press any key to close it.
```

Can I try to manually define that MSBFIRST to see if that solves the problem?

THPP123, 2022-02-17:
When writing the following: 
```
#ifndef MSBFIRST
#define MSBFIRST = SPI_MSBFIRST
#endif
````

The piece of code is grayed out, giving the impression that it is defined. When looking up the definition, I can see that it is defined in Arduino.h in the ASR650x directory (chip on this board)

2bndy5, 2022-02-17:
A C macro does not use the assignment operator (`=`):
```c
#define MSBFIRST SPI_MSBFIRST
```

---
Thanks for the compiler output. I doesn't seem to complain about `MSBFIRST` not defined, so I'm guessing it is defined somewhere... Thanks to your lead on the Arduino.h, I found it is defined:
https://github.com/HelTecAutomation/CubeCell-Arduino/blob/e23d3496e1e9560ecba171da3731f3134bf33c2d/cores/asr650x/Arduino.h#L48

I'm guessing if you run the actual (older) `printDetails()` (not `printPrettyDetails()`), then you'd see a lot more `0x0e`. Given that you didn't modify the example to get that erroneous output, I have to assume there is something wrong with the execution of SPI transaction (likely from the MCU side - not the radio or RF24 library side)

2bndy5, 2022-02-17:
Since you're using platformIO, you could define `SERIAL_DEBUG` to see a more in-depth output of SPI transactions:
```ini
# in platformio.ini
[env-name]
board = "..."
build_flags = 
    -DSERIAL_DEBUG=1
```

I'm kinda tapped out for possible problems/solutions at this point.

THPP123, 2022-02-17:
This is the output
```
Copyright @2019-2020 Heltec Automation.All rights reserved.
write_register(04,5f)
write_register(06,07)
write_register(1d,00)
write_register(1c,00)
write_register(01,3f)
write_register(02,03)
write_register(11,20)
write_register(12,20)
write_register(13,20)
write_register(14,20)
write_register(15,20)
write_register(16,20)
write_register(03,03)
write_register(05,4c)
write_register(07,70)
write_register(e2)
write_register(e1)
write_register(00,0c)
RF24/examples/GettingStarted
radioNumber = 1
write_register(06,0b)
write_register(11,04)
write_register(12,04)
write_register(13,04)
write_register(14,04)
write_register(15,04)
write_register(16,04)
write_register(02,0e)
write_register(00,0f)
write_register(07,70)
write_register(02,0e)
SPI Frequency           = 10 Mhz
Channel                 = 14 (~ 2414 MHz)
RF Data Rate            = 2 MBPS
RF Power Amplifier      = PA_MAX
RF Low Noise Amplifier  = Disabled
CRC Length              = 16 bits
Address Length          = 4 bytes
Static Payload Length   = 4 bytes
Auto Retry Delay        = 250 microseconds
Auto Retry Attempts     = 14 maximum
Packets lost on
    current channel     = 0
Retry attempts made for
    last transmission   = 14
Multicast               = Disabled
Custom ACK Payload      = Enabled
Dynamic Payloads        = Enabled
Auto Acknowledgment     = 0b001110
Primary Mode            = TX
TX address              = 0x646f4e320e
pipe 0 (closed) bound   = 0x646f4e320e
pipe 1 ( open ) bound   = 0x646f4e310e
pipe 2 ( open ) bound   = 0x0e
pipe 3 ( open ) bound   = 0x0e
pipe 4 (closed) bound   = 0x0e
pipe 5 (closed) bound   = 0x0e
[Reading 4 bytes 0 blanks]
write_register(07,40)
Received 4 bytes on pipe 3: 0.00
[Reading 4 bytes 0 blanks]
write_register(07,40)
Received 4 bytes on pipe 0: 0.00
[Reading 4 bytes 0 blanks]
write_register(07,40)
Received 4 bytes on pipe 0: 0.00
[Reading 4 bytes 0 blanks]
write_register(07,40)
Received 4 bytes on pipe 0: 0.00
[Reading 4 bytes 0 blanks]
write_register(07,40)
Received 4 bytes on pipe 0: 0.00
[Reading 4 bytes 0 blanks]
write_register(07,40)
Received 4 bytes on pipe 0: 0.00
[Reading 4 bytes 0 blanks]
write_register(07,40)
Received 4 bytes on pipe 0: 0.00
[Reading 4 bytes 0 blanks]
write_register(07,40)
Received 4 bytes on pipe 0: 0.00
[Reading 4 bytes 0 blanks]
```
This output is cut-off, since the same keeps happening. In a couple of hours, I should have access to a (basic) logic analyzer. Mayby that will give some more insight. Like you said yourself, I'm at a loss for a possible solution/cause.

THPP123, 2022-02-17:
This is the output of the printDetails() function. 
```
SPI Speedz      = 10 Mhz
STATUS          = 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1    = 0x646f4e320e 0x646f4e310e
RX_ADDR_P2-5    = 0x0e 0x0e 0x0e 0x0e
TX_ADDR         = 0x646f4e320e
RX_PW_P0-6      = 0x0e 0x0e 0x0e 0x0e 0x0e 0x0e
EN_AA           = 0x0e
EN_RXADDR       = 0x0e
RF_CH           = 0x0e
RF_SETUP        = 0x0e
CONFIG          = 0x0e
DYNPD/FEATURE   = 0x0e 0x0e
Data Rate       = 2 MBPS
Model           = nRF24L01+
CRC Length      = 16 bits
PA Power        = PA_MAX
ARC             = 14
```

2bndy5, 2022-02-17:
Debug output looks right, so I don't think its a RF24 lib problem. If you get the logic analyzer setup, there are a couple undocumented things you should know:
1. `write_register(07,40)` translates to "wrote 0x40 to register 0x07". And, `write_register(e2)` is a nRF24L01 SPI command (like flushes the TX or RX FIFO). The only nRF24L01 SPI commands that use more than 1 byte are the ones reading/writing payloads to the respective FIFOs.
2. `RF24::available()` uses a 1 byte SPI transaction to get an updated STATUS byte. When comparing the logic analyzer to the debug statements, you'll notice that the calls to `available()` show up as `0xFF` in the logic analyzer's MOSI output, but they aren't shown in the debug output for brevity and speed.

---
I am fairly certain the the `0x0e` should be the first byte returned by an SPI transaction (the nRF24L01 is a full-duplex slave device), then anything that follows should be the actual register data. I wouldn't be surprised if the logic analyzer confirms my suspicion.

2bndy5, 2022-02-17:
I wasn't aware you'd made alterations to anything but the example. I'm glad you got it working on a fresh slate.