nRF24/RF24 #830: NRF24.read() causes my CPU to spike

Satnet, 2022-02-19:
NRF24.read(); causes my CPU to go to 97% after running for about 15 seconds. The system slows down and after a while it eventually hangs or becomes so slow that it seems like it. I am using interrupt and besides this issue it all works very well. Those first 15 seconds the whole thing is super fast. Running a Teensy at 396mhz. I have narrowed the issue down to this function, if I comment it out, CPU is at 3%.

The transmitter is sending a payload every 250ms. Might be a bug or something else, I am unsure and I don't know what to do at this point.

I appreciate any help given!

2bndy5, 2022-02-19:
src for `read()`:
https://github.com/nRF24/RF24/blob/aaf0f53cfa4484d9bc0d3f6c2d5be91d8cdfe931/RF24.cpp#L1495-L1504

src for `read_register()`:
https://github.com/nRF24/RF24/blob/aaf0f53cfa4484d9bc0d3f6c2d5be91d8cdfe931/RF24.cpp#L377-L439

I really can't imagine why this function would cause a CPU spike. The lib is simply reading the data from the radio's register and resetting the RX_READY flag. I suspect it has something to do with how your app is using interrupts; please post your code if you can.

Satnet, 2022-02-19:
Thanks for the super fast reply! Here is the test code for my receiver.

```
#include "Arduino.h"
#include <SPI.h>
#include <RF24.h>

int NRF_IRQ = 1;

RF24 NRF24(0, 10);

const uint64_t pAddress = 0xB00B1E5000LL;  //Create a pipe addresses for the 2 nodes to communicate over, the "LL" is for LongLong type
uint32_t CAN_ID;
uint8_t payloadArray[12];

bool readyToRock = false;

void NRF24_InterruptReceive();

void NRF24_Initiate() {
  NRF24.begin();  //Start the nRF24 module
  NRF24.setAutoAck(1);                    // Ensure autoACK is enabled so rec sends ack packet to let you know it got the transmit packet payload
  NRF24.enableAckPayload();         //allows you to include payload on ack packet
  NRF24.maskIRQ(1,1,0);               //mask all IRQ triggers except for receive (1 is mask, 0 is no mask)
  NRF24.setPALevel(RF24_PA_LOW); //Set power level to low, won't work well at higher levels (interfer with receiver)
  NRF24.openReadingPipe(1, pAddress);      //open pipe o for recieving meassages with pipe address
  NRF24.startListening();                 // Start listening for messages
  attachInterrupt(NRF_IRQ, NRF24_InterruptReceive, FALLING);  //Create interrupt: 0 for pin 2 or 1 for pin 3, the name of the interrupt function or ISR, and condition to trigger interrupt
}

void NRF24_InterruptReceive() {
  while (NRF24.available()) {
    NRF24.read(&payloadArray, NRF24.getPayloadSize()); // CAUSES CPU TO SPIKE. COMMENT IT OUT AND CPU IS CHILLING.

    CAN_ID = (uint32_t)(payloadArray[8] << 24) | (uint32_t)(payloadArray[9] << 16) | (uint32_t)(payloadArray[10] << 8 ) | (uint32_t)(payloadArray[11]);
    Serial.println(CAN_ID, HEX);
  }
}

void setup() {
  Serial.begin(57600);
  Serial.print(CrashReport);
}

void loop() {
  if (!readyToRock && millis() > 5000) {
    NRF24_Initiate();

    readyToRock = true;
  }
}
```

2bndy5, 2022-02-19:
Somehow I knew you were using ACK payloads. When the radio receives a payload, the RX_DR flag is asserted. Additionally, if the radio responds with an ACK payload, then the TX_DS ("data sent") flag is also asserted.

I see you've used `maskIRQ()` to prevent the IRQ pin from going active when a ACK payload was sent. But, its still good practice to call `whatHappened()` on an IRQ callback (that's what is specifically for).

Try using `whatHappened()` to reset the flags upon enter the `NRF24_InterruptReceive()` function. Furthermore, you can determine if `read()` should be called by examining the results from `whatHappened()`.

---
```cpp
NRF24.read(&payloadArray, NRF24.getPayloadSize());
```
You are reading 32 bytes into a 12 byte array (and possibly causing a memory overflow). There are several ways to fix this, but I'm not going to tell you how to write your app.

Also, I find it pretty weird that you initialize the radio in `loop()`. Please consider moving `NRF24_Initiate()` into `setup()`. Why does your app need to wait 5 seconds? Is it because you're not using `while (!Serial) {}` in `setup()`?

2bndy5, 2022-02-19:
I forgot to mention that ACK payloads require `enableDynamicPayloads()`, and you should be using `getDynamicPayloadSize()` in this case. But it is still important that you have allocated enough memory for the received payload.

The lib is smart enough to enable the dynamic payload size feature for pipes 0 and 1 when you call `enableAckPayloads()`. But if you intend to use any other pipes, then you should really call `enableDynamicPayloads()`.

Satnet, 2022-02-19:
Many valid points there!

ACK is used to tell the transmitter that the receiver has received the payload right? I really don’t need that part if it can be disabled. For my use case, if the receiver doesn’t receive then so be it. Most likely because of range issues. Basically I just want the transmitter to spit out data and the receiver will pick it up if it can without confirming it was received.

I used an interrupt example I found online. It also had a YouTube video where a guy explained the mask(1,1,0) thing. I’ll look into whatHappened().

regarding the 12 and 32 byte, the transmitter also only sends max 12 byte. But maybe uint32_t is still 32 and overflows? What would be the best solution for that?

I use a flag that’s set false upon start. After 5 seconds I initiate the NRF and change that flag. This is because I want the system to boot up first as I have a screen running LVGL. 

I’m only going to use one pipe. It’s set to 1 from the example I found.

thanks for taking the time, I appreciate the help!

2bndy5, 2022-02-20:
Have you looked at our [interruptConfigure example](https://github.com/nRF24/RF24/blob/master/examples/InterruptConfigure/InterruptConfigure.ino)? I get annoyed when people ignore official examples and go off of third party tutorials. I put a lot of work into those examples to explain proper usage.

You should also look at the [acknowledgementPayloads example](https://github.com/nRF24/RF24/blob/master/examples/AcknowledgementPayloads/AcknowledgementPayloads.ino) if you intend to keep using ACK payloads.

If you know the payloads received will only be 12 bytes, then pass 12 to `read()`. If you follow our examples, then you'd see how we use `sizeof(variable)`.

`uint32_t` is 4 bytes; `uint8_t` is 1 byte, but `uint8_t arr[12]` is a 12 element array in which each element allocated to 1 byte. If you never call `setPayloadSize()`, then `getPayloadSize()` will return 32 (despite the state of dynamic payloads feature for any pipe).

Satnet, 2022-02-20:
I understand that these examples take time to make. I have looked at them. However I do like videos as it explains in more depth. But obviously I need to learn more about this library. I will test that interrupt example and see where it takes me.

what I don’t understand and don’t see in the examples is what ACK is really doing. Is it a conformation function for the transmitter to know what the receiver has received? Or does it have to do with the interrupt functionality?

regarding the bytes, I thank you for that explanation. It makes sense and I will change my code.

2bndy5, 2022-02-20:
Are you aware we have written [extensive docs](https://nrf24.github.io/RF24/classRF24.html)? I understand the ease of watching videos, but there's no way they go deeper than reading the docs or datasheet will. When looking at the examples, it helps a lot to pay attention to the comments.

>what I don’t understand and don’t see in the examples is what ACK is really doing

ACK packets are a feature used by the radio to ensure payload delivery. ACK packets can have custom payloads attached. Looking at your code, I realize that you're not calling `writeAckPayload()`, so you don't need to call `enableAckPayloads()`. Auto-ack is enabled by default, so you don't have to call `setAutoAck(1)` either. By default, the ACK packets are have no payloads, but people sometimes want to send data back to the transmitter by adding a payload to the ACK packet (which is done using `writeAckPayload()`).

It sounds like you just want data to go from point A to point B. That's exactly what the [gettingStarted example](https://github.com/nRF24/RF24/blob/master/examples/GettingStarted/GettingStarted.ino) demonstrates. The lib's default configuration is well suited to do just that. I think your confusion has come from being saturated with too much general advice (and not enough explanation) from those youtubers.

2bndy5, 2022-02-20:
BTW, `NRF24.setPALevel(RF24_PA_LOW);` is a power saving hack meant to compensate for inadequate power supplies when using a PA/LNA radio (the kind with bigger/detachable antennas). If you want to span distances greater than a few feet, then you should use a higher value.

Be warned, that the PA/LNA radios require a lot more power (~150 mA) than regular nRF24L01 radios at higher PA (power amplitude) levels. If your display is hogging a lot of the current from your MCU's power regulator, then the radio will suffer and result in failed transmissions.

Satnet, 2022-02-20:
I’m going through all your points today. Hopefully I’ll get rid of my issue where CPU power is spiking. I’ll read those docs for sure!

regarding power, I have plenty of power onboard my PCB.

Satnet, 2022-02-20:
Miss clicked close with comment…

I’m going to read those docs today and hopefully get past the CPU spike problem I’m having. The getStarted example is good and I’ve tested it before. But in my case I need interrupt because the MCU is single thread so it gets busy if I don’t use interrupt.

there is plenty of power on my board. I have successfully transmitted with MAX power between the transmitter and the receiver. Everything works great to be honest except for my CPU spiking after 15 seconds. I’m unsure why but LVGL and the NRF code doesn’t seem to like eachother.

I’ll test more and see where it all gets me.

Satnet, 2022-02-20:
I have now been testing all morning, and I do wanna underline that I know it's not a fault or anything like that with this library. It just happen to be so that LVGL and NRF24 isn't playing nice together. Seperately the both work perfectly. This NRF library is very nice and I can successfully transmit and receive on long distances.

LVGL needs to run it's course perfectly or it'll freeze or crash. The NRF library get's stuck sometimes somewhere for a short period of time. This is enough to freeze LVGL.

Could you think of any place where this would happen?

2bndy5, 2022-02-20:
I haven't seen or heard of anyone else that has tried to use RF24 lib + LVDL. TBH, I never heard of LVGL before you mentioned it. 

Its starting to sound like you need to use the radio on a separate MCU like a networking co-processor. I'm afraid to ask what "isn't playing nce together" actual entails because I don't know anything about the LVGL lib nor the display you're using. I also don't know why you called the Teensy a single thread CPU....

I honestly don't know how to help any further.

Satnet, 2022-02-20:
> I haven't seen or heard of anyone else that has tried to use RF24 lib + LVDL. TBH, I never heard of LVGL before you mentioned it.
>
> Its starting to sound like you need to use the radio on a separate [...]

From what I can narrow it down to, LVGL don't like the While loops of NRF24 library and the core SPI library of the Teensy. LVGL is very sensitive it seems.

I will investigate this further with LVGL and Teensy, thank you for taking the time, I learned a few things from you! :)

closed 2022-02-20 by Satnet

Satnet, 2022-02-21:
@2bndy5, I just wanted to let you know that this was solved by a guy in the PJRC forum. And as usual, it was the most simple thing you could imagine.

The receive interrupt function I have, contains the receive code logic, basically printing out the received data.
Instead of doing it this way, he suggested putting a flag in the interrupt function and then have the actual receive code in the loop in an IF (flag), and it just works now. It's always the smallest stuff.

I wanna thank you once again for trying to help, I appreciate that. And I will never forget how important flags can be when coding!