nRF24/RF24 #831: ATTiny85 3 pin config with correct analog reading

kiredewit, 2022-02-26:
I have a problem with reading analog input from the ATTINY85(PB3) when using the setup described in the [ATTiny manual](https://github.com/nRF24/RF24/blob/master/docs/attiny.md)/[instructables](https://www.instructables.com/NRF24L01-With-ATtiny85-3-Pins/)/[source](http://nerdralph.blogspot.com/2014/01/nrf24l01-control-with-3-attiny85-pins.html).

when I connect a [capacitive soil moisture level 1.2](https://www.circuitschools.com/interface-capacitive-soil-moisture-sensor-v1-2-with-arduino-lcd-and-oled/) to PB3, I expect a value around the 600. 

With only the sensor connected to the bread board, the values are correct. 

When I use the setup as described in the manuals, the analog read value is building up to 1023.(the code is without communication with the nrf2401, just a test code for reading the analog input on PB3). 

There is no problem with sending and receiving data. In this case the sensor is only transmitting data.(receiver is a wemos D1 mini)

Setup:
FT232RL (for power and serial communication, 5V, original power source will be a 9V battery)
ATTiny85
[LM1117T 3.3V ](https://nl.mouser.com/ProductDetail/Texas-Instruments/LM1117T-33-NOPB?qs=X1J7HmVL2ZF7RIQB6O0mNg%3D%3D)
nrf24l01

[image]

### Transmitter code
**platform.ini**:

```
[env:program_via_ArduinoISP]
board = attiny85
platform = https://github.com/platformio/platform-atmelavr.git
framework = arduino
upload_protocol = custom
board_build.f_cpu = 8000000L ; MCU frequency
upload_port = /dev/ttyUSB0
upload_speed = 19200
board_fuses.lfuse = 0xE2
board_fuses.hfuse = 0xDF
board_fuses.efuse = 0xFF
upload_flags =
    -C
    $PROJECT_PACKAGES_DIR/tool-avrdude/avrdude.conf
    -p
    $BOARD_MCU
    -P
    $UPLOAD_PORT
    -b
    $UPLOAD_SPEED
    -c
    stk500v1
upload_command = avrdude $UPLOAD_FLAGS -U flash:w:$SOURCE:i
lib_deps =
    SPI
    https://github.com/nRF24/RF24.git

```

**main.cpp**: (only sensorread is active)
```
#define CE_PIN 2
#define CSN_PIN 2 //Since we are using 3 pin configuration we will use same pin for both CE and CSN

#define RF24_CSN_SETTLE_HIGH_DELAY 64
#define RF24_CSN_SETTLE_LOW_DELAY 8

#include <Arduino.h>
#include <SoftwareSerial.h>
// #include "RF24.h"

// RF24 radio(CE_PIN, CSN_PIN);

struct DataStruct{
    uint8_t id;
    uint8_t humidity;
} sensorData;

int soilMoistureValue = 0;
uint8_t soilmoisturepercent = 0;
const int AirValue = 650;   //you need to replace this value with Value_1
const int WaterValue = 220;  //you need to replace this value with Value_2

int sensorPin = A3; // pin 3, PB3, IC pin 2.

SoftwareSerial mySerial(-1,4);  //rx, tx

const byte serverAddress[6] = "00001";

void setup() {
    mySerial.begin(9600);
    pinMode(sensorPin,INPUT);

    delay(1000);

//    radio.begin();
//    radio.openWritingPipe(serverAddress);
//    radio.setPALevel(RF24_PA_MIN);
//    radio.stopListening();
//
//    sensorData.id = 13;
}

void loop() {
    soilMoistureValue = analogRead(A1);
    mySerial.println(soilMoistureValue);
    soilmoisturepercent = map(soilMoistureValue, AirValue, WaterValue, 0, 100);

    sensorData.humidity = soilmoisturepercent>100?100:soilmoisturepercent<0?0:soilmoisturepercent;

    mySerial.print("--[");
    mySerial.print(sensorData.id);
    mySerial.print("---");
    mySerial.print(sensorData.humidity);
    mySerial.print("---");
    mySerial.print(sizeof(sensorData));
    mySerial.println("]");

    // radio.write(&sensorData, sizeof(sensorData));
    delay(1000);
}
```

**Some Serial output**(i was expecting a value around the 600/700):
1022
--[0---100---2]
1015
--[0---100---2]
1013
--[0---100---2]
1011
--[0---100---2]
1008
--[0---100---2]
1010
--[0---100---2]
1010
--[0---100---2]
1016
--[0---100---2]
1021
--[0---100---2]
1023
--[0---100---2]

If I remove the ceramic capacitor and the VCC for nrf2401 the analog value for PB3 is changing to a lower number, still not in the correct range. Can someone explain why the  analog read on PB3 give me not the expecting value with the full setup?

2bndy5, 2022-02-26:
This sounds like a power problem. The [L1117 datasheet recommends tantalum capacitors (1 on input and 1 on output)](https://www.ti.com/lit/ds/symlink/lm1117.pdf?HQS=dis-mous-null-mousermode-dsf-pf-null-wwe&DCM=yes&ref_url=https%3A%2F%2Fcn.mouser.com%2F&distId=26#%5B%7B%22num%22%3A120%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C0%2C171.9%2C0%5D). Its hard to tell from the picture what you're using in the circuit.

2bndy5, 2022-02-26:
If you're using the 3-wire approach, then you need to drop the SPI speed to 4MHz. This can be done either in platformio.ini
```ini
[env...]
board="..."
build_flags=
    -DRF24_SPI_SPEED=4000000
```
or in RF24 c'tor
```cpp
RF24 radio(CE_PIN, CSN_PIN, 4000000);
```

kiredewit, 2022-02-26:
> If you're using the 3-wire approach, then you need to drop the SPI speed to 4MHz. This can be done either in platformio.ini
> 
> ```ini
> [env...]
> board="..."
> build_flags=
>     -DRF24_SPI_SPEED=4000000
> ```
> 
> or in RF24 c'tor
> 
> ```c++
> RF24 radio(CE_PIN, CSN_PIN, 4000000);
> ```

I try this before. 

The communication part is working with and without setting the SPI speed. When I only disable the code for communications there are still bad results. I don`t think that the SPI speed is the problem.

kiredewit, 2022-02-26:
:( it`s not closed. How can i reopen this?

2bndy5, 2022-02-26:
I was trying to focus on the capacitors and resistors you're using, not the regulator. BJTs (like your regulator) often require capacitors on each end to function properly. If the regulator is adjustable, then there are required resistors in addition to the capacitors. This is all detailed by the L1117 datasheet (see the link I posted). When I say "power", I'm talking about the voltage multiplied by the current (P = V * I). If 800mA is the specification maximum, then that heavily depends on the circuit (made of resistors and capacitors) you're using. Note that it helps to measure the actual voltage and current rather than telling me what you've read in the specifications or tutorials.

This is hard to diagnose since you only describe the problem with very little about the circuit. An actual schematic would help.

kiredewit, 2022-02-26:
> s

Thanks for the advice and tips for so far. You  are right about missing components when using a L1117.  To limit the possibility for more mistakes  I removed the regulator and use a 3.3v power source(from the FTDI).

When I use 3.3v from the FTDI and remove the regulator, I still get the same problem with analog read on PB3. 

The actual scheme is now almost the same as in the manual. I removed the led and 5v, because the power source is 3.3V. 

I connected the sensor AOUT to PB3. 

```text
                                                                                       ^^
                             +-\/-+           nRF24L01   CE, pin3 ------|              //
                       PB5  1|o   |8  Vcc --- nRF24L01  VCC, pin2 ------x----------x--|<|-- 5V
               NC      PB3  2|    |7  PB2 --- nRF24L01  SCK, pin5 --|<|---x-[22k]--|  LED
               NC      PB4  3|    |6  PB1 --- nRF24L01 MOSI, pin6  1n4148 |
nRF24L01 GND, pin1 -x- GND  4|    |5  PB0 --- nRF24L01 MISO, pin7         |
                    |        +----+                                       |
                    |-----------------------------------------------||----x-- nRF24L01 CSN, pin4
                                                                  10nF
```

[image]

kiredewit, 2022-02-26:
> 0nF wh

_Original RC combination was 1K/100nF. 22K/10nF combination worked better._ 
I use the 22k/10nF combination.  (ceramic capacitor 103))

_It is possible that you caused damage to the soil sensor (via current inrush spikes) when not properly using the regulator._
Uhmm I didn`t think about that. I will try directly with a new sensor.

_The radio may have been protected by the diode but that's not a guarantee because it is an LED (not a power diode). I've always been against nerdRalph having used an LED in his blog post. Power stability is very important (especially with the ICs_
thanks.

2bndy5, 2022-02-26:
@kiredewit use `>` to denote a blockquote 😉 
```md
> a blockquote ends in a blank line

this a reply to the quoted text
```

kiredewit, 2022-02-26:
> @kiredewit use `>` to denote a blockquote wink
> 
> ```lisp
> > a blockquote ends in a blank line
> 
> this a reply to the quoted text
> ```

:) thanks. That`s more readable.

With a new sensor and 3.3v power source, i got the same result :(.

2bndy5, 2022-02-28:
closing this since there doesn't seem to be any need for action  on our end.

Feel free to report your findings about why PB2 works and why PB3 doesn't.

closed 2022-02-28 by 2bndy5