nRF24/RF24 #833: [Question] best addresses and settings for the longest range

flexphperia, 2022-03-12:
As we know, choosing the right address for RF24 communication is important.
There is a nice tutorial on which addresses to avoid etc.: [http://maniacalbits.blogspot.com/2013/04/rf24-addressing-nrf24l01-radios-require.html](http://maniacalbits.blogspot.com/2013/04/rf24-addressing-nrf24l01-radios-require.html)

Is there anything else that can be done **programmatically** during setup to get the best results?

Does this address seem okay to you?
`const uint8_t receiverAddress[5] = {0b00111011, 0b10001010, 0b01101011, 0b00101011, 0b00110110};`

I typically initiates the radio like this?
```
radio.setCRCLength(RF24_CRC_8); 
radio.setAddressWidth(5); 
radio.setPALevel(RF24_PA_MAX,  1);
radio.setDataRate( RF24_250KBPS );
radio.setRetries(5, 15);
```
It would be a nice post for beginners if experienced people added their comments here.

2bndy5, 2022-03-12:
> Is there anything else that can be done programmatically during setup to get the best results?

BLE specs require "whitening" the payload to avoid repetitious bit patterns. You could look into that. Its kinda like long polynomial division (as used by CRC also), but with a small divisor.

>Does this address seem okay to you?

`0b10001010` <- this one seems problematic if directly adjacent to the transmission's preamble. But since it doesn't start the address (big endian IIRC) then you should be ok.

TMRh20, 2022-03-12:
1. Using a lower channel should generally provide better penetration through walls etc.
2. Per https://github.com/nRF24/RF24Network/blob/master/RF24Network.cpp#L1136 RF24Network uses the following for constructing addresses: `static uint8_t address_translation[] = {0xc3, 0x3c, 0x33, 0xce, 0x3e, 0xe3, 0xec};`  or ` 11000011, 00111100, 00110011, 11001110, 00111110, 11100011, 11101100`
3. In general, RF24Network does a lot of things that users would/should implement like using a header with a counter that increments every send. This keeps the devices from dropping packets when receiving sequential payloads with the same CRC.

Instead of getting into the weeds on all the details, for most communication scenarios, I usually just recommend using one of the higher layer libraries (RF24Network/Mesh/Ethernet) depending on needs. They handle most scenarios, and a lot of work has gone into optimization/functionality. If you want to know it all, it takes a while. The datasheets and documentation and code-base contain all the info we have.

closed 2022-07-30 by TMRh20