nRF24/RF24 #835: transmission delay when request data from i2c device.

chenxianming, 2022-04-04:
Please read [about common issues](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md) first. It addresses the most common problems that people have (whether they know it or not).

**Describe the bug**
A clear and concise description of what the bug is.

Please include:

`Wire.begin();

radio.setPALevel(RF24_PA_LOW);

radio.setPayloadSize(sizeof(payload));

radio.openWritingPipe(address[0]);
radio.enableDynamicPayloads();
radio.enableAckPayload();
radio.stopListening();`

on the loop function

`void loop(){
    Wire.requestFrom(8, 20);
    while (Wire.available()) {
      char c = Wire.read();
      Serial.print(c);
    }
    memcpy(payload.tranData, a.c_str(), 200);
    
   if( report ){}else{...}
}`

2. Expected behaviour

on the rx print logs

`ok

ok

ok
`

attention it was received empty line between results when i add  
`Wire.requestFrom(8, 20);`

to the loop.
and the same trouble with INPUT_PULLUP and digitalRead( Button_PIN ).
is that setting incorrect？

4. What device(s) are you using? Please specify make, model, and Operating System if applicable.
master ESP32, additional arduino uno for i2c
slave sec arduino uno.

**Additional context**
Add any other context about the problem here.

2bndy5, 2022-04-04:
I'm not really sure what the problem is here. It seems like you want to know why there are empty lines in the RX output when you added `Wire.requestFrom()`?

I don't see a problem with the radio's RX output - the blank lines _could_ be whitespace that are output when `while (Wire.available())` is true.

Unless there's a problem with the radio or RF24 library, I think this issue would be better described with all the code (not just some of it). You can use your native language if you find that easier (we can Google translate it on our end).

2bndy5, 2022-04-04:
> ```
memcpy(payload.tranData, a.c_str(), 200);
```

200 bytes is excessive. Are you aware the radio can only transmit 32 bytes at a time?

Keep in mind that any bye returned from I2C is likely an ASCII integer representing a single char (`(int)48 == (char)'0'`). So, printing those characters out may include end-of-line (LF or CRLF) and other unreadable characters.

chenxianming, 2022-04-04:
I don't see any others, it's running stable until member actions.

`ok
ok
ok`

2bndy5, 2022-04-04:
If there is a problem, then it doesn't seem to have anything to do with the RF24 library. It would help to see the entire code as well.

If you're looking for some general Arduino help, then you can try the [Arduino forums](https://forum.arduino.cc/).

closed 2022-05-13 by 2bndy5