nRF24/RF24 #836: Cannot build pyRF24 on bullseye 64 bit

ErikApption, 2022-04-06:
I'm on bullseye 64 bit with RPI4, I was able to compile the rf24 library but there seems to be an architecture mismatch when it comes to python3. 

```
librf24.so found at /usr/local/lib
running build
running build_ext
building 'RF24' extension
arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -L/usr/local/lib -I/usr/local/include -DRF24_NO_INTERRUPT -fPIC -I/usr/include/python3.9 -c pyRF24.cpp -o build/temp.linux-aarch64-3.9/pyRF24.o

In file included from /usr/include/boost/python/detail/wrap_python.hpp:57,
                 from /usr/include/boost/python/detail/prefix.hpp:13,
                 from /usr/include/boost/python/args.hpp:8,
                 from /usr/include/boost/python.hpp:11,
                 from pyRF24.cpp:1:
/usr/include/python3.9/pyconfig.h:13:12: fatal error: arm-linux-gnueabihf/python3.9/pyconfig.h: No such file or directory
   13 | #  include <arm-linux-gnueabihf/python3.9/pyconfig.h>
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
```

The compilation seems to be looking for arm instead of aarch64. Is there any way to change this?

2bndy5, 2022-04-06:
When you installed the RF24 library, how did you go about doing it? Did you use CMake or the older configure script?

---
The setup script does not specify a compiler. It simply tries to use the default (unless you used the configure script). The configure script was originally written only for armhf (32 bit variant aarch64). Thus it is recommended to build/install the C++ RF24 lib using CMake for any 64 bit OS.

---
On a side note, we've been reworking the python wrappers into its own repo. For more details on the pyRF24 package, see www.github.com/nRF24/pyRF24

ErikApption, 2022-04-06:
> When you installed the RF24 library, how did you go about doing it? Did you use CMake or the older configure script?

I just followed the instructions with the configure and it all just worked out after I installed the missing packages.

I did a bit of reading and aarch64 seems to be a bit of a mess with python but I will try the new cmake version.  I did suspect that the arm was something from defaults but I was not able to figure out how the weird setup and crosscompiler were figuring out the architecture. Will update this issue with the new library tomorrow.

2bndy5, 2022-04-06:
Its important to remove the previous build artifacts before re-running setup.py. Especially, the Makefile.inc generated by the configure script.

2bndy5, 2022-04-14:
ping

ErikApption, 2022-04-20:
@2bndy5 you are correct. Cleaning the whole repo, and rebuilding with cmake first DID work :-)

closed 2022-04-20 by ErikApption