nRF24/RF24 #840: RPi4(Receiver) + Pico(Sender) Transmission failed/timed out, but data received.[Question]

git-aniket, 2022-04-20:
Dear Library authors,
First of all thank you for a very nice library.

I have a task where I need to collect data wirelessly from sensors attached to two piPico's and send the data(56Bytes) at 100Hz to an RPi4. For this, I chose to go with NRF modules. I hope I made the correct choice given their high data rates?

For now, I have limited my testing to a single pico and a Pi4, and I am using a stripped-down version of the multiceiver codes provided in the library examples.

My problem is this:

Although I am able to receive the data in the PI4 from the pico, the serial console of the pico always reads "Transmission failed or timed out". Seldom does it succeed and print the success message.

I am wondering why is this happening. 

I have included the codes and my screenshots for the pico and PI4 terminals.

Also, given the data rates and the amount of data I need to collect, is the NRF a good choice for the application.?

Could you please help me with this?

[image]
[image]
[main_PICO.txt](https://github.com/nRF24/RF24/files/8523944/main_PICO.txt)
[main_RPI4.txt](https://github.com/nRF24/RF24/files/8523945/main_RPI4.txt)
Please read [about common issues](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md) first. It addresses the most common problems that people have (whether they know it or not).

git-aniket, 2022-04-20:
I shall try soldering a capacitor as suggested in the common issues and see the results.

2bndy5, 2022-04-20:
With an average of 520 us (from my personal experience testing the examples) per transmission (using auto-ack enabled), It looks like you'll be able to transmit 32 bytes @ 1923 HZ. This calculation is ("in a perfect world") assuming the sensor readings don't occupy any time, radios are always in range, no interference, etc...

As for the auto-ack not receiving, you're right to consider power supply stability (typically addressed by adding capacitance). Once you get that working, you might want to consider RF24Network because it makes transmitting payloads larger than 32 bytes very easy.

git-aniket, 2022-04-21:
Thank you very much for the reply and suggestions. I shall keep update this post once i implement them.

git-aniket, 2022-07-25:
Dear Authors,
This is an issue that is somewhat related to the previous one. 

Just as a recap, I am using a raspberry pico to transmit IMU data to a PI4. The objective is to transmit as many data samples as possible every second.
For this, I am using the RF24Network Library since I want to transmit using multiple nodes. 

My problem is although I am able to receive data into the PI4, I see that while sending the data, there is always a slight pause every second, although I am not using any delays in the code. I am trying to transfer a packet that contains a float and an int8_t. From the previous post, I understand that I should be able to get a fairly high sending rate. I have added the capacitor as has been suggested before.

Since I am reading an IMU using i2c, I initially thought that it was the sensor sampling rate that was limiting data throughput, however, even when I tried with a constant data value, the problem of pausing slightly every second, exists. 

I am attaching a bare bone code example and collected data for your reference.

Kindly help.

Regards
Aniket

[test.txt](https://github.com/nRF24/RF24/files/9180403/test.txt)
[testdata.txt](https://github.com/nRF24/RF24/files/9180404/testdata.txt)

Avamander, 2022-07-25:
Updating the network might take time, if you can't handle the delay you should be using simply RF24.

TMRh20, 2022-07-25:
Theoretically, a delay caused by RF24Network should only happen if there is a communication issue, and the network layer is waiting for an ack or something. I would go with Avamander's suggestion of trying the same/similar code with RF24 and see if the delay persists.

git-aniket, 2022-07-25:
Thank you very much for the prompt suggestions. I shall update this post in case I have further issues.

Kind regards
Aniket

git-aniket, 2022-07-25:
Dear library authors,

I tried using a barebone code from the examples of the pico and RPI4 with the rf24 library and removed the use of rf24network. However, the problem still seems to persist. 

There is no problem in receiving and transmitting the data. However, there is always a small pause every second or so. This is quite consistent as if there is a sleep command somewhere in the code(I didn't find one anywhere). My radios are kept pretty close and I have the capacitors on them. I am powering the pico using USB 5V and the PI4 using its power supply. 

The issue is certainly not the side of the receiver PI as even when PI is off, the weird pauses happen consistently accompanied by a transmission failure.

Could you kindly let me know what might be actually causing this problem.
Could any one of the implemented blocking functions be resposible for this?

In the attachment is a slightly modified version of the code from the examples that I used for my testing.

[main_PICO.txt](https://github.com/nRF24/RF24/files/9182274/main_PICO.txt)
?

Avamander, 2022-07-25:
Actually the issue might exactly be on the side of the receiver, you're not acknowledging the packets. (Some clones of the chip also have broken auto-ACK) If you don't care about if the packet is received or not, then use RF24 and send without ACK'ing.

git-aniket, 2022-07-25:
Thank you for the reply.

closed 2022-07-25 by git-aniket