nRF24/RF24 #841: How to I build a modified .cpp example for raspberry pi?

JeffRocchio, 2022-04-22:
I've got a question I'm embarrassed to ask - I have created a modified version of scanner.cpp for my own testing purposes. But...I don't actually know how to compile it. It needs to **compile and run on a raspberry pi**. It has been a long, long time since I had to compile my own hand-written C code and I can see that the make files have gotten a lot more complex. So much so that I can't really decode from the examples what is going on in them. Can anyone help me; or point me in the right direction?

One theory I have is to just copy my new "scanner_ch4C-highlighted.cpp" file to the ~/rf24libs/RF24/examples_linux folder on the Pi, then run ' make Makefile.examples" and see what happens. I'm guessing that the line | all: $(PROGRAMS) | is meant to compile/link all .cpp files in that directory. The question is, even if that is true, will rerunning it make a mash of the existing install - e.g., new directories or whatnot?

Another theory is that to build this one, simple, program it should be pretty straight-forward to 'manually' run the compile and linker. But I haven't done that since Windows NT! (Which just gave you my age...)

So - any help appreciated!

TMRh20, 2022-04-22:
Probably the easiest way is to open up the makefile and scroll down to where it lists all the example file names, then just add yours in there. Then as long as your file is in same dir as the examples, just run ‘make’ or make -B and it should build for you

JeffRocchio, 2022-04-22:
Right, OK I see the list in Makefile. However, after adding my new program running "make" or "make -B" returns:
`Makefile:15: *** recipe commences before first target.  Stop.`

JeffRocchio, 2022-04-22:
Ok, I think I have it. I found, and ran ./configure. then I had remove .cpp from the file name in the list in the makefile (I wasn't careful enough to notice the other file names didn't have a file extension on them). Now the make file executes. Although I still get an error saying that it can't fine SPIDEV/interrupt.h. Tomorrow I'll try to track that down.

2bndy5, 2022-04-22:
I have a patch for the "can't find interrupt.h" problem (reported in #834). It comes from not updating the Makefiles and configure script enough when switching to pigpio for interrupt support. See the changes in PR #839. 

Ideally, we've been moving to CMake instead of the configure script (which won't work well for 64 but OS).

---
I think that compiling your source code should be as simple as
```
c++ -lrf24 -pthread <modified_example_filename>
```
Assuming you have the RF24 lib installed. If you added any other libs to your example, then you may need to tell the compiler where to look for them (std libs should be the exception).

JeffRocchio, 2022-04-23:
Still unable to get this to work. Here is what I have tried today:

lpigpio was not installed on my pi; so I did apt-get that and it reported a successful install. 

After doing that I re-ran ./configure (my install was 3/25, after the 'pigpio' updates were made to the github - I did also compare my installed configure file to the one from PR #819 and it is an exact match). Running configure produces the following output --   
```
pi@raspberrypi:~/rf24libs/RF24` $ ./configure
./configure: line 57: file: command not found
[SECTION] Detecting arm compilation environment.
  [OK] arm-linux-gnueabihf-gcc detected.
  [OK] arm-linux-gnueabihf-g++ detected.
[SECTION] Detecting target machine.
[OK] machine detected: SoC=BCM2835, Type=RPi, CPU=armv6l.
[SECTION] Detecting DRIVER
  [OK] DRIVER detected:RPi.
[SECTION] Detecting OS.
  [INFO] OS detected:LINUX.
[SECTION] Preparing configuration.
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] `Finished`
```

Then tried compiling again. Which still produces --
```
pi@raspberrypi:~/rf24libs/RF24/examples_linux $ c++ -lrf24 -lpigpio -pthread scanner_ch4C-highlighted.cpp
In file included from /usr/local/include/RF24/RF24_config.h:62,
                 from /usr/local/include/RF24/RF24.h:18,
                 from scanner_ch4C-highlighted.cpp:33:
/usr/local/include/RF24/utility/includes.h:8:14: fatal error: SPIDEV/interrupt.h: No such file or directory
    8 |     #include "SPIDEV/interrupt.h"
      |              ^~~~~~~~~~~~~~~~~~~~

```
I think I somewhat/sorta understand the chain of include files (I'm *not* saying I understand what's in them all, just that I think I am following the chain of includes from rf24.h). And so when I look into /usr/local/include/RF24/utility/includes.h what I see is: #include "RPi/interrupt.h." Full content of includes.h below --
```
#ifndef __RF24_INCLUDES_H__
#define __RF24_INCLUDES_H__

#define RF24_RPi

#include "RPi/bcm2835.h"
#include "RPi/RF24_arch_config.h"
#ifndef RF24_NO_INTERRUPT
    #include "RPi/interrupt.h"
#endif

#endif
```

And as I look at the include files, configure script, and some of the Makefiles, I am getting the impression that a given configuration should be either going for SPIDEV or RPi - as in these are mutually exclusive choices.

So - why is the compile process looking for SPIDEV/interrupt.h"??

Probably I'm missing something elementary.

JeffRocchio, 2022-04-24:
I think I finally have it. Running the configure script did not copy either the updated _~/rf24libs/RF24/utility/includes.h_ file nor the _~/rf24libs/RF24/utility/RPi_ directory over to **/usr/local/include/RF24/utility** folder. That folder still had the old includes.h and SPIDEV subdirectory (from the original install I guess?). So I copied the RPi files, and the proper includes.h file over. Now all the errors I get are in my own code, lol. But the good news is that some of my distant memory of working with C is starting to come back.

2bndy5, 2022-04-24:
Just looking at the output of the configure script, I can tell you that you're not using the changes in #839 

You're problem would be more easily solved if you use CMake because this interrupt.h problem is already addressed for that implementation. PR #839 addresses the interrupt.h problem using the older configure/Makefile implementation.

So, let's start over.
1. Remove the installed RF24 lib (.so files and the installed header files)
   ```
   sudo rm /usr/local/lib/librf24.*
   sudo rm /usr/local/lib/librf24-bcm.so
   sudo rm -r /usr/local/include/RF24
   ```
2. Remove any build artifacts from using the configure script. This is only important if you plan on also installing the python wrapper.

   Execute this in the RF24 repo's root folder:
   ```
   sudo rm Makefile.inc *.o *.so *.so.*
   ```
3. Install the RF24 lib again using [these CMake instructions](https://github.com/nRF24/RF24/blob/master/docs/using_cmake.md#building-from-source-code)

   - Before compiling the examples (step 7), edit the _examples_linux/CMakeLists.txt_ file. Append your modified example's file name (no extension):
      ```diff
      # iterate over a list of examples by filename
      set(EXAMPLES_LIST
          gettingstarted
          acknowledgementPayloads
          manualAcknowledgements
          streamingData
          multiceiverDemo
          scanner
     +   modified_scanner_example
      )
      ```
   - Then continue with step 7 as instructed.

JeffRocchio, 2022-04-24:
Ok - yes, I will do that; that will be a cleaner way to get myself squared away. Hopefully tomorrow as I'm off to bed now.

JeffRocchio, 2022-04-24:
Apologies if I'm being a bother on this. But I'm still struggling. I have just followed your steps above - 1, 2 to remove the old install assets. I then proceeded to follow the CMake instructions @ https://github.com/nRF24/RF24/blob/master/docs/using_cmake.md#building-from-source-code.

I installed **librf24-SPIDEV_1.4.2-1_armhf.deb**. It didn't give any errors.

I could not tell from that instruction page if I was then supposed to continue with the '_Building from Source_' section as normally you either install from package OR build from source. But the package install didn't appear to change any of the files in the RF24 directories. And when I tried to run make on the examples it was all the same routine and errors.

So, I then started again, with your steps to remove any of the assets in lib and in include; and then proceeded to follow the 'Building from Source' instructions. But when I got to Step-5 here is what happened:
```
pi@raspberrypi:~/rf24libs/RF24 $ cmake .. -D RF24_DRIVER=SPIDEV
CMake Error: The source directory "/home/pi/rf24libs" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```
Which is true, as CMakeLists.txt is in /home/pi/rf24libs/RF24.

I don't know, I supposed the next obvious thing is to truly start from scratch - from a complete reinstall of the Pi OS.

2bndy5, 2022-04-24:
You're so close. The `..` tells cmake to look in the parent folder for a CMakeLists.txt file. The instructions should tell you to navigate to the build folder that you created in the RF24 folder (last command in step 4).
So,
```
pi@raspberrypi:~/rf24libs/RF24 $ cd build
pi@raspberrypi:~/rf24libs/RF24/build $ cmake .. -D RF24_DRIVER=SPIDEV
pi@raspberrypi:~/rf24libs/RF24/build $ make
pi@raspberrypi:~/rf24libs/RF24/build $ sudo make install
```
Then modify _examples_linux/CmakeLists.txt_ and add the modified example name as shown in my last comment. Now you should be able to build the example
```
pi@raspberrypi:~/rf24libs/RF24/build $ cd ../examples_linux
pi@raspberrypi:~/rf24libs/RF24/examples_linux $ mkdir build && cd build
pi@raspberrypi:~/rf24libs/RF24/examples_linux/build $ cmake .. -D RF24_DRIVER=SPIDEV
```
In the examples_linux/build folder, you should find the built binary executables.

---
> I installed librf24-SPIDEV_1.4.2-1_armhf.deb. It didn't give any errors.

This is fine, but it doesn't include the latest changes since v1.4.2 was released. Thus, I'm endorsing the "build from source" instructions.

>Apologies if I'm being a bother on this

No worries! We're a bit overdue for a release since pigpio support was added (after v1.4.2 was released). You're experience is valuable feedback nonetheless. Feedback like this is what drives me to alter the docs to be more user friendly, so don't hold back on suggested changes to the instructions or anything.

JeffRocchio, 2022-04-25:
**Thanks for sticking with me Brendan** - I do appreciate the help and support! And - **Success!**

fwiw, here is what I did: I removed all files and directories from the original build. Reversed out the 'package' install. Navigated around my /home and all the usr/include, lib, usr/local/include and lib directories and made sure nothing remained in any of those.

Then I rigorously followed, step-by-step, the '_Build from Source_' instructions. And now I have my modified program running fine.

One note on the instructions: In Step-7 where you have this: "_and edit the pin numbers as directed in the linux/RPi general documation_." <- the URL you have there returns a **404 error**.

And from my perspective it could be more clear that users will either install the 'package' OR 'Build from Source' and not both.

Thanks again for the help...and thanks for everyone building and maintaing this repo!

2bndy5, 2022-04-25:
>And from my perspective it could be more clear that users will either install the 'package' OR 'Build from Source' and not both.

Excellent feedback! I was also think that the install via package instructions should have a lesser priority since build by source is more preferred for Linux platform. I'll keep this issue open as reminder of this feedback and close it when I've adjusted the doc.

>the URL you have there returns a 404 error.

Its a relative URL. The doc you're reading on github is meant to be used as input to Doxygen which creates the static website we host as our docs @ https://nRF24.github.io/RF24/pages.html (or https://rf24.readthedocs.io/en/latest/pages.html for millennials 😄 . So, there isn't a way to avoid using relative URLs in the docs folder.

JeffRocchio, 2022-04-25:
> Its a relative URL.
Ah, understood. (HTML, esp via dynamic PHP/Apache, is something I am actually very familiar with.)

PR #844 (2022-06-12): various improvements to docs
This includes spelling fixes, updates for the latest release of sphinx-immaterial theme, and fixes #841 (as per feedback provided there).

I also reduced the Doxyfile to only options that we change from their default value. This should help avoid warnings in future Doxygen output.

Note to devs: We cannot use Doxygen v1.9.3 (& possibly v1.9.2) due to erroneous XML output that broke compatibility with Sphinx's breathe extension. Recommend v1.9.4+

In reference to #814 The sphinx docs use the prototype [idea from that thread](https://github.com/nRF24/RF24/issues/814# [... cut]
closed 2022-06-12 by TMRh20