nRF24/RF24 #845: Python 3 Wrapper not working on Rasp4

mslookup, 2022-06-14:
Hello everybody,
rasp4 with fresh Raspberry Pi OS lite.

Installed RF24 like stated in https://nrf24.github.io/RF24/md_docs_linux_install.html (automatic install with SPI option)
Tried to install Python 3 Wrapper using https://nrf24.github.io/RF24/md_docs_python_wrapper.html, install itself seems to work, but during last step when testing, i only get this error:

root@rasp4-ahoy:~/rf24libs/RF24/examples_linux# python3 getting_started.py
Traceback (most recent call last):
  File "/root/rf24libs/RF24/examples_linux/getting_started.py", line 9, in <module>
    from RF24 import RF24, RF24_PA_LOW
ModuleNotFoundError: No module named 'RF24'

Tried several other OS (buster, ...), everytime the same message. What am I doing wrong? Or is the Wrapper install broken?

Thank you :)

mslookup, 2022-06-14:
Hi and thanks for your reply,
i did the steps of the manual again and unfortunately using the root user seems to not be the problem :/

Here a full output code:
`pi@rasp4-ahoy:~/rf24libs/RF24/pyRF24 $ python3 setup.py build
librf24.so found at /usr/local/lib
running build
running build_ext
building 'RF24' extension
creating build
creating build/temp.linux-armv7l-cpython-39
arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -L/usr/local/lib -I/usr/local/include -DRF24_NO_INTERRUPT -fPIC -I/usr/include/python3.9 -c pyRF24.cpp -o build/temp.linux-armv7l-cpython-39/pyRF24.o
In file included from /usr/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:2,
                 from /usr/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/python/converter/shared_ptr_to_python.hpp:12,
                 from /usr/include/boost/python/converter/arg_to_python.hpp:15,
                 from /usr/include/boost/python/call.hpp:15,
                 from /usr/include/boost/python/object_core.hpp:14,
                 from /usr/include/boost/python/args.hpp:22,
                 from /usr/include/boost/python.hpp:11,
                 from pyRF24.cpp:1:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/detail/iterator.hpp:13:1: note: ‘#pragma message: This header is deprecated. Use <iterator> instead.’
   13 | BOOST_HEADER_DEPRECATED("<iterator>")
      | ^~~~~~~~~~~~~~~~~~~~~~~
creating build/lib.linux-armv7l-cpython-39
arm-linux-gnueabihf-g++ -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -L/usr/local/lib -I/usr/local/include -DRF24_NO_INTERRUPT build/temp.linux-armv7l-cpython-39/pyRF24.o -L/usr/lib -lrf24 -lboost_python3 -o build/lib.linux-armv7l-cpython-39/RF24.cpython-39-arm-linux-gnueabihf.so
pi@rasp4-ahoy:~/rf24libs/RF24/pyRF24 $ sudo python3 setup.py install
librf24.so found at /usr/local/lib
running install
/usr/local/lib/python3.9/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/local/lib/python3.9/dist-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
creating RF24.egg-info
writing RF24.egg-info/PKG-INFO
writing dependency_links to RF24.egg-info/dependency_links.txt
writing top-level names to RF24.egg-info/top_level.txt
writing manifest file 'RF24.egg-info/SOURCES.txt'
reading manifest file 'RF24.egg-info/SOURCES.txt'
adding license file '/home/pi/rf24libs/RF24/LICENSE'
writing manifest file 'RF24.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_ext
creating build/bdist.linux-armv7l
creating build/bdist.linux-armv7l/egg
copying build/lib.linux-armv7l-cpython-39/RF24.cpython-39-arm-linux-gnueabihf.so -> build/bdist.linux-armv7l/egg
creating stub loader for RF24.cpython-39-arm-linux-gnueabihf.so
byte-compiling build/bdist.linux-armv7l/egg/RF24.py to RF24.cpython-39.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
copying RF24.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying RF24.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying RF24.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying RF24.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
writing build/bdist.linux-armv7l/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
__pycache__.RF24.cpython-39: module references __file__
creating dist
creating 'dist/RF24-1.4.2-py3.9-linux-armv7l.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing RF24-1.4.2-py3.9-linux-armv7l.egg
removing '/usr/lib/python3.9/site-packages/RF24-1.4.2-py3.9-linux-armv7l.egg' (and everything under it)
creating /usr/lib/python3.9/site-packages/RF24-1.4.2-py3.9-linux-armv7l.egg
Extracting RF24-1.4.2-py3.9-linux-armv7l.egg to /usr/lib/python3.9/site-packages
RF24 1.4.2 is already the active version in easy-install.pth

Installed /usr/lib/python3.9/site-packages/RF24-1.4.2-py3.9-linux-armv7l.egg
Processing dependencies for RF24==1.4.2
Finished processing dependencies for RF24==1.4.2
pi@rasp4-ahoy:~/rf24libs/RF24/pyRF24 $ cd ../examples_linux
pi@rasp4-ahoy:~/rf24libs/RF24/examples_linux $ sudo python3 getting_started.py
Traceback (most recent call last):
  File "/home/pi/rf24libs/RF24/examples_linux/getting_started.py", line 9, in <module>
    from RF24 import RF24, RF24_PA_LOW
ModuleNotFoundError: No module named 'RF24'`

Thank you!

2bndy5, 2022-06-14:
> ```
> RF24 1.4.2 is already the active version in easy-install.pth
> ```

This means that you didn't re-install the wrapper.

---
Before we go any further, I'm going to ask you try our newer python pkg called [pyRF24](https://nrf24.github.io/pyRF24/). The install is much more automated and it doesn't need the C++ libs installed. I'm recommending this because troubleshooting the older python wrappers (what you're playing with now) is difficult when working remotely with... um, "non-experts" (no offense intended). These older python wrappers have become highly "nuanced" to get working correctly over the last 6 years.
- The pyRF24 API is slightly different from the C++ API, so that might be a hassle if you're going to use a python script that was meant for the older wrappers.

2bndy5, 2022-06-14:
FYI I just tried a fresh install on my RPi4, and it worked without a problem.

<details><summary>Here is what I executed</summary>

```shell
pi@rpi4:~/github/RF24/build $ export RF24_DRIVER=SPIDEV
pi@rpi4:~/github/RF24/build $ echo $RF24_DRIVER
SPIDEV
pi@rpi4:~/github/RF24/build $ cmake ..
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting build type to 'Release' as none was specified.
-- RF24 library version: 1.4.2
CMake Warning at cmake/Cache.cmake:27 (message):
  ccache is enabled but was not found.  Not using it
Call Stack (most recent call first):
  CMakeLists.txt:39 (include)

-- detected SoC: BCM2711
-- detected CPU type: armv7l
-- tool name being used: arm-linux-gnueabihf
-- Found pigpio library: /usr/lib/libpigpio.so
-- RF24_DRIVER (set from env var) = SPIDEV
-- Using driver: SPIDEV
-- linking to pigpio lib for interrupt functionality
-- ready to package: librf24-SPIDEV_1.4.2-1_armhf.deb
-- ready to package: librf24-SPIDEV-1.4.2-1.armhf.rpm
-- Using utility library: /usr/lib/libpigpio.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/github/RF24/build
pi@rpi4:~/github/RF24/build $ make
Scanning dependencies of target rf24
[ 16%] Building CXX object CMakeFiles/rf24.dir/RF24.cpp.o
[ 33%] Building CXX object CMakeFiles/rf24.dir/utility/SPIDEV/gpio.cpp.o
[ 50%] Building CXX object CMakeFiles/rf24.dir/utility/SPIDEV/spi.cpp.o
[ 66%] Building CXX object CMakeFiles/rf24.dir/utility/SPIDEV/compatibility.cpp.o
[ 83%] Building CXX object CMakeFiles/rf24.dir/utility/SPIDEV/interrupt.cpp.o
[100%] Linking CXX shared library librf24.so
[100%] Built target rf24
pi@rpi4:~/github/RF24/build $ sudo make install
[100%] Built target rf24
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/include/RF24/utility/SPIDEV/includes.h
-- Up-to-date: /usr/local/include/RF24/utility/SPIDEV/gpio.h
-- Installing: /usr/local/include/RF24/utility/SPIDEV/spi.h
-- Up-to-date: /usr/local/include/RF24/utility/SPIDEV/compatibility.h
-- Up-to-date: /usr/local/include/RF24/utility/SPIDEV/RF24_arch_config.h
-- Installing: /usr/local/include/RF24/utility/SPIDEV/interrupt.h
-- Installing: /usr/local/lib/librf24.so.1.4.2
-- Up-to-date: /usr/local/lib/librf24.so.1
-- Up-to-date: /usr/local/lib/librf24.so
-- Installing: /usr/local/include/RF24/RF24.h
-- Up-to-date: /usr/local/include/RF24/nRF24L01.h
-- Installing: /usr/local/include/RF24/printf.h
-- Up-to-date: /usr/local/include/RF24/RF24_config.h
-- Installing: /usr/local/include/RF24/utility/includes.h
-- Updating ldconfig
pi@rpi4:~/github/RF24/build $ cd ../pyRF24/
pi@rpi4:~/github/RF24/pyRF24 $ ls
build  crossunixccompiler.py  dist  __pycache__  pyRF24.cpp  RF24.egg-info  setup.py
pi@rpi4:~/github/RF24/pyRF24 $ rm -r ./build/ ./dist/ ./RF24.egg-info/ ./__pycache__/
pi@rpi4:~/github/RF24/pyRF24 $ ls
crossunixccompiler.py  pyRF24.cpp  setup.py
pi@rpi4:~/github/RF24/pyRF24 $ python -m pip install --upgrade pip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /usr/lib/python3/dist-packages (20.3.4)
Collecting pip
  Downloading https://www.piwheels.org/simple/pip/pip-22.1.2-py3-none-any.whl (2.1 MB)
     |████████████████████████████████| 2.1 MB 3.6 kB/s
Installing collected packages: pip
  WARNING: The scripts pip, pip3 and pip3.9 are installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed pip-22.1.2
pi@rpi4:~/github/RF24/pyRF24 $ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
pi@rpi4:~/github/RF24/pyRF24 $ export PATH=$PATH:/home/pi/.local/bin
pi@rpi4:~/github/RF24/pyRF24 $ python -m pip install .
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/github/RF24/pyRF24
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: RF24
  Building wheel for RF24 (setup.py) ... done
  Created wheel for RF24: filename=RF24-1.4.2-cp39-cp39-linux_armv7l.whl size=497147 sha256=9ea9e57338ddf36b68ec7f7507a3a4c045da56da22cbc8307bd931d91095075a
  Stored in directory: /tmp/pip-ephem-wheel-cache-r41tdpw3/wheels/78/48/3f/1b4e1fb4bdc1515eb6f8083035a3e707815f75452f4f37d821
Successfully built RF24
Installing collected packages: RF24
Successfully installed RF24-1.4.2
pi@rpi4:~/github/RF24/pyRF24 $ pip list
Package           Version
----------------- ---------
...
RF24              1.4.2
...
pi@rpi4:~/github/RF24/pyRF24 $ python
Python 3.9.2 (default, Mar 12 2021, 04:06:34)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from RF24 import RF24
>>>
```

</details>

So, to summarize:
1. Install the C++ RF24 using CMake
   ```
   sudo apt install cmake
   export RF24_DRIVER=SPIDEV
   cd RF24/build
   cmake ..
   make
   sudo make install
   ```
2. Install the python wrapper. But, make sure any previous build artifacts are removed and pip is updated. pip must be updated!!!
   ```
   cd ../pyRF24
   rm -r ./build/ ./dist/ ./RF24.egg-info/ ./__pycache__/
   python3 -m pip install --upgrade pip
   python3 -m pip install .
   ```

3. You should see `RF24` when you execute
   ```
   python3 -m pip list
   ```

mslookup, 2022-06-15:
Good morning!

Thank you so much for your answer - with your tips it's now working perfectly!
Some small notes i want to add for other guys that may find this thread looking for a solution:

This will lead to the files being downloaded locally on the rpi:
sudo apt install git
git clone https://github.com/nRF24/RF24.git

Of course you need to install python and pip first:
sudo apt-get install python3-dev libboost-python-dev python3-pip python3-rpi.gpio

Don't forget to link the libboost afterwards:
sudo ln -s $(ls /usr/lib/$(ls /usr/lib/gcc | tail -1)/libboost_python3*.so | tail -1) /usr/lib/$(ls /usr/lib/gcc | tail -1)/libboost_python3.so

BR and ty again!

closed 2022-06-15 by mslookup

2bndy5, 2022-06-15:
I'm now adjusting the pyRF24 API to be drop-in compatible with these older python wrappers (pending nRF24/pyRF24#9)