nRF24/RF24 #853: Is the python wrapper install steps supposed to work on RPi Zero too?

alanmiller, 2022-07-18:
I built and installed the rf24libs on my RPi 3B+ and my Pi Zero W and it partially worked as described
in https://nrf24.github.io/RF24/md_docs_linux_install.html

 My Pi3B can transmit and my PiZero can receive, BUT I cant get the Pi3B to receive and the PiZero to transmit.

Then I tried to build the python3 wrapper on both Pis  via steps in https://nrf24.github.io/RF24/md_docs_python_wrapper.html

On the RPi 3B+  I ran "python3 setup.py build" and "python3 setup.py install" in rf24libs/RF24/pyRF24 directory.
and the RF24 module is installed in:
    /usr/local/lib/python3.9/dist-packages/RF24-1.4.3-py3.9-linux-armv7l.egg

On the Pi Zero W I ran the same commands and it says it installed the wrapper but I get this error when running 
python3 getting_started.py

```
  root@pizero:~/NRF24L01/rf24libs/RF24/examples_linux# python3 getting_started.py
  Traceback (most recent call last):
    File "/root/NRF24L01/rf24libs/RF24/examples_linux/getting_started.py", line 9, in <module>
      from RF24 import RF24, RF24_PA_LOW
  ModuleNotFoundError: No module named 'RF24'
```

Both Pis are running Bullseye

```
  root@rpi3b:~/NRF24L01/rf24libs/RF24/examples_linux# lsb_release -d
  Description:	Raspbian GNU/Linux 11 (bullseye)

  root@pizero:~/NRF24L01/rf24libs/RF24/examples_linux# lsb_release -d
  Description:	Raspbian GNU/Linux 11 (bullseye)
```

When I installed the wrapper on Pi Zero W I got this

```
root@pizero:~/NRF24L01/rf24libs/RF24/pyRF24# python3 setup.py install
librf24.so found at /usr/local/lib
running install
/usr/local/lib/python3.9/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/local/lib/python3.9/dist-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
writing RF24.egg-info/PKG-INFO
writing dependency_links to RF24.egg-info/dependency_links.txt
writing top-level names to RF24.egg-info/top_level.txt
reading manifest file 'RF24.egg-info/SOURCES.txt'
adding license file '/root/NRF24L01/rf24libs/RF24/LICENSE'
writing manifest file 'RF24.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv6l/egg
running install_lib
running build_ext
creating build/bdist.linux-armv6l
creating build/bdist.linux-armv6l/egg
copying build/lib.linux-armv6l-cpython-39/RF24.cpython-39-arm-linux-gnueabihf.so -> build/bdist.linux-armv6l/egg
creating stub loader for RF24.cpython-39-arm-linux-gnueabihf.so
byte-compiling build/bdist.linux-armv6l/egg/RF24.py to RF24.cpython-39.pyc
creating build/bdist.linux-armv6l/egg/EGG-INFO
copying RF24.egg-info/PKG-INFO -> build/bdist.linux-armv6l/egg/EGG-INFO
copying RF24.egg-info/SOURCES.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying RF24.egg-info/dependency_links.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying RF24.egg-info/top_level.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
writing build/bdist.linux-armv6l/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
__pycache__.RF24.cpython-39: module references __file__
creating 'dist/RF24-1.4.3-py3.9-linux-armv6l.egg' and adding 'build/bdist.linux-armv6l/egg' to it
removing 'build/bdist.linux-armv6l/egg' (and everything under it)
Processing RF24-1.4.3-py3.9-linux-armv6l.egg
removing '/usr/lib/python3.9/site-packages/RF24-1.4.3-py3.9-linux-armv6l.egg' (and everything under it)
creating /usr/lib/python3.9/site-packages/RF24-1.4.3-py3.9-linux-armv6l.egg
Extracting RF24-1.4.3-py3.9-linux-armv6l.egg to /usr/lib/python3.9/site-packages
RF24 1.4.3 is already the active version in easy-install.pth

Installed /usr/lib/python3.9/site-packages/RF24-1.4.3-py3.9-linux-armv6l.egg
Processing dependencies for RF24==1.4.3
Finished processing dependencies for RF24==1.4.3
```

2bndy5, 2022-07-21:
@alanmiller Is this resolved now? Was using `sudo` the problem? I ask because you're the second person to report this problem about "RF24 module not found". If the advice about not using `sudo` during py wrapper install is helpful, then I'll augment the docs' build instructions.

alanmiller, 2022-07-21:
I switched to the packaged versions of RF24 and RF24Network 

```
pi@pizero ~/PYRF24 $ apt list --installed |grep ^rf24
rf24/now 1.4.4 armhf [installed,local]
rf24network/now 1.0.17 armhf [installed,local]
```
and tried to build the new Python wrapper but got errors

```
pi@pizero ~/PYRF24 $ sudo apt install python3-rpi.gpio
pi@pizero ~/PYRF24 $ sudo apt install python3-dev cmake
pi@pizero ~/PYRF24 $ git clone --recurse-submodules https://github.com/nRF24/pyRF24.git
pi@pizero ~/PYRF24 $ cd pyRF24
pi@pizero ~/PYRF24/pyRF24 $ sudo python -m pip install .
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/PYRF24/pyRF24
  Installing build dependencies ... \
...
  ERROR: Command errored out with exit status 1:
```

Full output attached.

[pyrf24-build.log](https://github.com/nRF24/RF24/files/9159973/pyrf24-build.log)

2bndy5, 2022-07-21:
> ```
> sudo python -m pip install .
> ```

PLEASE stop using `sudo` when running python or pip. You shouldn't need root access to install python modules/pkgs. Doing so alters the python runtime env and limits where pythonlooks for installed modules/pkgs.

2bndy5, 2022-07-21:
I can reproduce the build error using `pip install .`. However, the process is fine if I run 
```
pip install -r requirements.txt
python setup.py bdist_wheel
pip install dist/pyrf24*.whl
```
With no modifications to the pyrf24 lib.

alanmiller, 2022-07-21:
After I posted last, I tried building the wheel and it worked.

```
pi@pizero ~/PYRF24/pyRF24 $ sudo python3 -m pip install -r requirements.txt
pi@pizero ~/PYRF24/pyRF24 $ python setup.py bdist_wheel
....
pi@pizero ~/PYRF24/pyRF24 $ sudo python3 -m pip install  dist/pyrf24-0.1.dev138+gff659b5-cp39-cp39-linux_armv6l.whl
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./dist/pyrf24-0.1.dev138+gff659b5-cp39-cp39-linux_armv6l.whl
Installing collected packages: pyrf24
Successfully installed pyrf24-0.1.dev138+gff659b5
```

The example works now:

```
pi@pizero ~/PYRF24/pyRF24 $ cd examples
pi@pizero ~/PYRF24/pyRF24/examples $ python3   getting_started.py  -r 1
Which radio is this? Enter '0' or '1'. Defaults to '0' 1
================ SPI Configuration ================
CSN Pin		= /dev/spidev0.1
CE Pin		= Custom GPIO13
SPI Speedz	= 10 Mhz
================ NRF Configuration ================
STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
TX_ADDR		= 0x65646f4e32
RX_PW_P0-6	= 0x04 0x04 0x04 0x04 0x04 0x04
EN_AA		= 0x3f
EN_RXADDR	= 0x03
RF_CH		= 0x4c
RF_SETUP	= 0x03
CONFIG		= 0x0e
DYNPD/FEATURE	= 0x00 0x00
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_LOW
ARC		= 15
getting_started.py
Transmission successful! Time to Transmit: 826.995 us. Sent: 0.0
Transmission successful! Time to Transmit: 835.994 us. Sent: 0.01
Transmission successful! Time to Transmit: 949.993 us. Sent: 0.02
```

closed 2022-07-21 by alanmiller

2bndy5, 2022-07-21:
I'm in the process of updating the pyrf24 package. I hope to be able to build and distribute wheels to pypi in the future (see nRF24/pyRF24#6) which would bypass the need to manually build the wheel (and make install much quicker). For now, I'll just change the README to instruct using the `bdist_wheel` command... Turns out the `setup.py install` command is now deprecated.