nRF24/RF24 #856: pico-sdk port?

DatanoiseTV, 2022-07-21:
It would be nice to see a non-arduino port for the Pico SDK. I am currently working on small LiPo powered board for NRF24 powered sensors where I am planing to utilise dormant sleep of the RP2040 and wake up via the nRF interrupt pin.

2bndy5, 2022-07-21:
The PikoSDK is already supported. Instruction for how to include RF24/RF245Network/RF24Mesh in your RP2040 projrct are in the docs: https://nrf24.github.io/RF24/md_docs_pico_sdk.html

DatanoiseTV, 2022-07-21:
@2bndy5 I've just tried with the following pinout, but i am getting "radio hardware is not responding":
GP1 -> CE
GP2 -> SCK
GP3 -> MOSI
GP4 -> MISO
GP5 -> CSN
Powered by 3.3V from a XC6206P332MR LDO.

2bndy5, 2022-07-21:
I need to see the steps you took to get there. What exact RP2040 board are you using? Are you still putting a capacitor in parallel to the radio's power pins? I can't help much based on the limited info you're giving me.

2bndy5, 2022-07-21:
These are the pins used for the pico board, if you're testing this lib's pico examples.
https://github.com/nRF24/RF24/blob/a04f32b1fb0b2aa3070179d6abcde0aacaf60ebb/examples_pico/defaultPins.h#L25-L30

DatanoiseTV, 2022-07-21:
That did the job. Thanks.

2bndy5, 2022-07-21:
Closing this as it is getting off topic. You can open another issue to ask any other questions. Happy coding.

closed 2022-07-21 by 2bndy5