nRF24/RF24 #860: nrf24l01+ packet loss in 250kbps and 1mbps

SauravNiraula, 2022-08-04:
my nrf24l01+ seems to work fine with dataRate set to 2MBPS. I get like 300 packets per second(PPS) with 3byte payload size. Setting dataRate to 1MBPS or 250KBPS i get like 0 to 10 packets per second. 

I am using amplified antenna version and taking out antenna from both tx and rx side doesnot seem to affect PPS when dataRate is 2MBPS but increase PPS when dataRate is set to 1MBPS ,250KBPS.

Can it be due to buck converter that i am using to supply power ?
I have also soldered 10uf capacitor soldered between vcc and gnd and o/p voltage is set to 3.28v

2bndy5, 2022-08-04:
> Can it be due to buck converter that i am using to supply power ?

Most likely. PA/LNA modules require an upwards of ~120mA (varies on the manufacturer). Removing the antennae doesn't affect the power requirement because they use another IC (the chip that consumes the extra power) to implement the extra PA/LNA features.

It has also been reported that some cheap buck converters produce considerable ElelctroMagnetic noise. This can be solved by adding shielding to either or both buck converter/radio module.

Most of this is addressed in the [COMMON_ISSUES.md](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#my-palna-module-fails-to-transmit)

SauravNiraula, 2022-08-04:
Will try changing the power source.

As addressed in COMMON_ISSUES.md  "Moreover, this can sometimes be seen in odd ways such as the radio module working better when touched. This again is likely a power stability issue".

radio module was working better when touched.

2bndy5, 2022-08-04:
> radio module was working better when touched.

Then the manufacturer skipped adding a capacitor to the module. This is pretty common; most of my non-PA/LNA modules are missing this capacitor.

You may also want to try adding more capacitance to the power supply of the radio. Smaller capacitors will smooth out small ripples, but larger capacitors will smooth out larger ripples. Basically, use a small and large capacitors in parallel with the radio's power supply lines.

SauravNiraula, 2022-08-05:
Those buck converters were the problem. Changing power source completely solved the issue.
Thanks for your help.

closed 2022-08-05 by SauravNiraula