nRF24/RF24 #861: STM32 crashing when defining other SPIClass

V8Power, 2022-08-11:
Please read [about common issues](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md) first. It addresses the most common problems that people have (whether they know it or not).

**Describe the bug**
A clear and concise description of what the bug is.
I've got an STM32F103C8 that keeps crashing when I define SPI2 as the port it is connected to. I've tried defining SPI1 on a different board in the same way and I get the same issue. I can not hook the module to a different port as it is on a custom board.
Both transmitting and receiving functions seem to cause problems.
The module is detected on reset.

`

#include <Arduino.h>
#include <SPI.h>
#include "printf.h"
#include "RF24.h"

// instantiate an object for the nRF24L01 transceiver
//RF24 radio(7, 8);  // using pin 7 for the CE pin, and pin 8 for the CSN pin
RF24 radio(PB5, PB12); // using pin 7 for the CE pin, and pin 8 for the CSN pin
// Let these addresses be used for the pair
uint8_t address[][6] = { "1Node", "2Node" };
// It is very helpful to think of an address as a path instead of as
// an identifying device destination

// to use different addresses on a pair of radios, we need a variable to
// uniquely identify which address this radio will use to transmit
bool radioNumber = 0;  // 0 uses address[0] to transmit, 1 uses address[1] to transmit

// Used to control whether this node is sending or receiving
bool role = false;  // true = TX role, false = RX role

// For this example, we'll be using a payload containing
// a single float number that will be incremented
// on every successful transmission
float payload = 0.0;

void setup() {
SPIClass mySPI(PB15,PB14,PB13);
mySPI.begin();
  Serial.begin(115200);
  while (!Serial) {
    // some boards need to wait to ensure access to serial over USB
  }

  // initialize the transceiver on the SPI bus
  if (!radio.begin(&mySPI)) {
    Serial.println(F("radio hardware is not responding!!"));
    //while (1) {}  // hold in infinite loop
  }

  // print example's introductory prompt
  Serial.println(F("RF24/examples/GettingStarted"));

  // To set the radioNumber via the Serial monitor on startup
  Serial.println(F("Which radio is this? Enter '0' or '1'. Defaults to '0'"));
  while (!Serial.available()) {
    // wait for user input
  }
  char input = Serial.parseInt();
  radioNumber = input == 1;
  Serial.print(F("radioNumber = "));
  Serial.println((int)radioNumber);

  // role variable is hardcoded to RX behavior, inform the user of this
  Serial.println(F("*** PRESS 'T' to begin transmitting to the other node"));

  // Set the PA Level low to try preventing power supply related problems
  // because these examples are likely run with nodes in close proximity to
  // each other.
  radio.setPALevel(RF24_PA_LOW);  // RF24_PA_MAX is default.

  // save on transmission time by setting the radio to only transmit the
  // number of bytes we need to transmit a float
  radio.setPayloadSize(sizeof(payload));  // float datatype occupies 4 bytes

  // set the TX address of the RX node into the TX pipe
  radio.openWritingPipe(address[radioNumber]);  // always uses pipe 0

  // set the RX address of the TX node into a RX pipe
  radio.openReadingPipe(1, address[!radioNumber]);  // using pipe 1
Serial.println("alive"); 
  // additional setup specific to the node's role
  if (role) {
    radio.stopListening();  // put radio in TX mode  //crash point
    Serial.println("transmitting"); 
  } else {
    Serial.println("getting ready to listening");
    radio.startListening(); 
    Serial.println("listening");  // put radio in RX mode
  }

  // For debugging info
  // printf_begin();             // needed only once for printing details
  // radio.printDetails();       // (smaller) function that prints raw register values
  // radio.printPrettyDetails(); // (larger) function that prints human readable data

}  // setup

void loop() {
Serial.println("alive"); 
  if (role) {
    // This device is a TX node
    Serial.println("pre_alive"); 
    unsigned long start_timer = micros();                // start the timer
    bool report = radio.write(&payload, sizeof(float));  // transmit & save the report    //crash point
    unsigned long end_timer = micros();  
    Serial.println("alive");                // end the timer

    if (report) {
      Serial.print(F("Transmission successful! "));  // payload was delivered
      Serial.print(F("Time to transmit = "));
      Serial.print(end_timer - start_timer);  // print the timer result
      Serial.print(F(" us. Sent: "));
      Serial.println(payload);  // print payload sent
      payload += 0.01;          // increment float payload
    } else {
      Serial.println(F("Transmission failed or timed out"));  // payload was not delivered
    }

    // to make this example readable in the serial monitor
    delay(1000);  // slow transmissions down by 1 second

  } else {
    // This device is a RX node
Serial.println("starting to listening");
    uint8_t pipe;
    if (radio.available(&pipe)) {       //crashes point
      Serial.println("listening at a pipe");            // is there a payload? get the pipe number that recieved it
      uint8_t bytes = radio.getPayloadSize();  // get the size of the payload
      radio.read(&payload, bytes);             // fetch payload from FIFO
      Serial.print(F("Received "));
      Serial.print(bytes);  // print the size of the payload
      Serial.print(F(" bytes on pipe "));
      Serial.print(pipe);  // print the pipe number
      Serial.print(F(": "));
      Serial.println(payload);  // print the payload's value
    }
    Serial.println("listening... didnt have data");
  }  // role
Serial.println("alive"); 
  if (Serial.available()) {
    // change the role via the serial monitor

    char c = toupper(Serial.read());
    Serial.println("got serial data"); 
    if (c == 'T' && !role) {
      // Become the TX node

      role = true;
      Serial.println(F("*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK"));
      radio.stopListening();

    } else if (c == 'R' && role) {
      // Become the RX node

      role = false;
      Serial.println(F("*** CHANGING TO RECEIVE ROLE -- PRESS 'T' TO SWITCH BACK"));
      radio.startListening();
    }
  }

}  // loop
`

2bndy5, 2022-08-11:
I'm assuming you're using a [the STM32duino core](https://github.com/stm32duino/Arduino_Core_STM32/tree/main/libraries/SPI) despite the fact you mentioned using a custom board.

Is there any erroneous output you can post (like from compiler or seg fault indication in Serial monitor)?

V8Power, 2022-08-11:
Yes, your absolutely correct, I'm using the official STM32 core. The board is defined as generic STM32F103C8.
I'm not getting any errors or warning from the compiler. I'm using VS Code as my editor. 
Could it be a problem with the CS pin? I tried setting the CS pin in my SPI definition but that didn't change anything

2bndy5, 2022-08-11:
I think the PB5 can be used as the SPI1_MOSI, according to the [STM32F103c8 datasheet pinout](https://www.st.com/resource/en/datasheet/stm32f103c8.pdf), but only if the core is using a remap of the ppin's alternate function; I doubt that this is the case, but just wanted to point that out.

Looking at the [stm32duino SPI library's readme](https://github.com/stm32duino/Arduino_Core_STM32/tree/main/libraries/SPI), the `transfer()` might need to be altered for the core's API. Specifically, the CS pin may need to be the first param.

When you pass an SPI obj to `RF24::begin()`, you end up using the calls to `_spi->function()` in RF24 src. For instance, this is the code that reads a buffer from the radio's registers:
https://github.com/nRF24/RF24/blob/89358b2df7de37d64f2bba180d61b6d200c54070/RF24.cpp#L177-L183

2bndy5, 2022-08-11:
I also have to a verify what version of the RF24lib you are using. There was a bug in v1.4.3-1.4.4 that affected arduino platforms about `sprintf()` support. Best to use v1.4.5

V8Power, 2022-08-11:
I was using v1.4.3. I switched to v1.4.5 but it didn't make any difference. I checked in in MXCube and it specifies the pins just as they are hooked up.
I set up my proper debug environment and found that it gets stuck in an HAL error handler.
```

#include "stm32_def.h"
#include "core_debug.h"

#ifdef __cplusplus
extern "C" {
#endif

/**
  * @brief  This function is executed in case of error occurrence.
  * @param  None
  * @retval None
  */
WEAK void _Error_Handler(const char *msg, int val)
{
  /* User can add his own implementation to report the HAL error return state */
  core_debug("Error: %s (%i)\n", msg, val);
  while (1) {
  }
}

#ifdef __cplusplus
}
#endif
```

In a different instance it went into an endless loop with the SPI active flag

```

_STATIC_INLINE uint32_t LL_SPI_IsActiveFlag_TXE(SPI_TypeDef *SPIx)
{
  return ((READ_BIT(SPIx->SR, SPI_SR_TXE) == (SPI_SR_TXE)) ? 1UL : 0UL);///exception occured
}
```
```

  while (size--) {
#if defined(STM32H7xx) || defined(STM32MP1xx)
    while (!LL_SPI_IsActiveFlag_TXP(_SPI));
#else
    while (!LL_SPI_IsActiveFlag_TXE(_SPI));    //stuckhere
#endif
    LL_SPI_TransmitData8(_SPI, *tx_buffer++);

```

So I guess the problem is the SPI active flag staying active. While debugging I managed to go through about 6-8 void loop() cycles before it got stuck

2bndy5, 2022-08-11:
This may be a simple name conflict...
https://github.com/nRF24/RF24/blob/89358b2df7de37d64f2bba180d61b6d200c54070/RF24_config.h#L109-L113
https://github.com/nRF24/RF24/blob/89358b2df7de37d64f2bba180d61b6d200c54070/RF24.h#L127-L129
vs a [similar pointer declared in the STM32duino core](https://github.com/stm32duino/Arduino_Core_STM32/blob/375f6fbafbebf124d5260ac27b70c6c8feb3a5d3/libraries/SPI/src/utility/spi_com.c#L482):
```diff
spi_status_e spi_transfer(spi_t *obj, uint8_t *tx_buffer, uint8_t *rx_buffer,
                          uint16_t len, uint32_t Timeout, bool skipReceive)
{
  spi_status_e ret = SPI_OK;
  uint32_t tickstart, size = len;
-  SPI_TypeDef *_SPI = obj->handle.Instance;

  if ((obj == NULL) || (len == 0) || (Timeout == 0U)) {
    return Timeout > 0U ? SPI_ERROR : SPI_TIMEOUT;
  }
  tickstart = HAL_GetTick();

#if defined(SPI_CR2_TSIZE)
  /* Start transfer */
  LL_SPI_SetTransferSize(_SPI, size);
  LL_SPI_Enable(_SPI);
  LL_SPI_StartMasterTransfer(_SPI);
#endif

  while (size--) {
#if defined(SPI_SR_TXP)
    while (!LL_SPI_IsActiveFlag_TXP(_SPI));
#else
-    while (!LL_SPI_IsActiveFlag_TXE(_SPI)); // indicated crash point
#endif
    LL_SPI_TransmitData8(_SPI, *tx_buffer++);
```
This declaration happens just before the crash point you identified. I suppose we could namespace the RF24 `_SPI` or just rename it to `_RF24_SPI`. I'd opt for the later for quick search-&-replace testing. Namespacing the SPI objects/types might be a bit more complicated because of all the specific hardware files in RF24/utility folder.

V8Power, 2022-08-11:
I tried renaming _SPI to RF24_SPI, didn't change a thing. I just redefined the SPI pins at the beginning of my code. Seems to work fine now

2bndy5, 2022-08-11:
If you could post the working code, I'll add it as an example snippet to the arduino.md doc. We don't have a stm32 example because I couldn't test it.

V8Power, 2022-08-11:
I'll do some more testing and then post it here

2bndy5, 2022-09-07:
@V8Power Any updates on this?

closed 2023-01-17 by 2bndy5