nRF24/RF24 #867: updating the docs without publishing a release

2bndy5, 2022-09-01:
I just had a thought that would allow us to publish docs without having to publish a release... It would involve a manually triggered the docs CI from the master branch (see [`workflow_dispatch` event](https://docs.github.com/en/actions/using-workflows/events-that-trigger-workflows#workflow_dispatch)).

Of course this should be used lightly because the we don't want to publish docs that don't correspond to latest release.

2bndy5, 2022-09-01:
RTD docs already do this:
- "latest" = master's head
- "stable" = latest tagged commit

[image]

Although, I'm not currently relying on RTD builds as a proper reference because the "Related Pages" have misplaced admonitions at the bottom (a [bug in breathe](https://github.com/michaeljones/breathe/issues/835)).

TMRh20, 2022-09-01:
I think this would be a pretty good idea, there are probably lots of times the docs can be updated without a release being made, plus RPi/Linux users would be compiling from source in many cases. Thinking we should do this for all the RF24 repos.

2bndy5, 2022-09-01:
> Thinking we should do this for all the RF24 repos.

Oops, I forgot to mention that was part of my intention.

2bndy5, 2022-09-02:
@TMRh20 @Avamander FYI
[image]
1. select the docs CI workflow
2. open the dropdown menu
3. manually run the workflow on the master branch to update the docs. 

Manually running this workflow on any other branch will not publish the docs (as usual).

closed 2022-09-02 by 2bndy5

2bndy5, 2022-09-02:
This wasn't needed for RF24Audio. If I missed a repo, let me know. The CirPy & pyRF24 repos already rely on RTD, so we don't need this for them either.

TMRh20, 2022-09-02:
Nice work!