nRF24/RF24 #874: v1.4.6 not showing up in Arduino Library Manager

TMRh20, 2022-09-24:
The latest RF24 release doesn't seem to be showing up in the lib manager. Wondering if its just me or ???

2bndy5, 2022-09-24:
strange, It shows up for me in IDE v2.0.0
[image]

2bndy5, 2022-09-24:
I did something a little different for that release. I created a draft release, then push the version bump, then published the drafted release. I noticed the CI didn't auto-upload release assets for this release (I added them manually), and I assumed it was because the release event was slightly different from usual. The tag does point to the version bump commit, so it looks right to me...

TMRh20, 2022-09-24:
Looks like it was just me. Using IDE 2.0. I still have 1.8.x installed so tried that, it worked, went back to 2.0.0 and it worked.

closed 2022-09-24 by TMRh20

2bndy5, 2022-09-24:
v2.0.0 IDE just hit stable designation (last week I think), but I've been using the beta for a while (kinda addicted to the electron-based editor because of VSCode). I also started using a separate serial monitor called [Simply Serial](https://github.com/fasteddy516/SimplySerial) because it respects the return feed char (`\r` is useful for scanner output), but user input isn't echoed to the terminal for that app.