nRF24/RF24 #876: [Question]

janezlapajne, 2022-10-23:
Have anyone tried to use this library on Jatson Nano? Is it possible to use python calls to utilize wireless communication?

2bndy5, 2022-10-23:
It might be possible using the newer [pyrf24 pkg](https://github.com/nRF24/pyRF24). But, there's an old issue about using the Jetson TX2 #525 which we can't resolve without serious funding. I'm not sure if that will also affect the Jetson Nano.

janezlapajne, 2022-10-23:
Hello, thank you for your answer. I will try to build it in a few days on the jetson nano - I will inform you whether it will be successful. Can I kindly ask you about which examples should I use to test wireless communication between two devices (one receiver and one transmitter) on which the transceiver modules are connected? Also, in your readme, the hyperlink to Examples does not work ;)

2bndy5, 2022-10-23:
> I will try to build it in a few days on the jetson nano

I'm not sure you need to build it yourself. You should be able to `pip install pyrf24`, but if that gives errors, then you'll need to build it from src (note that the pyrf24 pkg uses git submodules).

> which examples should I use to test wireless communication between two devices

The [getting_started example](https://pyrf24.readthedocs.io/en/latest/examples.html#getting-started) is always a good idea for basic hardware tests.

> Also, in your readme, the hyperlink to Examples does not work ;)

That's probably because I used a relative URL. Try reading the [pyrf24 docs](https://pyrf24.readthedocs.io/en/latest/) instead.

janezlapajne, 2022-10-25:
As suspected the installation unfortunately doesn't work. First I tried to install it from pyPI, but it throws me an error: 
`
ERROR: Could not find a version that satisfies the requirement pyrf24 (from versions: none)
ERROR: No matching distribution found for pyrf24
`

Then I followed the procedure on how to install it locally, but it also throws me an error:
`
ERROR: No matching distribution found for setuptools>=61
  Installing build dependencies ... error
WARNING: Discarding file:///home/jetson/Documents/pyRF24. Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.6/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-ke5j4hlp/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=61' wheel 'setuptools_scm[toml]>=6.2' Check the logs for full command output.
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.6/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-ke5j4hlp/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=61' wheel 'setuptools_scm[toml]>=6.2' Check the logs for full command output.
`

Obviously the version of setuptools doesn't meet the requirements. I tried to update pip and setuptools using:
`
python3 -m pip install --upgrade pip
`
`
pip3 install -U setuptools
`
, but this doesn't help either, since I use python 3.6 and I already have the latest version of setuptools installed. I cannot update python to the newest version, because I use some dependencies that only run with python 3.6.

Any Ideas for an easy fix? 🧙‍♂️

2bndy5, 2022-10-25:
> I cannot update python to the newest version, because I use some dependencies that only run with python 3.6.

Well, you should take that up with your dependencies. Python 3.6 is no longer maintained and deemed deprecated. The pyRF24 package requires v3.7 or newer because the examples use `time.monotonic_ns()` to calculate elapsed milliseconds. If that were the only thing blocking you, then I'd tell you how to bypass that, but it also sounds like `setup_tools` v61+ doesn't support python 3.6. You need v61+ to use the pyproject.toml, otherwise you'd have to rewrite setup.py with all the relevant info in pyproject.toml (excluding mypy & pylint settings).

---
As a last resort, you could try installing the RF24 C++ libs and their individual python wrappers, but we are less interested in maintaining the individual wrappers (now that we have the pyRF24 package deployable).

If you go this route, please follow the CMake build/install instructions and then install the python wrappers. The older `./configure` instructions might not work with the Jetson Nano because they assumed using the RPi with an armhf compiler.

janezlapajne, 2022-10-26:
Yes, I agree with everything you wrote in the first paragraph. Believe me I would definitely upgrate the python version, but the problem is the current version of Jetpack  (v4.6), which only supports python 3.6. See the introduction of this [article](https://qengineering.eu/install-pytorch-on-jetson-nano.html#:~:text=PyTorch%201.12%2C%201.11.,3.7%2C%20found%20in%20JetPack%205.0.) why -  Copied: _Since JetPack 4.6 has Python 3.6, you cannot install PyTorch 1.11.0 on a Jetson Nano.
It looks like Nvidia has no plans to release the new JetPack 5.0 for the Jetson Nano for now. It's only available for the Xavier series._ To be sincere, I will probably find another quick solution for now i.e. right now we are using raspberry pi attached to jetson nano for communication and it works, but it adds new complexity and cost to the whole system. So for now I will thank you for all your responses and maybe I will try in the future when they provide new JetPack version :) cya!

closed 2022-10-26 by janezlapajne

2bndy5, 2022-10-26:
Yet another reason why I didn't like working with the TX2: Nvidia's dev cycle is very slow. And the Jetpack wasn't very well documented IMHO, but maybe that's improved over the last 2 years.

The individual wrappers were written to support all versions of python (even the dead v2.7) with boost.python, so if you do revisit this, I would suggest using the individual wrappers (located along side the C++ src in RF24* libs). They are current with the C++ sources (which rarely changes now).

2bndy5, 2023-04-21:
Looks like Nvidia is no longer supporting the Nano and TX2/TX1 series. See [this Nvidia dev comment](https://forums.developer.nvidia.com/t/jetpack-5-0-2/223564/22). The end-of-life for Jetpack 4.6.x isn't concrete (indefinite maintenance period), but it may be obsoleted when Ubuntu18.04 reaches EoL in May 2023.

IMO, this is sad news and another reason why people shouldn't invest in Nvidia Jetson modules for long-term projects. 😞

janezlapajne, 2023-04-21:
We shall see what happens, thanks for the info!