nRF24/RF24 #879: [Question]

WellingtonMaricato, 2022-11-29:
Hi, I'm have a project, and than one raspberry send a data to Arduino, in one variable string ex [34, 42, 67, 12]. 
I'm read all documentation and all exemplos, but don't found how solver this in radio.write(variable) (raspberry) and radio.read(variable) in Arduino. 
I'm use struct in python, but don't have a success.
Tks.

2bndy5, 2022-11-29:
> but don't found how solver this in radio.write(variable) (raspberry) and radio.read(variable) in Arduino.

I don't understand. `radio.read()` is clearly demonstrated in the Arduino examples (see [GettingStarted.ino](https://github.com/nRF24/RF24/blob/master/examples/GettingStarted/GettingStarted.ino)). `radio.write()` is demonstrated in the python examples (see [getting_started.py](https://github.com/nRF24/RF24/blob/master/examples_linux/getting_started.py)).

> I'm use struct in python, but don't have a success.

"don't have success" could mean many things. Can you post the error text that you are seeing?

On a side note, how did you install the python wrappers? We have a [new python package called pyrf24](https://pyrf24.readthedocs.io/en/latest) that is [much easier to install](https://pyrf24.readthedocs.io/en/latest/#installing-from-pypi). There are detailed [examples for how to use pyrf24](https://pyrf24.readthedocs.io/en/latest/examples.html) on the RPi.

WellingtonMaricato, 2022-11-29:
My problem is in to send a array variable. I'm don't know used "struct" in phyton for send a array to Arduino. Do you have a simple exemplo for this?

2bndy5, 2022-11-29:
It sounds like you need to learn more about python. I am not going to teach you python, but I can point you in the right direction.

The [`struct` module is a standard python library](https://docs.python.org/3/library/struct.html). You can find various tutorials around the web for this.

The key important factor is that the `RF24.write()` function only accepts [`bytes`](https://docs.python.org/3/library/stdtypes.html#bytes) or [`bytearray`](https://docs.python.org/3/library/stdtypes.html#bytearray) objects, and the examples use [`struct.pack()`](https://docs.python.org/3/library/struct.html#struct.pack) to easily turn data into a bytes object. In python, the `RF24.read()` returns a [`bytearray`](https://docs.python.org/3/library/stdtypes.html#bytearray), and the examples use [`struct.unpack()`](https://docs.python.org/3/library/struct.html#struct.unpack) to easily covert that into whatever you need it to be.

### Some advanced python tips
If all integers in the array will always be `0 <= x <= 255` (the size of a single unsigned byte), then you can send the array using python's `bytes` constructor.
```py
my_list = [34, 42, 67, 12]

radio.write(bytes(my_list))
```
This can be done in reverse using the `list` constructor. Again this assumes that all integers in the list are `0 <= x <= 255`.
```py
payload = radio.read(4)

my_list = list(payload)
```

closed 2022-12-01 by 2bndy5