nRF24/RF24 #881: How to close communication after receiving data ?

dharmikP17, 2022-12-22:
I am using nrf24 with esp32 in light sleep mode. Esp32 wakes up on IRQ and than I have to use another memory IC that works on SPI too. 

Problem is after radio.begin(), I am not able to initiate communication with the IC. If I dont use nrf24 than it works fine so there is no problem with the hardware.

I am using winbond 16MB flash IC and SPIMemory.h library. So my guess is I have to close the communication from nrf24 in order to make it work. I dont want to use both of them at same time anyway. I could not find any function in the library related to this. So is there any way to do so ??

Another little off topic quetion. I was not able to wakeup ESP32 from deep sleep using IRQ. Is ther any other configuration required for deep sleep mode ??

Let me know if any other details are required. Thank you.

2bndy5, 2022-12-22:
The radio's CSN pin is automatically made inactive upon the end of an SPI transaction, thus there's no need to manually close the SPI connection.

If you're going to have more than 1 slave device on the same SPI bus, then each slave device must use a CS or CSN pin to know which slave the master is talking to. If you could provide link to the libs using the same SPI bus, we might be able to help more.

---
Deep sleep mode is specific to the ESP32, so there's nothing in the RF24 lib that will affect that. You could power down the radio with `RF24::powerDown()`, but that would prevent incoming transmissions from triggering the IRQ pin.

dharmikP17, 2022-12-22:
I have both nrf24 and flash IC on different pins. Flash IC is on standard ESP32 pins. Error occurs even before any SPI transaction. 
I am using in build SPIMemory.h library. For simplicity I have this sample code, 

`#include <SPIMemory.h>
#include <FastLED.h>
#include "nRF24L01.h"
#include "RF24.h"

#define FlashChip_CS 5
#define USE_SERIAL true
#define S \
  if (USE_SERIAL) Serial

RF24 radio(16, 17, 26, 25, 27);  // CE, CS, SCK, MISO, MOSI
const uint64_t pipe_1 = 0xE8E8F0F0E1LL, pipe_2 = 0xE8E8F0F0E2LL;

SPIFlash flash;
void setup() {
  S.begin(115200);
  // radio.begin();
  if (!flash.begin()) {
    S.println("\n[ ERROR ] : accessing flash IC");
    S.println(flash.error(VERBOSE));
    while (1)
      ;
  }
  S.println("Flash chip up!");
}

void loop() {
  // put your main code here, to run repeatedly:
}`

It works fine but if I uncomment radio.begin() , I will not reach setup end and ERROR line will execute. I tried manually deactivating nrf24 after begin with digitalWrite on CS pin but that didnt work.

Lets leave sleep modes.

2bndy5, 2022-12-22:
> ```cpp
> RF24 radio(16, 17, 26, 25, 27);  // CE, CS, SCK, MISO, MOSI
>```

That's not how the constructor works. Please read the docs about the [RF24 constructor](https://nrf24.github.io/RF24/classRF24.html#a123231bd955698373d221a84ad40055a) and [`RF24::begin()`](https://nrf24.github.io/RF24/classRF24.html#a048a20c73c7d9b2e02dcbae6fb9c4ba8).

> ```cpp
> const uint64_t pipe_1 = 0xE8E8F0F0E1LL, pipe_2 = 0xE8E8F0F0E2LL;
> ```

You should be using the newer form of addresses: `uint8_t pipe1[5]`. This will save on memory (3 bytes per address), but beware the endianess difference.

```cpp
const uint8_t pipe_1[] = { 0xE1, 0xF0, 0xF0, 0xE8, 0xE8 };
const uint8_t pipe_2[] = { 0xE2, 0xF0, 0xF0, 0xE8, 0xE8 };
```

2bndy5, 2022-12-22:
You also might want to review the [example snippet for the ESP32 in our arduino.md](https://github.com/nRF24/RF24/blob/master/docs/arduino.md#esp32-example) doc.

dharmikP17, 2022-12-24:
Thank you for your help. 

I tried method mentioned in doc but faced compilation error also constructor was different too. Turns out library version I was using was 1.3.1. I have recently downloaded this library so didnt check but I must have downloaded it from someone else's source. 

With latest library version everything works fine. Again thank you very much for your help and time.

closed 2022-12-24 by dharmikP17