nRF24/RF24 #883: [Question]not able to connect nrf24l01 + antenna with arduino uno (used a capacitor too)

AnshumanGu, 2022-12-22:
SPI Speedz	= 10 Mhz
STATUS		= 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1	= 0x0000000000 0x0000000000
RX_ADDR_P2-5	= 0x00 0x00 0x00 0x00
TX_ADDR		= 0x0000000000
RX_PW_P0-6	= 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		= 0x00
EN_RXADDR	= 0x00
RF_CH		= 0x00
RF_SETUP	= 0x00
CONFIG		= 0x00
DYNPD/FEATURE	= 0x00 0x00
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= Disabled

this is what i am getting when i am uploading the following code:

 #include <SPI.h>
#include <RF24.h>
#include <printf.h>

RF24 radio(7, 8);

byte addresses[][6] = {"1Node", "2Node"};

void setup() {
  radio.begin();
  radio.setPALevel(RF24_PA_LOW);
  
  radio.openWritingPipe(addresses[0]);
  radio.openReadingPipe(1, addresses[1]); 
  radio.startListening();
  
  Serial.begin(9600);
  printf_begin();

  radio.printDetails();
  
}

void loop() {
//  empty

}

i have used a 100uf 16v capacitor between the power supply of nrf

2bndy5, 2022-12-22:
Jugding from the `printDetails` output, you probably have it wired up wrong (a common symptom when everything is `0`). Check your SPI connections are connected to the right pins.

You can use the rewritten value of `RF24::begin()` to know if it was successful in setting up the radio. See the lib examples, not a third party tutorial.

It might also be possible that the power supply isn't pumping enough power (Power = Volts * Current). But this would more likely get seen only during transmissions with the PA/LNA modules.

AnshumanGu, 2022-12-23:
I have tried all these things 
btw is it possible that my nrf is not working properly because earlier I supplied it power with 5v pin in arduino?

2bndy5, 2022-12-23:
Capacitors don't protect from over-voltage. If you fed 5v to the radio's VCC (expecting 3v - not 5v tolerant), then you may have damaged the board. Some modules are different though; the sparkfun nRF24 module has a voltage regulator. There are also adapter boards that have a voltage regulator made for the PA/LNA modules.

AnshumanGu, 2022-12-23:
Ok Thanks
I will buy a new nrf module to see whether its something else or the problem is with nrf only.

2bndy5, 2022-12-23:
Cool. You can re-open this thread if needed or open a new issue if about a different topic.

closed 2022-12-23 by 2bndy5