nRF24/RF24 #884: Reduce consumption of nrf24l01 in RX mode

dharmikP17, 2022-12-27:
I am trying to make an energy efficient device with ESp32 and nrf24l01. 

The thing is, I want the device to wake up only when there is a new data on nrf24. After radio.startListening() I can use IRQ pin easily but power consumption is still pretty high.  From what I have read online, there is no way to reduce power from receiver's side. 

Is that really the case ?? Is there any way to put nrf24l01 is standby mode by some other method ?? (Every mA counts here for me.) 
If I use the standby mode , will I be able to generate interrupt through IRQ pin on data receive ?? (I tried this but didnt worked)

Thank you.

TMRh20, 2022-12-27:
The radio needs to be active to receive data so you are looking at 15mA power usage for the radio.

Avamander, 2022-12-27:
If you have accurate time on both sides, you could synchronize at which point in time you're keeping reception enabled, agree upon time slots basically.

2bndy5, 2022-12-27:
If ACK packets are being transmitted in RX mode, then you can reduce the power consumption via [`RF24::setPALevel()`](https://nrf24.github.io/RF24/classRF24.html#ab6a711a9cc14fb459a4a1698b8665d82). This will 
1. Reduce power consumption for TX-ing packets (including ACK packets)
2. Reduce the radio's effective range, so playing with the PA level will depend on your project's spanned area.

I guess you could also disable auto-ack during sleep mode, but the wake up packet will go unacknowledged. **Otherwise**, there is no other option to reduce power consumption because that aspect was hard-wired into the radio's chip.

NOTE:
PA/LNA modules are famous for excessive power consumption during TX. This is why the lib examples call `setPALevel(RF24_PA_LOW)` (a soft hack) to reduce power consumption.

2bndy5, 2022-12-27:
I like a avamander's idea, but time slots is a point of failure (especially with channel hopping) if the clocks don't stay in sync. However, if all radios are driven by an ESP32, then the boards' clock (or an RTC if available) can be synced to an NTP pool; that would keep the clocks accurately in sync.

dharmikP17, 2022-12-28:
Transmission will be handle by customer so timing will be random. Most of the time device will be in sleep, so what what I do after receiving data does not matter much. I guess I probably have to spare 15mA.
I just wanted conclusive answer before I switch to another method or technology and I think I got it.

Thank you all. I will look for alternatives and post if find anything.

closed 2022-12-28 by dharmikP17