nRF24/RF24 #885: Most MCU libraries tend to be MIT or the Lesser GPL is there a reason why this one uses GPL?

chrisco484, 2022-12-30:
This library looks perfect for what we need but we are looking at commercializing the end product which we can't do if we include this library without open sourcing the entire product's source code - of which this library would be a tiny, optional component for some configurations.

Is there a chance this library's license may be changed to, say, the Lesser GPL license one day or is that extremely unlikely?

Is there a chance it may be dual licensed so that it could be used in commercial products? If so what sort of pricing would you be looking at?

2bndy5, 2022-12-30:
I can't imagine why you couldn't dual license your project. Since you say the use of this lib is optional, I would think that opting to use the feature (using this lib) would imply user/dev acceptance of the GPLv2 license. Though, you might have to disclaim that license acceptance upon opting in.

The license itself is not likely to change. We would only dual license the lib if we use software that carries a different license (like RF24Ethernet is).

chrisco484, 2022-12-30:
> I can't imagine why you couldn't dual license your project.
The project relies on a huge investment in some commercially sensitive IP with significant competitive advantage compared to existing similar products. Fully disclosing that source which would be required if we went with dual licensing so that is not an option at this stage.

>Since you say the use of this lib is optional, I would think that opting to use the feature (using this lib) would imply user/dev acceptance of the GPLv2 license. Though, you might have to disclaim that license acceptance upon opting in.
The product is a hardware device designed for end users (no devs) who typically don't have any knowledge of/interest in such software licensing models - other than some text they will likely skim over in the docs. Agreed, if we used this library it would imply user acceptance of GPLv2 but that could only happen if we have disclosed our entire software base as per first paragraph - hence the stumbling block in using it.

> The license itself is not likely to change. We would only dual license the lib if we use software that carries a different license (like RF24Ethernet is).

So, if I understand correctly, there will not likely be a change in the future where this software could be used in closed source commercial products.

TMRh20, 2022-12-30:
The library was initially based on gpl 2 code by maniacbug so I’m not even sure it would be our call to make. It’s not likely to change.

2bndy5, 2022-12-30:
Not sure which paragraph you're referring to, but the way I understand it is:
- According to Section 1 of the GPLv2: If you didn't modify the RF24 lib code, then you should be able to distribute it in binary form as long as it is accompanied with a copy of the license (which should also include the copyright notice and disclaimer of warranty).
   ```text
   1. You may copy and distribute verbatim copies of the Program's
   source code as you receive it, in any medium, provided that you
   conspicuously and appropriately publish on each copy an appropriate
   copyright notice and disclaimer of warranty; keep intact all the
   notices that refer to this License and to the absence of any warranty;
   and give any other recipients of the Program a copy of this License
   along with the Program.
  
   You may charge a fee for the physical act of transferring a copy, and
   you may at your option offer warranty protection in exchange for a fee.
   ```
- According to section 2 of the GPLv2: If you made modifications, then you're obligated to distribute a copy of the lib with modifications along with the license. I think this could be as simple as a link to your RF24 fork.
   ```text
   2. You may modify your copy or copies of the Program or any portion
   of it, thus forming a work based on the Program, and copy and
   distribute such modifications or work under the terms of Section 1
   above, provided that you also meet all of these conditions:
   ```
So, did you make modifications?

chrisco484, 2022-12-30:
> The library was initially based on gpl 2 code by maniacbug so I’m not even sure it would be our call to make.

Ah, in that case it would be maniacbug's call to make and changes to the licensing.

chrisco484, 2022-12-30:
> So, did you make modifications?

No but I think the issue with GPL is that it doesn't matter if you make changes to a GPLed library: Any app that links to a GPLed library must have its source made open. That's the 'contagion' effect of GPL which is why the Lesser GPL exists - to avoid that contagion effect and allow use of Lesser GPL licensed libraries in commercial products (and some organizations even fear there may be issues with using libraries with the Lesser GPL license in closed source apps - though I don't know why)

Even though all the core Linux libraries are GPLed (not Lesser GPLed) Linus added a special clause to his version of the GPL that covers Linux which explicitly excludes the need for apps written in Linux to open source their code. This was a smart move and was necessary for the success of Linux otherwise Oracle, IBM etc., would never have created a version of their apps that run on Linux.

2bndy5, 2022-12-30:
> No but I think the issue with GPL is that it doesn't matter if you make changes to a GPLed library: Any app that links to a GPLed library must have its source made open.

You seem to be drawing on knowledge from other sources I'm not aware of. I don't see anything that strict in [the license](https://github.com/nRF24/RF24/blob/master/LICENSE). Actually, it is rather vague about the forms of distribution (wording does seem to be geared toward open source dists).

@Avamander Are we missing something important here? Should we seriously consider an optional license for closed source projects?

2bndy5, 2022-12-30:
Maniacbug has been unresponsive for years (concerning the RF24 stack). I don't think we can rely on maniacbug's involvement.

chrisco484, 2022-12-31:
> Maniacbug has been unresponsive for years (concerning the RF24 stack). I don't think we can rely on maniacbug's involvement.

I managed to make contact with the original author, maniacbug, almost 2 years ago to see what he thought about re-licensing to a commercially friendly license or dual licensing.

I'll try to make contact again to see if the's thought any further about it.

Avamander, 2022-12-31:
> Ah, in that case it would be maniacbug's call to make and changes to the licensing.

Nope, all significant contributors have to make that call. Even if he could with the original, it does **not** apply to this fork.

> No but I think the issue with GPL is that it doesn't matter if you make changes to a GPLed library: Any app that links to a GPLed library must have its source made open.

That is basically true and intentional. GPL is meant to function as tit-for-tat rather than take-it-and-run.

> Are we missing something important here? Should we seriously consider an optional license for closed source projects?

I don't get the impression we are. GPL is not hard to comply with, Linux and how people run proprietary kernel modules is a rather good example.

2bndy5, 2022-12-31:
So, I guess that means the answer is: no you cannot use this lib in closed source projects?

Personally, I'm not against a dual license. My only real main concern is that there is no guarantee of undamaged hardware because that's very much out of our hands. Obviously, I can be rather lenient about decided legalese though.

Avamander, 2022-12-31:
> So, I guess that means the answer is: no you cannot use this lib in closed source projects?

Not in fully closed ones, no.

Nothing stops you from only giving the source to your customers though. But they get the rights that come with the license.

chrisco484, 2023-01-01:
> GPL is not hard to comply with, 

GPL's issue isn't "difficulty of compliance" it's whether or not it makes commercial sense to open source your commercial products because that's what it mandates: even if you use one single GPLed library with 1000 lines of code in a 10 million line commercial app.

> Linux and how people run proprietary kernel modules is a rather good example.

As I said above, Linus used a modified GPL for Linux which explicitly excludes the need for closed source apps to open their source if they link to the binaries of any of Linux' GPL (modified) licensed libraries.

He knew that he needed Linux to be friendly to commercial app development and indeed some of those players are now big contributors to some open source code bases in Linux and elsewhere.

Linux is probably a good example of why a more permissive license is better than pure GPL.

Avamander, 2023-01-02:
> even if you use one single GPLed library with 1000 lines of code in a 10 million line commercial app.

Yep, if that's the approach to integration then that's the way it's supposed to work.

Very frankly it's an anti-leech license. 

> Linus used a modified GPL for Linux which explicitly excludes the need for closed source apps to open their source if they link to the binaries of any of Linux' GPL (modified) licensed libraries.

Nope. It's GPLv2.

> Linux is probably a good example of why a more permissive license is better than pure GPL.

That's be the BSDs, probably more widespread in proprietary products but often hostile to the users and the larger community.

chrisco484, 2023-01-02:
> > Linus used a modified GPL for Linux which explicitly excludes the need for closed source apps to open their source if they link to the binaries of any of Linux' GPL (modified) licensed libraries.
> 
> Nope. It's GPLv2.

Not according to:

https://github.com/torvalds/linux/blob/master/COPYING

which states:

> The Linux Kernel is provided under:
> 
> 	SPDX-License-Identifier: GPL-2.0 WITH Linux-syscall-note
> 
> Being under the terms of the GNU General Public License version 2 only,
> according with:
> 
> 	LICENSES/preferred/GPL-2.0
> 
> With an explicit syscall exception, as stated at:
> 
> 	LICENSES/exceptions/Linux-syscall-note
> 

So it's not pure GPLv2 - it's GPLv2 with a very deliberate modification/exception.

The explicit exception permits apps to compile their source using Linux header files without needing to open their own source code as would be the case with GPL without this explicit exception.

There are also some other exceptions if you are interested:

https://github.com/torvalds/linux/tree/master/LICENSES/exceptions

Avamander, 2023-01-02:
> So it's not pure GPLv2 - it's GPLv2 with a very deliberate modification/exception.

Yes, but that is an explicit statement about syscalls, not linking as you originally said. I think you would find the threads about "nvidia GPL condom" very interesting.

I think there isn't anything left to discuss, so I'm closing this issue.

chrisco484, 2023-01-02:
> > So it's not pure GPLv2 - it's GPLv2 with a very deliberate modification/exception.
> 
> Yes, but that is an explicit statement about syscalls, not linking as you originally said. I think you would find the threads about "nvidia GPL condom" very interesting.
> 

Did you read it?

2bndy5, 2023-01-02:
@chrisco484 I'm sorry we couldn't be more accommodating. There are other nRF24L01 libs for the Arduino platform, though I doubt their API is as convenient.

Sorry for the good cop/bad cop routine. This dual licensing decision must be unanimous, and I don't know if that is achievable.

chrisco484, 2023-01-02:
> @chrisco484 I'm sorry we couldn't be more accommodating. There are other nRF24L01 libs for the Arduino platform, though I doubt their API is as convenient.
> 

That's fine. I have looked at a few alternatives with more permissive licenses. Given the SPI interface to these nRF24L01 devices there's a few alternative libraries and also many direct examples available.

> Sorry for the good cop/bad cop routine. This dual licensing decision must be unanimous,

No worries. I was just trying to make the point that even Linux made explicit exceptions to the GPLv2 to try to accommodate/encourage the use by close source/commercial apps.

> and I don't know if that is achievable.

I completely understand - especially given that any change would, as only a first step in a much larger process, need to come from the original author, maniacbug.

Paid for/closed source software isn't all evil. Most of use probably have day jobs at commercial software entities with closed source which pay all the bills so we have time for open source side projects ;)

2bndy5, 2023-01-02:
> few alternative libraries 

That _may_ be for the _Arduino_ framework. Nordic had nRF24L01 support in the (now deprecated) nRF Go SDK -> this was then [adapted into an Arduino lib](https://github.com/pstolarz/NRF_HAL) (which carries Nordic Semi's original license) that you can also [use in PlatformIO](https://registry.platformio.org/libraries/pstolarz/NRF_HAL).

chrisco484, 2023-01-03:
> That _may_ be for the _Arduino_ framework. Nordic had nRF24L01 support in the (now deprecated) nRF Go SDK -> this was then [adapted into an Arduino lib](https://github.com/pstolarz/NRF_HAL) (which [...]

Thanks for that reference - I had not seen that one before. It looks reasonably easy to use.