nRF24/RF24 #888: Wemos D1 mini and nRF24l01 compatibility issue?

DavethaSlave, 2023-01-04:
I have a Wemos D1 mini that i am trying to use with the nRF24l01 radios. I seem to be having a compatibility issue? I have loaded a simple example sketch that has just the include libraries and it wont compile. I have changed the board type to any other board not in the ESP8622 section and it complies fine. Are the D1 mini's and RF24 library compatible? Thank you in advance. I have include the sample code.

`#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

 RF24 radio(7, 8); // CE, CSN

 const byte address[6] = "00001";

void setup() {
 /*
  radio.begin();
  radio.openWritingPipe(address);
  radio.setPALevel(RF24_PA_MIN);
  radio.stopListening();
  */
 
}

void loop() {
  /*
  const char text[] = "Hello World";
  radio.write(&text, sizeof(text));
  delay(1000);
  */
  
}`

2bndy5, 2023-01-04:
This library is compatible with the ESP8266 core. We even have an [ESP8266 example that demonstrates using a different SPI bus](https://github.com/nRF24/RF24/blob/master/docs/arduino.md#esp8266-example).

We would help more if you could provide what the compiler is saying.

2bndy5, 2023-01-04:
Please make sure that your ESP8266 core is up-to-date (via Arduino IDE board manager) and that you're using the latest release of RF24 lib (currently v1.4.6). I just test compiled our GettingStarted example for the Wemos D1 mini boards, and I got no compiler error.

2bndy5, 2023-01-04:
Looks like a conflict in commonly named symbols. I think the proper fix is to namespace our defines (a non-trivial solution).

It's weird that I don't get that error with the examples. Does removing `#include <nRF24L01.h>` help?

DavethaSlave, 2023-01-04:
I found a new version of RF24 in the library manager. The version I had was from a link in some "how to" videos. It said RF24 master. I deleted the old one and installed the version in library manger and it compiles now. Thank you. Looks like I had possibly an incomplete third party one.

2bndy5, 2023-01-04:
PLEASE don't use unofficial sources! Use the Lib manager in PlatformIO/Arduino IDE. I would also ask you to review our lib examples because youtube videos can be outdated (or just erroneous).

closed 2023-01-04 by 2bndy5

DavethaSlave, 2023-01-04:
Thank you  have a good day.