nRF24/RF24 #889: Will two nrf24l01's work with two different boards?

DavethaSlave, 2023-01-06:
Hello. I am trying to use a wemos D1 mini to transmit to an Arduino pro mini through nrf24l01. I have successfully used the sample code with matching boards and it works, however when taking a working d1 mini and pairing it with a Arduino pro mini, it doesn't work. I read the FAQ section on nrf24 but it doesn't seem to hit this topic. I did make sure to adjust the pin assignments for the new board in the code. I saw that someone had mentioned possible "sizeof" differences in some boards, but I can't find that discussion anymore. Does this have anything to do with it?  Thank you.

2bndy5, 2023-01-06:
> Will two nrf24l01's work with two different boards?

yeah, that's the beauty of these transceivers.

> however when taking a working d1 mini and pairing it with a Arduino pro mini, it doesn't work.

Can you explain this more? How would I reproduce it?

2bndy5, 2023-01-06:
> I saw that someone had mentioned possible "sizeof" differences in some boards

I think the ESP8266 core has different `malloc` behavior, so certain data types aren't always the same size in memory compared to other MCUs (like the pro mini).

DavethaSlave, 2023-01-06:
I tried an experiment to make sure that my connections and pin assignments were good. I used a pair of D1 mini's and radios with a simple code that transmits "Hello World". One was setup as a transmitter, the other a receiver. Everything worked fine. That made me feel better that the connections on the D1 mini were good. I substituted the receiver for the pro mini, and that is where I couldn't get it to work. Connections on the pro mini are as follows:
CE - 8
CSN - 10
SCK - 13
Mosi - 11
Miso - 12
Also I am using the radio adapters with the built in voltage regulator. Everything lights up with both experiments. 
I plan on trying the same experiment with a pairs of pro mini's to satisfy that the connections are good so I can stop focusing on those. Also I have included both transmitter code (for the D1 mini), and the receiver code ( for the pro mini). Thanks.
Transmitter code:
`[code]
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

RF24 radio(D2, D8); // CE, CSN

const byte address[6] = "00001";

void setup() {
  radio.begin();
  radio.openWritingPipe(address);
  radio.setPALevel(RF24_PA_MIN);
  radio.stopListening();
}

void loop() {
  const char text[] = "Hello World";
  radio.write(&text, sizeof(text));
  delay(1000);
}
[/code]
`
receiver code:
`[code]
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

RF24 radio(8,10); // CE, CSN

const byte address[6] = "00001";

void setup() {
  Serial.begin(9600);
  radio.begin();
  radio.openReadingPipe(0, address);
  radio.setPALevel(RF24_PA_MIN);
  radio.startListening();
}

void loop() {
  if (radio.available()) {
    char text[32] = "";
    radio.read(&text, sizeof(text));
    Serial.println(text);
  }
}
[/code]
`

2bndy5, 2023-01-06:
> `radio.openReadingPipe(0`

You should avoid using pipe 0 for reading. Using pipe 1-5  for reading is preferred. See [the docs](https://nrf24.github.io/RF24/classRF24.html#a9edc910ccc1ffcff56814b08faca5535) for more info.

I don't see anything else wrong with the code.

> and that is where I couldn't get it to work

This may seem abvious to you, but how does it not work? Are there compiler errors? Does `RF24::begin()` return false? Or do all transmissions just fail?

2bndy5, 2023-01-06:
> Also I am using the radio adapters with the built in voltage regulator.

This just means that you have to feed 5v to the adaptor board's VCC (not 3v).

2bndy5, 2023-01-06:
> Where can I find documentation for all the functions and their arguments for the RF24 library?

Its the first line in the lib's root README: https://nRF24.github.io/RF24

DavethaSlave, 2023-01-06:
Everything compiles and uploads fine. By not working, I mean that nothing is being diplayed in the serial monitor for the receiver board. Is there any commands lines that I can add to the code to check connection status of the two radios? Im not sure how to check for begin() = false.
Maybe:
if(!radio.begin()){
Serial.println("Radio not found");
}
 Do this on both radios and monitor the results?

2bndy5, 2023-01-06:
> Is there any commands lines that I can add to the code to check connection status of the two radios?

We often use `RF24::printDetails()` (or `RF24::printPrettyDetails()`) to get debug info off the radio.

> Im not sure how to check for begin() = false.

There's an example snippet in the [`RF24::begin()` docs](https://nrf24.github.io/RF24/classRF24.html#a048a20c73c7d9b2e02dcbae6fb9c4ba8). Note: the "radio not responding prompt may not show if sketch does not wait for Serial monitor to be opened `while (!Serial) ;`.

> Do this on both radios and monitor the results?

Yes. Contrary to 3rd-party tutorials, `RF24::begin()` should be checked when it returns false as any further behavior will be undesirable.

DavethaSlave, 2023-01-07:
I got it working!. The checking for radio.begin() to be true is a great tool for troubleshooting. I was able to narrow down which side wasn't talking to the radio. I found that I assigned my CE and CSN wires backwards, or wired them backwards. Switching the numbers in the code was easier than switching the wires. This isn't the code that I am going to be running. I have a D1 mini already running a NPTclient to get network time and using that time to switch aquarium lighting from day mode to night mode. I am going to add the radio to that one and have it transmit across the house to the pro mini at another aquarium to switch that one with relays.  Again....thanks for the help.

closed 2023-01-07 by 2bndy5