nRF24/RF24 #891: [Question] Reasoning behind rp2 SPI functionality being static?

TheGarkine, 2023-01-20:
Hey, I am working with RF24 interfaces for a while now, and started to build a more complex use case where I need two of them to connect to an rp2040 microcontroller. For simplicity and debugging reasons, I thought I'd want to use the two SPI busses on this specific microcontroller, but I quickly realized this is not correctly working. Then I realized that the rpi2040-specific SPI components `/utility/rp2/spi.cpp` are static, such that it is no possible to use this class to communicate on two busses simultaneously.

Maybe someone can give my some insights why this design was chosen? I think I might just put my second nRF24 on the same bus, but was just wondering and am looking to learn.

2bndy5, 2023-01-20:
Are you using the PicoSDK? That's what the src in /utility/rp2/spi.cpp is used for. You can change the SPI bus used in the PicoSDK in source code [as documented here](https://nrf24.github.io/RF24/md_docs_pico_sdk.html#autotoc_md53).

If you're using the an Arduino core for the RP2040, then you can specify the SPI similarly [as described here](https://nrf24.github.io/RF24/md_docs_arduino.html#autotoc_md14) (though no specific RP2040 example snippet.

---
Please clarify the which you're using.

TheGarkine, 2023-01-20:
Thank you for your fast reply @2bndy5 , yes I am using the pico-sdk. I know I can use this functionality to change the SPI bus for a single device but as mentioned above, I want to use **two** nRF24 on two different busses, so I basically want to use spi0 for one device and spi1 for the other. The problem is that the class defined for rp2040 has all of its methods and the spi hw interface statically defined for its class (even though the class itself is not static, see "/utility/rp2/spi.cpp". So i cannot use `spi.begin(spi0/1,...)` twice, or i would overwrite the changes of the first invocation. Neither can i create two SPIs myself, since even though I create two individual instances, they share their hw_id due to its static definition.

TheGarkine, 2023-01-20:
So basically, I got my two nRF24 modules to work by removing all static keywords in [spi.h](https://github.com/nRF24/RF24/blob/master/utility/rp2/spi.h). And then made two instances of my SPI which i gave to the radios:
```cpp
static SPI SPI0;
static SPI SPI1;

RF24 radio(8, 5);
RF24 radio2(14, 13);

bool setup() {
    SPI0.begin(spi0, 6, 7, 4);
    if (!radio.begin(&SPI0)) {
        printf("Radio hardware is not responding!\n");
        return false;
    }
    printf("Radio1 initalized!");

    SPI1.begin(spi1, 10, 11, 12);
    if (!radio2.begin(&SPI1)) {
        printf("Radio2 hardware is not responding!\n");
        return false;
    }
    ....
}
```

I would just like to understand how this design decision was made, so maybe I am in the wrong here and need to do this better.

2bndy5, 2023-01-20:
Thank you for the code snippet. Yes, that is how the code is supposed to work, but the `static` keywords may have been simply an oversight (PicoSDK support was a collaborative effort).

You are welcome to submit a PR to remove the static attributes. Clearly this scenario isn't something we hardware tested, so thank you for reporting this.

TheGarkine, 2023-01-20:
Nice, will do a PR when I have time to properly write and document it, (and also test the current way, such that no old code breaks). Thank you for your support, I was just really confused when I found out! Will categorize this issue correctly now.

2bndy5, 2023-01-20:
You might find that our [/.github/workflows/build_rp2xxx.yml](https://github.com/nRF24/RF24/blob/master/.github/workflows/build_rp2xxx.yml) CI workflow can be helpful to build multiple examples (and archive the built binaries as [workflow artifacts](https://github.com/nRF24/RF24/actions/runs/3872833489) which can be downloaded and copied to the board in DFU mode).

2bndy5, 2023-01-20:
@krypton Can you recall why the SPI class uses `static` attributes? I didn't think anything of it at the time.

TheGarkine, 2023-01-20:
@2bndy5 , created a PR #892 for this and also added an example for my use case. Also tested the old method and it is not broken by the changes. Feel free to take your time to discuss internally, if my changes break something else.

Have a great weekend!

kripton, 2023-01-20:
> @kripton Can you recall why the SPI class uses `static` attributes? I didn't think anything of it at the time.

Hi :) To be honest, I don't really have an explanation as well. It might have been to reduce memory usage but it should ne neglectable. I can check the PR against my code in the following days but in general, I don't oppose thr change. It allows the code to be used in a more flexible way, so why not?

2bndy5, 2023-01-21:
I agree. The memory capacity isn't going to get hogged by this change.

closed 2023-01-23 by 2bndy5