nRF24/RF24 #896: typedef GPIO pin variables

2bndy5, 2023-02-11:
We might save a few bytes (on certain platforms/chips) by using `typedef` to configure the datatype used to specify pin numbers on different platforms/processors.

For example:
in utility/ATTiny/RF24_arch_config.h
```cpp
typedef uint8_t rf24_gpio_t;
```
in utility/ATXMegaD3/RF24_arch_config.h
```cpp
typedef uint16_t rf24_gpio_t;
```
in RF24.h
```cpp
    rf24_gpio_t ce_pin;    /* "Chip Enable" pin, activates the RX or TX role */
    rf24_gpio_t csn_pin;   /* SPI Chip select */

    RF24(rf24_gpio_t _cepin, rf24_gpio_t _cspin, uint32_t _spi_speed = RF24_SPI_SPEED);
```

I'm inclined to use `uint8_t` (as it was from maniacbug days) in RF24_config.h for any platform/chip that doesn't have its own utility/\*/RF24_arch_config.h, but maybe it is safer to just use `uint16_t` for continuity.

### Additional context
As usual, I have an ulterior motive here. In order to extend support to the zephyr project (which uses DeviceTree specs to declare GPIO pins), I would need a way to change what datatype is passed to the RF24 c'tor. By `typedef`ing the gpio pins' datatype, I can more easily implement wrappers for the `pinMode()` and the like.

TMRh20, 2023-02-12:
I don't think this should be an issue, seems straight forward enough.

2bndy5, 2023-02-16:
I found that [most arduino cores have `NUM_DIGITAL_PINS` defined](https://github.com/search?q=org%3Aarduino+NUM_DIGITAL_PINS&type=code) in the board variant's pins_arduino.h header!

This will help us adequately determine whether to use `uint8_t` or `uint16_t` in a future-compatible way! I'm still going with `uint16_t` if `!defined(NUM_DIGITAL_PINS)` for boards that aren't specifically handled in utility/\* or RF24_config.h.

PR #898 (2023-04-22): Alias gpio pin datatype
resolves #896
closed 2023-04-22 by TMRh20