nRF24/RF24 #897: Is this compatible with the ATSAMD21 Processor? [Question]

AceSimulations, 2023-02-14:
I am trying to get the gettingStarted working on a Adafruit feather M0 processor but it will say "radio hardware not responding". I have tested this same code on a feather 32u4 and it works flawlessly. I have verified the CE and CSN pins are correct and am fairly certain that the MO, MI, SCK pins are also correct considering they are labeled on the board. I have done tons of searching and it seems like this has been an issue in the past but it does not seem to be resolved. 

The pinout for the M0 is here - https://cdn-learn.adafruit.com/assets/assets/000/107/293/original/adafruit_products_Adafruit_Feather_M0_Adalogger_Pinout.png?1639414889

VCC - 3.3
GND - GND
CE - 11
CSN - 12
SCK - SCK
MOSI - MOSI
MISO - MISO

2bndy5, 2023-02-14:
I can vouch that this works for the ATSAMD21. In fact, I've added enough code to make `printDetails()` work on that chip if using the Adafruit Core for their M0 boards. Its worth noting that I always have to change the CE & CSN pin numbers specified (via [`RF24::RF24(ce_pin, csn_pin)`](https://nrf24.github.io/RF24/classRF24.html#a123231bd955698373d221a84ad40055a)) in the examples to get them working on my various ATSAMD21 boards...

Looking at your pinout link, I want to be clear that when you say `11` & `12`, you really mean `D11` & `D12` (not the 11 & 12 outlined in green).

> I have done tons of searching and it seems like this has been an issue in the past but it does not seem to be resolved.

I'm not sure what you're referring to here.

2bndy5, 2023-02-15:
What radio module are you using? PA/LNA?

Sometimes I have to lower the SPI speed to 4000000 (3rd arg in c'tor) if my wires are a bit lengthy.

AceSimulations, 2023-02-15:
I tried D11 and D12 but it claims that it is undefined. I am not sure what I am doing wrong. 
`// instantiate an object for the nRF24L01 transceiver
RF24 radio(D11, D12);  // using pin 7 for the CE pin, and pin 8 for the CSN pin
`
I am using the NRF24L01+ PA module.

2bndy5, 2023-02-15:
The only board (that I know of) that actually has pins defined they way they are written in the silkscreen is the NodeMCU (ESP8266).

```cpp
RF24 radio = RF24(11, 12);
```
The way you are declaring the pins should be correct. I wish I could ensure confidence there, but the [pin definitions in the Arduino core](https://github.com/adafruit/ArduinoCore-samd/blob/master/variants/feather_m0/variant.cpp) are rather convoluted.

Have you tried [lowering the SPI speed](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#payloads-receivedsent-are-inaccurate-or-printdetails-outputs-the-unexpected-values)? Some boards have higher impedance than others, and I usually run into this with an ATSAMD21 based board.

AceSimulations, 2023-02-15:
I lowered the SPI speed quite a bit and still no luck. I have tried basically every pin combination on the board. I know the board and transceiver are functioning. I even swapped transceivers and still nothing. Is there a way I can define every pin I am using instead of just CE and CSN? (Ex. MOSI, MISO, SCK, CE, CSN) The Adafruit feather M0 uses GPIO 24/23/22 for SCK/MOSI/MISO.

AceSimulations, 2023-02-15:
Update: I did something I am not sure what and now it is working. I believe it was using `RF24 radio(5, 6, 4000000);`. I am not sure why it is now working but my guess is my wires were barely too long for the SPI speed I was testing it at.

2bndy5, 2023-02-15:
> Is there a way I can define every pin I am using instead of just CE and CSN? (Ex. MOSI, MISO, SCK, CE, CSN)

Technically yes, but that technique is meant for using a board's non-default SPI bus. More info is in the [RF24 docs' Arduino page](https://nrf24.github.io/RF24/md_docs_arduino.html#autotoc_md14). I didn't add an example for the ATSAMD21 because many boards don't usually have the required GPIO pins exposed.

If you were trying to refer to software-implemented SPI, then that may be possible [via the DigitalIO library](https://nrf24.github.io/RF24/md_docs_arduino.html#autotoc_md12) (required for SoftSPI in RF24). There might be some arch limitations in DigitalIO though.

> not sure why it is now working but my guess is my wires were barely too long for the SPI speed I was testing it at

In my experience the wires don't have to be very long (happens with 5 inch jumpers for me).

---
FWIW, I also tend to use A0 & A1 on my QtPy boards (1 of which is has the ATSAMD21):
```cpp
radio = RF24(0, 1);
```

I'll close this now that you're back on track. 🎉

closed 2023-02-15 by 2bndy5