nRF24/RF24 #899: radio.begin() hangs on raspberry pi pico since b5a905b

felher, 2023-03-02:
Hey all!

When using the library on my raspbery pi pico, calling `radio.begin()` hangs forever. This was not always the case. I used a former version of this library in an older project and it worked fine. I run a git bisect with the version I had used in the old project and it seems that since commit b5a905b, calling `radio.begin()` just hangs.

Here is a minmal code example:

```c
#define CE_PIN 7
#define CSN_PIN 8
#define IRQ_PIN 6

#include <RF24.h> // RF24 radio object

RF24 radio(CE_PIN, CSN_PIN);

int main() {
  const uint8_t LED_PIN = 25;
  stdio_init_all();
  gpio_init(LED_PIN);
  gpio_set_dir(LED_PIN, GPIO_OUT);

  radio.begin();

  while (true) {
    gpio_put(LED_PIN, 1);
    sleep_ms(100);
    gpio_put(LED_PIN, 0);
    sleep_ms(100);
  }
}
```

I did expect to see blinking, but I'm not seeing any. If I remove the `radio.begin()` line, the LED starts to blink again. If I reset library to a version before b5a905b I can leave `radio.begin()` in and it works fine.

I'm not an expert by any stretch, so this might well be an misunderstanding/mistake on my part. If so, I apologize for the spurious bug report

2bndy5, 2023-03-03:
The mentioned commit neglects
https://github.com/nRF24/RF24/blob/da8987470e6c67a0aed88a209609318c12a7ddd2/utility/rp2/RF24_arch_config.h#L31

which is the `SPI` instance that RF214 uses by default.

Maybe, the removal of `static` attributes requires the object be instantiated like so
```cpp
SPI spi = SPI();
```

2bndy5, 2023-03-03:
yep, we didn't thuroughly test the default `RF24::begin()` from that commit... IO got it working with the following patch:

```diff
diff --git a/RF24.cpp b/RF24.cpp
index c0ceef8a..68144a25 100644
--- a/RF24.cpp
+++ b/RF24.cpp
@@ -999,6 +999,7 @@ bool RF24::begin(void)
     _spi->begin(csn_pin);

 #elif defined(RF24_RP2)
+    _spi = new SPI();
     _spi->begin(PICO_DEFAULT_SPI ? spi1 : spi0);

 #else // using an Arduino platform || defined (LITTLEWIRE)
diff --git a/utility/rp2/RF24_arch_config.h b/utility/rp2/RF24_arch_config.h
index 7d5a1a93..3f24e949 100644
--- a/utility/rp2/RF24_arch_config.h
+++ b/utility/rp2/RF24_arch_config.h
@@ -28,8 +28,6 @@
 #define _SPI   SPI
 #define RF24_SPI_PTR

-static SPI spi;
-
 #ifdef SERIAL_DEBUG
     #define IF_SERIAL_DEBUG(x) ({ x; })
 #else
```
However, I'm not receiving any data, only sending data works. This might be a separate issue with my RPi4 setup, so I'll try again with a different device... I'm testing the examples_pico/getting_started.cpp on my Feather RP2040.

2bndy5, 2023-03-03:
> However, I'm not receiving any data, only sending data works

It wasn't my RPi setup, it was the adapter board I had connecting my nRF24 module to my Feather RP2040. This above patch works fine to fix this regression

@felher  I pushed the fix to the master branch, so you should be good to go now

closed 2023-03-03 by 2bndy5

felher, 2023-03-03:
Holy moly, this was fast! Thank you!

TMRh20, 2023-03-03:
That was fast! You rock @2bndy5

commit 557d165c (2023-03-03): resolves #899

a regression from #892 about
using the default instance of the SPI wrapping class for PicoSDK