nRF24/RF24 #901: [Question] why nrf24 compile on ubuntu server 22.04 but not excute?

mjk14, 2023-03-12:
I have done these steps in order to install and execute nrf24 on ubuntu server 22.04 (64-bit):
1- sudo apt update
2- sudo apt-get install -y gcc-arm-linux-gnueabihf  g++-arm-linux-gnueabihf  make
3- wget http://tmrh20.github.io/RF24Installer/RPi/install.sh
4- sudo chmod +x install.sh
5- ./install.sh
point: I answered yes to all the questions in the script and for the "install RF24 core using?" question, I chose option 2.SPIDEV(Comatibility, default)

6- ./RF24/configure --driver=SPIDEV
```
[SECTION] Detecting arm compilation environment.
  [OK] arm-linux-gnueabihf-gcc detected.
  [OK] arm-linux-gnueabihf-g++ detected.
[SECTION] Detecting target machine.
[OK] machine detected: SoC=BCM2835, Type=RPi, CPU=aarch64.
[SECTION] Detecting OS.
  [INFO] OS detected:LINUX.
[SECTION] Preparing configuration.
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] Finished.
```
7- make
```
arm-linux-gnueabihf-g++ -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT -c RF24.cpp
arm-linux-gnueabihf-g++ -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT -c utility/SPIDEV/spi.cpp
arm-linux-gnueabihf-g++ -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT -c utility/SPIDEV/gpio.cpp
arm-linux-gnueabihf-gcc -fPIC  -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT -c utility/SPIDEV/compatibility.cpp
[Linking]
arm-linux-gnueabihf-gcc -shared -Wl,-soname,librf24.so.1 -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT -o librf24.so.1.4.6 RF24.o spi.o gpio.o compatibility.o
```

8- sudo make install
```
[Installing Libs to /usr/local/lib]
[Installing Headers to /usr/local/include/RF24]
```

9- cd RF24/examples_linux && make
```
arm-linux-gnueabihf-g++ -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib gettingstarted.cpp -lrf24 -o gettingstarted
arm-linux-gnueabihf-g++ -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib acknowledgementPayloads.cpp -lrf24 -o acknowledgementPayloads
arm-linux-gnueabihf-g++ -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib manualAcknowledgements.cpp -lrf24 -o manualAcknowledgements
arm-linux-gnueabihf-g++ -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib streamingData.cpp -lrf24 -o streamingData
arm-linux-gnueabihf-g++ -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib multiceiverDemo.cpp -lrf24 -o multiceiverDemo
arm-linux-gnueabihf-g++ -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib scanner.cpp -lrf24 -o scanner
```

10- ./multiceiverDemo
```
-bash: ./multiceiverDemo: No such file or directory
```

The question is why the executable file does not run? Are there any changes to be made in the compilation process? the content of the Makefile.inc is as follows:

```
OS=LINUX
SOC=BCM2835
DRIVER=SPIDEV
CPUFLAGS=-marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard
CFLAGS=-marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -DRF24_NO_INTERRUPT
PREFIX=/usr/local
REMOTE_PREFIX=/usr/local
LIB=rf24
LIBNAME=librf24.so.1.4.6
LIB_VERSION=1.4.6
LIBSYMLINKS=librf24.so.1.4 librf24.so.1 librf24.so
LIBDEPRECATE=librf24-bcm.so
CC=arm-linux-gnueabihf-gcc
CXX=arm-linux-gnueabihf-g++
LIB_DIR=/usr/local/lib
REMOTE_LIB_DIR=/usr/local/lib
HEADER_DIR=/usr/local/include/RF24
REMOTE_HEADER_DIR=/usr/local/include/RF24
DRIVER_DIR=utility/SPIDEV
ARCH_DIR=utility
SHARED_LINKER_FLAGS= -shared -Wl,-soname,librf24.so.1
LDCONFIG=ldconfig
REMOTE_LDCONFIG=/sbin/ldconfig
EXAMPLES_DIR=/usr/local/bin
REMOTE_EXAMPLES_DIR=/usr/local/bin
```

2bndy5, 2023-03-12:
For a 64 bit OS, you must use the [cmake instructions](https://nrf24.github.io/RF24/md_docs_using_cmake.html). The old `configure` script still exists to not break existing user projects, but it was originally designed to be used with armhf compiler.

mjk14, 2023-03-12:
Thank you @2bndy5  for your quick reply. I used cmake but when use ./gettingstarted got the following error: 
```
terminate called after throwing an instance of 'GPIOException'
  what():  can't export GPIO pin .check access rights
Aborted (core dumped)
```
What is the reason for this error? Since I intend to use spi devices, I chose option 2(most compatible, default), is it correct?

closed 2023-03-13 by mjk14