nRF24/RF24 #902: Arduino 1-to-1 wireless communication using Nrf24L01

Miba786, 2023-03-16:
Hi, i need help with my Arduino wireless communication with NRF24L01

I am trying to do a 1-1 one-way communication between 2 arduinos using the NRF24L01 module.

On my transmitter side, I have 1 Ultrasonic Sensor and the readings will be sent over to the receiver side.

This is my code for my transmitter side:
```
#include <RF24.h>
#include <RF24Network.h>
#include <SPI.h>

RF24 radio(10, 9);               // nRF24L01 (CE,CSN)
RF24Network network(radio);      // Include the radio in the network
const uint16_t this_node = 00;   // Address of this node in Octal format
const uint16_t node01 = 01;      

void setup() {
  pinMode(trigPin, OUTPUT); // Sets the trigPin as an Output
  pinMode(echoPin, INPUT); // Sets the echoPin as an Input
  SPI.begin();
  radio.begin();
  network.begin(90, this_node);  //(channel, node address)
}

void loop() {

  digitalWrite(trigPin, LOW);
  delayMicroseconds(2);
  // Sets the trigPin on HIGH state for 10 micro seconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
  // Reads the echoPin, returns the sound wave travel time in microseconds
  duration = pulseIn(echoPin, HIGH);
  // Calculating the distance
  distance = duration * 0.034 / 2;
  network.update();

  unsigned long US1_Measurement = distance
  RF24NetworkHeader header(node01);     // (Address where the data is going)
  bool ok = network.write(header, &US1_Measurement , sizeof(US1_Measurement )); // Send the data
}
```

On my receiver side, it will receive the readings from the transmitter and display on the serial monitor.

This is my code for the receiver side:
```
#include <RF24.h>
#include <RF24Network.h>
#include <SPI.h>

RF24 radio(10, 9);               // nRF24L01 (CE,CSN)
RF24Network network(radio);      // Include the radio in the network
const uint16_t this_node = 01;   // Address of our node in Octal format ( 04,031, etc)

void setup() {
  SPI.begin();
  radio.begin();
  network.begin(90, this_node); //(channel, node address)
}

void loop() {
  network.update();
  while ( network.available() ) {     // Is there any incoming data?
    RF24NetworkHeader header;
    unsigned long incomingData;
    network.read(header, &incomingData, sizeof(incomingData)); // Read the incoming data
    Serial.println(incomingData); 
  }
}
```
This is working. Now what I am trying to do now is, on my transmitter side, i want to add an additional ultrasonic sensor. So now i want my transmitter to send both readings from my ultrasonic sensors to the receiver at the same time. I am not sure on how to amend the code.

Any help would be much appreciated. Thanks

2bndy5, 2023-03-16:
Sounds like your goal is bi-directional communication between nodes. You can use either
- ACK payloads (see [AcknowledgementPayloads.ino](https://github.com/nRF24/RF24/blob/master/examples/AcknowledgementPayloads/AcknowledgementPayloads.ino) example)
- manually acknowledged payloads (see [ManualAcknowledgements.ino](https://github.com/nRF24/RF24/blob/master/examples/ManualAcknowledgements/ManualAcknowledgements.ino) example).

Which approach you choose depends on the app's timing of data exchange.
1. ACK payloads are more reliable for beginners, but the ACK payload has to loaded into the TX FIFO _before_ a payload is received.
2. Manual acknowledgements allow for flexibility of timed transmissions between nodes and allows using latest payload data in either direction. This approach can be tricky because the nodes must be changed between RX & TX mode in unison. Using timed intervals for RX/TX mode is the most common approach.

2bndy5, 2023-03-16:
Wait I think I read that last paragraph wrong.

If all you want is to send multiple variables in 1 direction, then that is much easier. You could create a buffer that has copies of the multiple objects' data, then send the buffer. It might be easier to use a struct though.

```cpp
// TX side
struct SensorReadings {
    int reading1;
    int reading2;
}
myData = SensorReadings();
myData.reading1 = 0; // data from sensor 1
myData.reading2 = 1; // data from sensor 2
bool ok = radio.write(&myData, sizeof(myData));
```

```cpp
// RX side
struct SensorReadings {
    int reading1;
    int reading2;
}
myData = SensorReadings();
radio.read(&myData, sizeof(myData));
Serial.print(myData.reading1); // data from sensor 1
Serial.print(myData.reading2); // data from sensor 2
```

Miba786, 2023-03-16:
> Wait I think I read that last paragraph wrong.
> 
> If all you want is to send multiple variables in 1 direction, then that is much easier. You could create a buffer that has copies of the multiple objects' data, then send the buffer. It might be easier to use a struct though.
> 
> ```c++
> // TX side
> struct SensorReadings {
>     int reading1;
>     int reading2;
> }
> myData = SensorReadings();
> myData.reading1 = 0; // data from sensor 1
> myData.reading2 = 1; // data from sensor 2
> bool ok = radio.write(&myData, sizeof(myData));
> ```
> 
> ```c++
> // RX side
> struct SensorReadings {
>     int reading1;
>     int reading2;
> }
> myData = SensorReadings();
> radio.read(&myData, sizeof(myData));
> Serial.print(myData.reading1); // data from sensor 1
> Serial.print(myData.reading2); // data from sensor 2
> ```

Thanks for the prompt response, will try it out and update if it works.

2bndy5, 2023-03-16:
My pseudo code was incorrect about declaring the `myData` obj; I updated my suggestion.

Miba786, 2023-03-19:
> ```c++
> Serial.print(myData.reading1); // data from sensor 1
> Serial.print(myData.reading2); // data from sensor 2
> ```

Tried it out, it worked. Thanks!

2bndy5, 2023-03-19:
Great! I'm closing this now that the question is answered. 

In the future, for anything not directly related to the RF24 source code, you can use [the Arduino forums](https://forum.arduino.cc/). I know this question seemed like RF24 related, but it was really a basic understanding of using C++ pointers to a data structure (or buffer).

closed 2023-03-19 by 2bndy5