nRF24/RF24 #903: [OrangePi Zero] Chip enable not working on GPIO10 + workaround

fertinator, 2023-03-27:
I am using the gettingstarted.cpp to test the library. I use the following declaration:
RF24 radio(10, 10, 1000000).
So chip enable on GPIO10, Spi on 1.0 as 0.0 is used by flash. And speed for testing, I could probably ramp it up.
For a few days struggling to find out why the gettingstarted program failed with a "hardware fault". With a logic analyzer I saw the SPI is transmitting OK, but the CE pin never got high. 
It seems the following code is prohibiting the CE pin to go high:
RF24.cpp from line 103
`void RF24::ce(bool level)
{
    //Allow for 3-pin use on ATTiny
    if (ce_pin != csn_pin) {
        digitalWrite(ce_pin, level);
    }
}`
Because I use the 10,10 declaration it read as if both pins are the same. Except the Orange Pi Zero CS pin is GPIO13. See pinout: [https://linux-sunxi.org/Xunlong_Orange_Pi_Zero#Expansion_Port](url)
I got it to work by just commenting the if statement and only leaving the digitalWrite there. 
It might be a possible bug, but depends on the platform and gpio used

2bndy5, 2023-03-27:
Thank you for testing and reporting this! I don't have a OrangePi, so I can't reproduce this. But your diagnosis is sound.
https://github.com/nRF24/RF24/blob/6ad390fee19bd65b252cf8f18effc62b8996c151/RF24.cpp#L103-L109
We should wrap the conditional statement with `#ifndef RF24_LINUX`:
```cpp
void RF24::ce(bool level) 
{ 
#ifndef RF24_LINUX
    //Allow for 3-pin use on ATTiny 
    if (ce_pin != csn_pin) { 
#endif
        digitalWrite(ce_pin, level); 
#ifndef RF24_LINUX
    } 
#endif
} 
```

@fertinator I assume you built/installed RF24 lib with the SPIDEV driver. Is that correct?

fertinator, 2023-03-27:
Correct. I built it with the SPIDEV driver

2bndy5, 2023-03-27:
@fertinator If you could test this fix to make sure it works on your OrangePi, we can move forward with this fix (a PR would be welcome if the fix works).

commit 0d101ae2 (2023-08-28): [OrangePi Zero] Linux chip enable conditional statement (#904)

resolves #903