nRF24/RF24 #905: Does not read information on Raspberry Pi 3B

Fatonndev, 2023-04-29:
**Describe the bug**
I am trying to connect an identical nRF24 to a Raspberry Pi 3B, but it does not receive or send data. I tried to choose the following configurations, and it didn't work on any of them.

`./configure --driver=RPi`
`./configure --driver=SPIDEV`
`./configure --driver=wiringPi`

At the same time, the wires, the power supply and the module itself have already been rechecked several times, connected to esp32, this module worked there.

SPI enabled.

**Additional context**
Log on Raspberry Pi 3B:
```bash
pi@pi:~/temp_ff/RF24/examples_linux $ sudo ./scanner
./scanner================ SPI Configuration ================
CSN Pin         = /dev/spidev0.0
CE Pin          = Custom GPIO22
SPI Speedz      = 10 Mhz
================ NRF Configuration ================
STATUS          = 0x0f RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1    = 0x67786d7838 0x67786d7839
RX_ADDR_P2-5    = 0x3b 0x3b 0x3e 0x3f
TX_ADDR         = 0xf7f7f7f7f7
RX_PW_P0-6      = 0x30 0x30 0x30 0x30 0x30 0x30
EN_AA           = 0x00
EN_RXADDR       = 0x03
RF_CH           = 0x6e
RF_SETUP        = 0x07
CONFIG          = 0x0f
DYNPD/FEATURE   = 0x00 0x00
Data Rate       = 1 MBPS
Model           = nRF24L01+
CRC Length      = 16 bits
PA Power        = PA_MAX
ARC             = 0
000000000000000011111111111111112222222222222222333333333333333344444444444444445555555555555555666666666666666677777777777777
0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcd
------------------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------------------
```

Log on esp32:
```bash
SPI Speedz       = 10 Mhz
STATUS          = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1    = 0x4550495030 0xc2c2c2c2c2
RX_ADDR_P2-5    = 0xc3 0xc4 0xc5 0xc6
TX_ADDR         = 0x4550495030
RX_PW_P0-6      = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA           = 0x00
EN_RXADDR       = 0x02
RF_CH           = 0x4c
RF_SETUP        = 0x07
CONFIG          = 0x0f
DYNPD/FEATURE   = 0x00 0x00
Data Rate       = 1 MBPS
Model           = nRF24L01+
CRC Length      = 16 bits
PA Power        = PA_MAX
ARC             = 15
00000000000000001111111111111111222222222222222233333333333333334444444444444444555555555555555566666666666666667777777777777777
0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef
210024531001123653154bbd6142334cca2223444444223321566414344267222214533203111100100000000000000000000000000000000000000000000000
23442133223457643557668d0fa31447646636114353222334431313135543211101111353233441211000000000000000000000000000000000000000000000
00256541111123535314a3533342abce543000123210479a83364444444455543364101013553620000000000000000000000000000000000000000000000000
20013333112345253423acddb884323123cb23223322111101222356420111134322232233220011111000000000000000000000000000000000000000000000
000111122213355235332525338acd64642123314210124445321111224213333302334574432012021000000000000000000000000000000000000000000000
11446332331212011475e178b31325444bcb11444546343452242334364845463313431134111465310000000000000000000000000000000000000000000000
1112222221111113521321125ef98220111347435566535444400011225354352221013211122454200000000000000000000000000000000000000000000000
0012684542100033132508344242235cdcb866857541211445413220012213331124210222422121000000000000000000000000000000000000000000000000
133342654223465443335a6cdf333222124566643225231122223211121101422121111132375233320000000000000000000000000000000000000000000000
```

TMRh20, 2023-04-29:
First off, this probably isn't a software issue, so we generally would refer you to somewhere else, however...

1. What kind of RF24 module are you using? The PA+LNA module with antenna or the smaller ones?
2. How are you powering it? Do you have a power adapter connecting to 5V and converting to 3.3? This is recommended on the RPi.
3. How long are your connecting wires? They need to be short as possible on RPi.
4. After switching between RF24 RPi drivers, you typically need to either reboot or unload/reload the SPI kernel module.
5. I recommend the SPIDEV driver for RPI, since you don't need to run as root as long as your user has SPI access. Use the BCM driver if you have specific issues with SPIDEV.
6. Have you run the gettingstarted example? It typically will tell you what's wrong by failing at radio begin etc.

Overall it looks like you have some sort of SPI communication happening, but something is wrong with the physical connections, power supply etc because its not looking quite how it should.

Fatonndev, 2023-04-29:
1) I use NRF24L01+PA+LNA
2) I tried in three ways: directly from raspberry, using a 5v->3.3v power adapter, from an external 3.3v source
3) 15 centimeters, with this length of wires, the module worked on esp32
4) Changed the RPi to SPIDEV, after rebooting the same as it was
6) 
```
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
Transmission failed or timed out
```

TMRh20, 2023-04-29:
Ahh, OK for testing purposes with the PA + LNA modules you should call `radio.setPALevel(RF24_PA_MIN,0);` right after calling `radio.begin();` , to ensure you are using the least amount of power to start. Stick with an external 3.3v supply or the power adapter.
You can also lower the SPI speed to like 1,000,000 in the constructor `RF24 radio(22,0,1000000);`
Second do you have capacitors connected to the radio? Preferably 10-100uf and smaller 100nf caps directly to the GND + VCC.
If you look closely at the chip, is it labeled as nrf24l01 at least? Not Si24r1 etc?
15cm wires are a bit long...

Fatonndev, 2023-04-29:
OK, I put a 47uF and 220pF capacitors, changed the code as you suggested, and it didn't do anything :(  Chip is labeled as nrf24l01
[image]

Fatonndev, 2023-04-29:
The multimeter shows a voltage of 3.31V when the scanner is running

TMRh20, 2023-04-29:
Are you sure you have the CE/CS pins wired correctly? Its getting past 'begin()` so that means your SCK, MISO & MOSI are correct, but you may have CE pin off. It defaults to GPIO 22 which is pin 15 if I remember correctly.

Fatonndev, 2023-04-29:
I connected RF24 module according to this scheme
[image]

TMRh20, 2023-04-29:
Try `RF24 radio(25,0,1000000);`

Fatonndev, 2023-04-29:
It works, thank you so much, sorry for wasting your time

closed 2023-04-29 by Fatonndev

TMRh20, 2023-04-29:
Np, I've been having some odd behavior on my RPi3 B with SPIDEV so let us know if you have any problems with it that are resolved by switching to the BCM/RPi driver. That's why I was curious here :p