nRF24/RF24 #909: [Question] Mega not transmitting

FlailAway, 2023-06-28:
Weird stuff going on. I have an IoT setup that has been running for years. The base is a Mega as is one other node that needed the memory to cope with encrypted weather-data stream. The rest are Nano. All have PA-NLA-radios. Megas use CE/CSN 18/53 and 9/53 - if that matters.

Recently the 2-Mega intermittently stopped doing Tx. It slowly got worse over the past week or so and now they do not Tx at all. They Rx OK. I ran a scan and the channel I was using (90-dec) was close to some scatter so I moved them to 110-dec and updated all Nano-Nodes. No difference. I have an new Mega that I plugged the radio shields into and same, no Tx, but Rx OK.

The shields are made from a KeysStudio Mega-proto board (https://wiki.keyestudio.com/images/thumb/9/9f/Ks0005.png/500px-Ks0005.png) and have a 3.3v regulator and Sanyo 3300uF smoothing cap to power the nrf24L01. Power to the board shows 5.01vdc and never over 0.28A. I tested both shields on a new Mega, so probably not the shields. 

I have tried 5 different PA-NLA-radios still no Tx but Rx is good.

Any suggestions?

2bndy5, 2023-06-28:
Did you try replacing the capacitors with new ones? Like batteries, capacitors aren't immortal.

You didn't mention what version of the RF24* libs you are using. I do appreciate all the detail you did include.

Generally speaking, if a PA/LNA module stops TXing, then it is likely a power supply failure.

FlailAway, 2023-06-29:
Thanks for the thoughts. I could not find a version number in the Libraries. I checked the text files and a couple of .h files but none of then had the version or date -- that I could find. 

Moot though as I DLd the latest libraries and recompiled the offending 2 Mega but still nothing. Both Tx failed, both Rx are good.

I will swap the caps, but not holding out great hope as the power is pretty clean without them. Both Mega Nodes also have external Vin of 5vdc and a 3.3vdc 2-Amp Regulator with the Caps, so not a power supply problem. I stopped plugging them in when I discovered years back the the standard Arduino 5vdc rail works fine. Been like that for years. 

The power supply for each Node is a small Power Brick with Active Pass-though. Works like a UPS. It took me a long while to find one that had actual Live Pass-through, but they were only about 10-bucks a piece. I bought a **lot** and have tried powering direct and also with several other Power banks. No change.

2bndy5, 2023-06-29:
Does it help to lower the PA level?

FlailAway, 2023-07-01:
Thanks again but no. I had tried that at all Levels. I a coding error to force a fault and the Arduino IDE (1.1.19) seems to be using some older files as it should not compile -- but did.

OK, Library properties makes sense. NO idea why I didn't look in there. :)

2bndy5, 2023-07-01:
I still think its a hardware problem of some kind. This library intentionally doesn't do anything to prevent TX operations, rather, it just assumes that the radio is in working order and acts accordingly.

> I a coding error to force a fault and the Arduino IDE (1.1.19) seems to be using some older files as it should not compile -- but did.

This is a bit troubling. Have you tried the newer Arduino IDE v2.x? The Arduino team painstakingly tried to make the new IDE's behavior/experience as seamless as possible.

> Since it is a full rewrite ... go to something with embedded radio

That's always an option. Didn't know you did a complete rewrite. IMHO, its hard to get the same power efficiency from an embedded radio (assuming you mean WiFi or Bluetooth). You don't mention using batteries as a power source, so maybe power efficiency isn't a big concern for you.

Please feel free to close this issue if you're moving on to alternatives.

FlailAway, 2023-07-10:
Thanks for the comments. After a few days messing with ESP... I am back to why I originally chose the nrf24L01 radios, range, reliability and consistency.

I suspect my current issue might be more with the Node Numbering. Because this is a full re-write of the system I put Node numbers in a  ore logical fashion relative to physical location.

All nodes are mains-powered but with small UPS to cover the almost daily brief-outages. 

I briefly tried version 2 Arduino IDE, but like Platform-IO it needs to call home for whatever reasons and we do not have always-on Internet so I dropped back to 1.8.19. 

I may be over-thinking the Octal Node numbering system. [This image](https://nrf24.github.io/RF24Network/md_docs_tuning.html) has a combination of Decimal and Octal (no leading 0) numbers.

What would be more helpful, well, to me at a least would be a text display of all all actual Octal number dependents of a single Node 01.

I have 4 main Nodes, 01, 02, 03, 04 in range of the Base then a bunch of outliers in many directions.

Taking 01, it has 1 Child, that Child then has 3 Nodes in a garage, one of which has a further Child in a barn that then has 4 Child Nodes to cover a fairly long distance apart (out in the boonies -- stuff in barns etc). In the old system that is working fine but the Numbering is a mess from adding Nodes ad-hoc over the years. The physical numbering goes like this

* 01 --> 1...
*      --> 2... --> 1... --> 1...
*                                   2...
*                                   3...
*                                   4...
*     --> 03...

So, what must the Octal numbering look like?

2bndy5, 2023-07-10:
That image is actually using all octal numbers, but some of them don't have the leading 0 prefix (? picture was created before my time with this library). I drew a graph for the python wrapper library (which uses the same C++ code under the hood), so maybe that would be helpful. See [this topology page](https://pyrf24.readthedocs.io/en/latest/topology.html) but keep in mind that it uses python octal notation (`0o` prefix instead of the `0` prefix used in C++).

FlailAway, 2023-07-14:
Thanks, I finally figured out the problem with the Mega. For some reason it (IDE ?? compiler??) sets the Channel at 0 or 255 but only on the 2-Mega. I am using a single .h header file for all nodes and that has the Channel number defined so that everything is n the same channel. There are only 2-Mega the rest are Nano. 

I have gone back to much-much earlier Library and it is not happening. I will dig out the actual version for you as I just stumbled on to the Channel thing and now time to quit. Thanks for the topology map, Naming turns out to not to be the issue, but image looks better than the old one.

TMRh20, 2024-02-18:
Please update if there is further information on this, closing the issue.

closed 2024-02-18 by TMRh20