nRF24/RF24 #916: so i was trying to check if my PA/LNA nRF24l01 was working or not, but I seem to be having some problems

DevvratBani, 2023-08-14:
the code i used is this one:-

/*
  If your serial output has these values same then Your nrf24l01 module is in working condition :
  
  EN_AA          = 0x3f
  EN_RXADDR      = 0x02
  RF_CH          = 0x4c
  RF_SETUP       = 0x03
  CONFIG         = 0x0f
  
  This code is under public domain
  
  Last updated on 21/08/28 
  https://dhirajkushwaha.com/elekkrypt
 */

#include <SPI.h>
#include <RF24.h>
#include <printf.h>

RF24 radio(9, 10);

byte addresses[][6] = {"1Node", "2Node"};

void setup() {
  radio.begin();
  radio.setPALevel(RF24_PA_LOW);
  
  radio.openWritingPipe(addresses[0]);
  radio.openReadingPipe(1, addresses[1]); 
  radio.startListening();
  
  Serial.begin(9600);
  printf_begin();

  radio.printDetails();
  
}

void loop() {
//  empty

}

i keep getting weird outputs like :- 

STATUS		= 0xff RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1	= 0xffffffffff 0xffffffffff
RX_ADDR_P2-5	= 0xff 0xff 0xff 0xff
TX_ADDR		= 0xffffffffff
RX_PW_P0-6	= 0xff 0x7f 0xff 0x7f 0xff 0x7f
EN_AA		= 0xff
EN_RXADDR	= 0x7f
RF_CH		= 0xff
RF_SETUP	= 0x7f
CONFIG		= 0x7f
DYNPD/FEATURE	= 0x7f 0xff
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_MAX
ARC		= 15

            OR 

STATUS		= 0xff RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1	= 0xffffffffff 0xffffffffff
RX_ADDR_P2-5	= 0xff 0xff 0xff 0xff
TX_ADDR		= 0xffffffffff
RX_PW_P0-6	= 0xff 0xff 0xff 0xff 0xff 0x7f
EN_AA		= 0xff
EN_RXADDR	= 0xff
RF_CH		= 0xff
RF_SETUP	= 0xff
CONFIG		= 0xff
DYNPD/FEATURE	= 0xff 0xff
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_MAX
ARC		= 15
SPI Speedz	= 10 Mhz 

how do i fix this issue, i keep getting these weird values

2bndy5, 2023-08-14:
The true indicator of "working" behavior would be to check the return value of `radio.begin()` (as is done in all library examples).
https://github.com/nRF24/RF24/blob/a028813b285d6cec3090d2c96984bd67a399c52e/examples/GettingStarted/GettingStarted.ino#L41-L50

What MCU are you using? Does [decreasing the SPI speed help](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#payloads-receivedsent-are-inaccurate-or-printdetails-outputs-the-unexpected-values)?

DevvratBani, 2023-08-14:
eh, im actually quite new to using nRF modules, how do i change the SPI speed? also shouldn't they be pre-calibrated?

2bndy5, 2023-08-14:
Follow the link I posted (or simply look at our [COMMON_ISSUES.md](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#payloads-receivedsent-are-inaccurate-or-printdetails-outputs-the-unexpected-values)).

DevvratBani, 2023-08-14:
> Follow the link I posted (or simply look at our [COMMON_ISSUES.md](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#payloads-receivedsent-are-inaccurate-or-printdetails-outputs-the-unexpected-values)).

ill try and post the updates

2bndy5, 2023-08-14:
> shouldn't they be pre-calibrated?

There is no way to do that. Technically, your environment (temperature, humidity, physical obstacles, etc) is too unpredictable.

DevvratBani, 2023-08-14:
hmm, i cahnged the SPI speed to 4MHz, still the same response, 

SPI Speedz	= 4 Mhz
STATUS		= 0xff RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1	= 0xffffffffff 0xffffffffff
RX_ADDR_P2-5	= 0xff 0xff 0xff 0xff
TX_ADDR		= 0xffffffffff
RX_PW_P0-6	= 0xff 0xff 0xff 0xff 0xff 0xff
EN_AA		= 0xff
EN_RXADDR	= 0xff
RF_CH		= 0xff
RF_SETUP	= 0xff
CONFIG		= 0xff
DYNPD/FEATURE	= 0xff 0xff
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_MAX
ARC		= 15

do you think it could be a connection issue or a power issue??

2bndy5, 2023-08-14:
1. When printDetails() shows incorrect values, that usually indicates a wiring problem.
2. When a PA/LNA module fails to transmit (including auto-ack packets), that usually indicates a power supply problem.

Both of these answers are explained in more detail in our [COMMON_ISSUES.md doc](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md).

DevvratBani, 2023-08-14:
hmm, so i guess its a power supply problem since i tried changing the SPI value , maybe the USB port is not giving enough power, or my vcc and gnd connections are weak

2bndy5, 2023-08-14:
Double check your wires first. Once you get `radio.begin()` to return true, then we can start looking at your power supply.

DevvratBani, 2023-08-14:
[image]
wirings are all good

2bndy5, 2023-08-14:
What I meant was check that your Arduino pins are correctly connected to the appropriate radio pins. What you're seeing usually means one (or more) of SPI wires (CSN, MISO, MOSI, SCK) isn't going to the right radio pin. I can't tell from the picture you posted.

DevvratBani, 2023-08-14:
> What I meant was check that your Arduino pins are correctly connected to the appropriate radio pins. What you're seeing usually means one (or more) of SPI wires (CSN, MISO, MOSI, SCK) isn't going to the right radio pin. I can't tell from the picture you posted.

CE = dig 9
CSN = dig 10
SCK = dig 13
MOSI = dig 11
MISO = dig 12
VCC = 3.3v
Gnd = Gnd 

also, i did change the CE and CSN values to 9 and 10 in the code when i was running it, so no problem there. i was still getting similar results when the pins were 7 and 8 respectively.

DevvratBani, 2023-08-14:
this time i held the pins on the arduino side firmly while uploading, and i got this 

SPI Speedz	= 4 Mhz
STATUS		= 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1	= 0x0000000000 0x0000000000
RX_ADDR_P2-5	= 0x00 0x00 0x00 0x00
TX_ADDR		= 0x0000000000
RX_PW_P0-6	= 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		= 0x00
EN_RXADDR	= 0x00
RF_CH		= 0x00
RF_SETUP	= 0x00
CONFIG		= 0x00
DYNPD/FEATURE	= 0x00 0x00
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= Disabled
PA Power	= PA_MIN
ARC		= 0
 now the PA power went MIN...

2bndy5, 2023-08-14:
try using different set of wires (shorter if possible). Holding the connections firmly shouldn't affect anything.

DevvratBani, 2023-08-14:
> try using different set of wires (shorter if possible). Holding the connections firmly shouldn't affect anything.

yes i was just watching a video about this, i think i need shorter wires and also i need to turn the Gnd wire about the MOSI and MISO wires

2bndy5, 2023-08-14:
Just to be very clear, the pinout for an UNO board is included in our doc's [pinout table](https://nrf24.github.io/RF24/#autotoc_md45).

DevvratBani, 2023-08-14:
Meanwhile, i have another question i was actually thinking of using the nRF module for communication between my remote and a mini-tank project (my latest repository, UGV). should i use LoRA module instead, its just a mini-project and the LoRA module seems more robust. you can view the code if you want, im basically sending a data array from remote to the UGV

2bndy5, 2023-08-14:
This isn't the [Arduino Forums](https://forum.arduino.cc/) here. We tend to only focus on library problems here. Helping people troubleshoot sometimes reveals library problems, but this issue seems like that isn't the case.

DevvratBani, 2023-08-14:
could it be that my module is just delivery damaged??

2bndy5, 2023-08-14:
It depends on packaging. All transistor-based electronics are highly susceptible to static electricity, so if it came in just bubble wraps (and no type of electromagnetic blocking pouch), then they could have been damaged in transport due to ambient static electricity.

2bndy5, 2023-08-14:
We have also seen users fry their radios by accidentally routing 5v to the radio's 3v VCC pin. There is no way to recover from that.

2bndy5, 2023-08-17:
I'm sorry to hear that. Officially, per Nordic Semiconductors, "the nRF24L01 is no longer recommended for new designs". Since they open-sourced the module's PCB design, the market has been flooded with cheaply manufactured clones (beware of the Si24R1 👎🏼 ). Still it is hard to beat the power efficiency, price, and features of the nRF24L01.

If you have a couple nRF52840-based boards sitting around, then you can use their builtin radio with the [nrf_to_nrf Arduino library](https://github.com/TMRh20/nrf_to_nrf) for OTA compatibility with nRF24L01 (sacrificing BLE capability).

DevvratBani, 2023-08-20:
alright, so i got the new nRF module and this one is definately better than the last one, its giving out values like this:-

��00000001c 0x000�SPI Speedz	= 10 Mhz
STATUS		= 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1	= 0x000000001c 0x000000001c
RX_ADDR_P2-5	= 0x38 0x0e 0x38 0x38
TX_ADDR		= 0x6c6c6c6cec
RX_PW_P0-6	= 0x0e 0x1c 0x1c 0x38 0x0e 0x38
EN_AA		= 0x3f
EN_RXADDR	= 0x1c
RF_CH		= 0x38
RF_SETUP	= 0x1c
CONFIG		= 0x0f
DYNPD/FEATURE	= 0x70 0xe0
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_HIGH
ARC		= 0

the values seem to be changing with each upload, but nothing as serious as 0x00 or 0xff , so is this better, will this module work??

2bndy5, 2023-08-20:
The output looks right. You could check if it works by running one of our lib examples; typically you would begin with [GettingStarted.ino](https://github.com/nRF24/RF24/blob/master/examples/GettingStarted/GettingStarted.ino).

> **Note**
> Our examples use a lower PA level (`RF24_PA_LOW`) to avoid power problems with PA/LNA modules.

DevvratBani, 2023-08-25:
got the exact output while using an older version of the IDE(1.8.57) (idk why?) 
but yeah it was a problem caused by a shady dealer.  thanks for everyone who gave this query attention

closed 2023-08-25 by DevvratBani