nRF24/RF24 #917: RF24 for UNO R4?

bambuino, 2023-08-22:
My bloaty bloatware no longer runs well on an UNO R3 (AVR) as NODE #00.

So, I got an UNO R4 (Cortex M4) WiFi today. For some reason I thought my NODE #00 code had compiled in the Arduino IDE with the UNO R4 WiFi as the target board. Now, hmmm...does not work.

Any suggestions or leads to get this resolved? Thank you.

2bndy5, 2023-08-22:
Arduino IDE doesn't check for dependency versions... Make sure you have updated RF24Network and RF24Mesh.

If that doesn't help, then we'll need more info to help you. Specifically whatever errors/warnings that the Arduino IDE is showing.

bambuino, 2023-08-22:
I have RF24 1.3.2, Network 2.0.0 and Mesh 2.0.0

Am using 1.3.2 for my ATtiny84 nodes because 1.4.7 robs about 600 bytes from program storage. But 1.3.2 throws plenty errors with the NODE #00 code on the UNO R4.

However, the NODE #00 code compiled with 1.4.7 throws but one error on an AVR thing, avr/sleep.h.

```cpp
Alternatives for avr/sleep.h: []
ResolveLibrary(avr/sleep.h)
  -> candidates: []
c:\Users\mkapo\Arduino\libraries\RF24Network\RF24Network.cpp:42:14: fatal error: avr/sleep.h: No such file or directory
     #include <avr/sleep.h>
              ^~~~~~~~~~~~~
compilation terminated.
Multiple libraries were found for "SD.h"
  Used: C:\Users\mkapo\Arduino\libraries\SD
  Not used: C:\Users\mkapo\AppData\Local\Arduino15\libraries\SD
Using library Wire in folder: C:\Users\mkapo\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\libraries\Wire (legacy)
Using library Adafruit RGB LCD Shield Library at version 1.2.0 in folder: C:\Users\mkapo\Arduino\libraries\Adafruit_RGB_LCD_Shield_Library 
Using library RTClib at version 2.1.1 in folder: C:\Users\mkapo\Arduino\libraries\RTClib 
Using library Adafruit BusIO at version 1.14.3 in folder: C:\Users\mkapo\Arduino\libraries\Adafruit_BusIO 
Using library RF24Network at version 2.0.0 in folder: C:\Users\mkapo\Arduino\libraries\RF24Network 
Using library RF24 at version 1.4.7 in folder: C:\Users\mkapo\Arduino\libraries\RF24 
Using library SPI in folder: C:\Users\mkapo\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\libraries\SPI (legacy)
Using library SD at version 1.2.4 in folder: C:\Users\mkapo\Arduino\libraries\SD 
exit status 1

Compilation error: exit status 1
```
This NODE #00 does not sleep, just collects up data from the nodes. I commented out the avr/sleep.h and power.h in the RF24Network.cpp around line 42, after which the fatal error above was not thrown, but this breaks the library for all the other nodes.
```
#if defined(ENABLE_SLEEP_MODE) && !defined(RF24_LINUX) && !defined(__ARDUINO_X86__)
    #include <avr/sleep.h>
    #include <avr/power.h>
volatile byte sleep_cycles_remaining;
volatile bool wasInterrupted;
#endif
```

Is there a way to specifically "define" this UNO R4 board so that those #includes are not used?

Thanks folks for taking a read of this.

2bndy5, 2023-08-22:
A snippet from the [RF24Network_config.h docs](https://nrf24.github.io/RF24Network/md_docs_advanced_config.html):
> `#define ENABLE_SLEEP_MODE` -- Uncomment this option to enable sleep mode for AVR devices. (ATTiny,Uno, etc)

So, just make sure that it is commented out (like [it is when the lib is freshly installed](https://github.com/nRF24/RF24Network/blob/a334fa6ec420b2beedaf617018c11847cea80402/RF24Network_config.h#L93-L95))

---
I wasn't aware you are using an ATTiny84. What Arduino core are you using to compile the code for you ATTiny84? Officially, we only support the [SpenceKonde ATTinyCore](https://github.com/SpenceKonde/ATTinyCore) (which is noted in the RF24 ATTiny examples and [the RF24/ATTiny doc](https://nrf24.github.io/RF24/md_docs_attiny.html))

2bndy5, 2023-08-22:
Looks like the UNO R4 is rather newly supported board in the Arduino ecosystem (as evident by the [ArduinoCore-renesas tagged releases history](https://github.com/arduino/ArduinoCore-renesas/tags)). Additionally, the Renesas-based boards seem to be supported by PlatformIO (see [PIO docs](https://docs.platformio.org/en/latest//boards/index.html#renesas-ra)) which provides a better solution (compiler `-D` [build_flags](https://docs.platformio.org/en/latest//projectconf/sections/env/options/build/build_flags.html#build-flags)) than modifying RF24Network_config.h by hand for different board builds (which was an intentional design decision by the Arduino IDE devs).

bambuino, 2023-08-22:
Commenting out the two lines as mentioned above, in RF24Network.cpp, got this running on the UNO R4.

I had commented out the line in RF24Network_config.h as I thought that would enable both a "watchdog" timer and the interrupt from the RF24 with `network.sleepNode()` on the ATtiny84. It seems to work on the Nano V3 router nodes so far.

So by commenting back in the `#define ENABLE_SLEEP_MODE`, I will not have to comment out the two lines in RF24Network.cpp, is that right? Then I won't have `network.sleepNode()` available for the AVRs?

I am using the ATTinyCore, per your folks recommendation.

2bndy5, 2023-08-22:
This may sound repetitive, but I want to be very clear.
The `#include <avr/****.h>` lines should not need to be commented out if `#define ENABLE_SLEEP_MODE` is commented out. The recommended way to control that feature is through RF24Network_config.h (using `ENABLE_SLEEP_MODE` macro definition).

> So by commenting back in the #define ENABLE_SLEEP_MODE, I will not have to comment out the two lines in RF24Network.cpp, is that right? Then I won't have network.sleepNode() available for the AVRs?

If you un-comment the `#define ENABLE_SLEEP_MODE` and leave the `#include <avr/****.h>` lines commented out, then the feature will not work (probably won't even compile). Please restore the `#include <avr/****.h>` lines as they were (not commented) and use `ENABLE_SLEEP_MODE` to turn on (un-commented) or turn off (commented) that feature.

# Using PlatformIO simplifies things
As I hinted before, you can better use PlatformIO to build your project for different boards using the project's platformio.ini file. This means you can leave `ENABLE_SLEEP_MODE` commented out in RF24Networek_config.h and control it's definition through different platformio.ini env's [`build_flags`](https://docs.platformio.org/en/latest//projectconf/sections/env/options/build/build_flags.html) option. You can also control what version of RF24 to use for different boards using the [`lib_deps`](https://docs.platformio.org/en/latest//projectconf/sections/env/options/library/lib_deps.html#lib-deps) option.
```ini
# the env for your UNO R4
[env:uno_r4_minima]
platform = renesas-ra
board = uno_r4_minima
# or for the UNO R4 WiFi
# board=uno_r4_wifi
lib_deps=
  nRF24/RF24==1.4.7
  nRF24/RF24Network>=2.0.0
  nRF24/RF24Mesh>=2.0.0
  # ...

# the env for your ATTiny board
[env:attiny84]
platform = atmelavr
board = attiny84
lib_deps=
  nRF24/RF24==1.3.7
  nRF24/RF24Network>=2.0.0
  nRF24/RF24Mesh>=2.0.0
  # ...
# enable sleep feature in RF24Network
build_flags =
  -DENABLE_SLEEP_MODE
```

## Success?
It sounds like your problem here is solved, yes? Should we close this issue?

closed 2023-10-25 by 2bndy5

BugProg, 2023-10-28:
Hi !

I have some problems with my Arduino uno r4. 
I have an existing project that uses two Arduino boards (Leonardo and Mega). The two boards communicate perfectly with the RF24Network lib. 

I recently replaced the Arduino leonardo with an Arduino UNO r4 minima. The code is the same, but the data transmitted to the Arduino mega seems to be corrupted.  Do you think this could be coming from my arduino uno? 
One point that could be interesting is that I've noticed that the values returned by `sizeof` don't seem to be the same depending on whether it's an Arduino Leonardo or Uno r4. 

Could the problem come from this line : `network.write(header, &telemetryData, sizeof(Telemetry));`

Adrien

2bndy5, 2023-10-28:
@BugProg Your problem seems different from the issue originally reported (related to using sleep function in RF24Network). It sounds like the UNO R4 uses differently sized memory allocation for certain data types, but I don't know what data types the `Telemetry` struct holds. Since this problem is rather common between some Arduino boards and Linux, we recommend you ensure the data you are sending is aligned to 32 bits. Ultimately, this isn't a RF24 problem, so you can probably get better advice elsewhere.

BugProg, 2023-10-29:
Hi @2bndy5  !

Thank you for your message ! Yes, indeed, it was not the right discussion for that.
I was able to resolve partially the issue by replacing `int` by `uint8_t`, I'm steel investigating for `float` number.