nRF24/RF24 #919: SPIException Exception issue

eduardoeberhardt15, 2023-09-19:
I'm using the orange pi zero 3 with python 3.11 with wiringOp and wiringOp-Python installed and i'm trying to running this rf24 lib

[image]
This is my pinout

spidev1.1 and spidev1.0 are enabled 

The csn rf24 pinout it's on physic 24 pin
The ce rf24 pinout it's on physic 3 pin like is GPIO2

the radio config i'm try already RF24 radio(2, 11), radio (2,10)
trying also many others ce pinouts but always got the same error like

terminate called after throwing an instance of 'SPIException'
  what():  can't open device
Aborted

[image]

If i change csn to 0 i got the GPIOException instead.

Could you guys help me please?

eduardoeberhardt15, 2023-09-19:
> Have you tried running with sudo permission?

Yes, you can see on the image that i ran the sudo ./gettingstarted

2bndy5, 2023-09-19:
I should note that wiringPi support is deprecated because the original library is no longer maintained. The version of wiringPi that you are using in the OrangePi is a third-party port of the original library (& not officially supported by the RF24 library).

How did you install the RF24 library? Did you select `WiringPi` as your driver? I think the [`SPIException` is coming from the `SPIDEV` driver](https://github.com/nRF24/RF24/blob/eb790f98371f31cfb4b6987f080d5d9d1556554f/utility/SPIDEV/spi.cpp#L54).

2bndy5, 2023-09-19:
> If i change csn to 0 i got the GPIOException instead.

Again, this means you are using the `SPIDEV` driver because that's [where the `GPIOException` exception is coming from](https://github.com/nRF24/RF24/blob/master/utility/SPIDEV/gpio.cpp). But I think the proper CE pin number (according to your chart in OP & using `SPIDEV`) should be 229, right?

eduardoeberhardt15, 2023-09-19:
Should i remove the wiringPi?
i ran sudo ./configure --driver=SPIDEV, that is, i'm not running with wiringpi, but also i already test with all drivers and nothing work
I run this code to test the spi.
[image]

eduardoeberhardt15, 2023-09-19:
> Again, this means you are using the `SPIDEV` driver because that's [where the `GPIOException` exception is coming from](https://github.com/nRF24/RF24/blob/master/utility/SPIDEV/gpio.cpp). But I [...]

I had tried all kid of number

2bndy5, 2023-09-19:
I'm not sure where you got the `spidev_test` program... But I think that test requires a jumper from MOSI to MISO as it is likely a loop-back test.

eduardoeberhardt15, 2023-09-19:
> I'm not sure where you got the `spidev_test` program... But I think that test requires a jumper from MOSI to MISO as it is likely a loop-back test.

Yes, needs a jumper.
I used the orangepi doc above.
https://drive.google.com/file/d/1nqr1LWQKs2SN5GZzboN5SLwoNZHkSJ7b/view

2bndy5, 2023-09-19:
And that was the output with the jumper attached (not the nRF24L01)?

eduardoeberhardt15, 2023-09-19:
> And that was the output with the jumper attached (not the nRF24L01)?

Not, with jumper, rx is same tx.

eduardoeberhardt15, 2023-09-19:
> Just so I understand, the loop back test works with the jumper? Your English is hard to understand.

You, with the jumper work normal.

2bndy5, 2023-09-19:
As far as I can tell, it should work with `RF24 radio(229, 11)` using the `SPIDEV` driver. I don't know anybody that has an OPi3 to try or test.

eduardoeberhardt15, 2023-09-19:
> BTW, you don't need `sudo` permission when using the `SPIDEV` driver.
> 
> > Should i remove the wiringPi?
> 
> It should make no difference if you remove WiringPi because the compiler shouldn't be linking into that library if you configured RF24 to use `SPIDEV`.

Now I started from scratch again. I deleted the folder, cloned it in git and ran the commands in the attached image and it gave an access error.
[image]

2bndy5, 2023-09-19:
> CPU=aarch64

If you are using a 64-bit OS, then you should be [installing RF24 with CMake](https://nrf24.github.io/RF24/md_docs_using_cmake.html) instead. The older `configure` script is meant to prepare builds with `armhf` compilers, and it has not aged well with 64-bit environments.

2bndy5, 2023-09-19:
I see you are using the automated install script, and it shows that there is already a rf24libs folder with contents. It is best to start from a clean slate
You can delete the rf24libs folder and start again or install it manually from source:
```shell
# navigate to the RF24 library source
cd RF24

# reset the git repository
git reset --hard

# remove old build artifacts
rm Makefile.inc utility/includes.h

# uninstall previous RF24 installation
sudo rm /usr/local/lib/librf24.*
sudo rm /usr/local/lib/librf24-bcm.so
sudo rm -r /usr/local/include/RF24

# now install using cmake
mkdir build
cd build
cmake .. -DRF24_DRIVER=SPIDEV
make
sudo make install
```
Building the examples with CMake is similar to building the library
```shell
cd RF24/examples_linux
mkdir build
cd build
cmake .. -DRF24_DRIVER=SPIDEV
make

# run the gettingstarted example
./gettingstarted
```

eduardoeberhardt15, 2023-09-19:
> I see you are using the automated install script, and it shows that there is already a rf24libs folder with contents. It is best to start from a clean slate You can delete the rf24libs folder and [...]

Seems the same problem.

orangepi@orangepizero3:/RF24/examples_linux$ sudo nano gettingstarted.cpp
orangepi@orangepizero3:/RF24/examples_linux$ cd build
orangepi@orangepizero3:/RF24/examples_linux/build$ sudo cmake .. -DRF24_DRIVER=SPIDEV
-- detected SoC: sun50i-h616
-- detected CPU type: aarch64
-- tool name being used: aarch64-linux-gnu
-- Using driver: SPIDEV
-- using RF24 library: /usr/local/lib/librf24.so
-- Configuring done
-- Generating done
-- Build files have been written to: /RF24/examples_linux/build
orangepi@orangepizero3:/RF24/examples_linux/build$ sudo make
Consolidate compiler generated dependencies of target gettingstarted
[  7%] Building CXX object CMakeFiles/gettingstarted.dir/gettingstarted.cpp.o
[ 14%] Linking CXX executable gettingstarted
[ 14%] Built target gettingstarted
Consolidate compiler generated dependencies of target acknowledgementPayloads
[ 28%] Built target acknowledgementPayloads
Consolidate compiler generated dependencies of target manualAcknowledgements
[ 42%] Built target manualAcknowledgements
Consolidate compiler generated dependencies of target streamingData
[ 57%] Built target streamingData
Consolidate compiler generated dependencies of target multiceiverDemo
[ 71%] Built target multiceiverDemo
Consolidate compiler generated dependencies of target scanner
[ 85%] Built target scanner
Consolidate compiler generated dependencies of target interruptConfigure
[100%] Built target interruptConfigure
orangepi@orangepizero3:/RF24/examples_linux/build$ sudo ./gettingstarted
terminate called after throwing an instance of 'GPIOException'
  what():  can't access /sys/class/gpio/gpio%d/direction GPIO pin. check access rights
Aborted
orangepi@orangepizero3:/RF24/examples_linux/build$  ./gettingstarted
terminate called after throwing an instance of 'SPIException'
  what():  can't open device
Aborted (core dumped)

2bndy5, 2023-09-19:
Have you added your user to the `dialout` group? That is required for `SPIDEV` driver to have access to `/dev/spidevX.Y`.

```shell
usermod -a -G dialout $USER
```

Also, I think you have to add your user to whatever group the GPIO pins are registered to. I don't remember exactly how to do this; off the top of my head I think I had to install a udev rule from RPi.GPIO sources in Ubuntu on my RPi4.

> **Note**
> You can get the groups that are associated with your user by running `groups`

2bndy5, 2023-09-19:
A reboot is required for changes to the user groups to take effect.

2bndy5, 2023-09-19:
What I see on my RPi4:
```shell
brendan@rpi4-ubuntu-64bit:~$ groups
brendan adm dialout cdrom sudo dip plugdev lpadmin lxd sambashare
brendan@rpi4-ubuntu-64bit:~$ ls /dev/spi* -l
crw-rw---- 1 root dialout 153, 0 Mar 20  2023 /dev/spidev0.0
crw-rw---- 1 root dialout 153, 1 Mar 20  2023 /dev/spidev0.1
brendan@rpi4-ubuntu-64bit:~$ ls /dev/gpio* -l
crw-rw---- 1 root dialout 254, 0 Mar 20  2023 /dev/gpiochip0
crw-rw---- 1 root dialout 254, 1 Mar 20  2023 /dev/gpiochip1
crw-rw---- 1 root dialout 504, 0 Mar 20  2023 /dev/gpiomem
```

I don't remember what else I did to get non-root access to GPIO, but it wasn't as easy as adding a group to my user's groups. I think this is an [ongoing and unsolved issue with OrangePi](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/222).

eduardoeberhardt15, 2023-09-19:
> What I see on my RPi4:
>
> ```shell
> brendan@rpi4-ubuntu-64bit:~$ groups
> brendan adm dialout cdrom sudo dip plugdev lpadmin lxd sambashare
> brendan@rpi4-ubuntu-64bit:~$ ls /dev/spi* -l
> crw-rw---- 1 root [...]

[image]

2bndy5, 2023-09-19:
What OS are you using?

This issue is beginning to seem like a problem with OrangePi3 and less of a problem with RF24.

eduardoeberhardt15, 2023-09-19:
> What OS are you using?
> 
> This issue is beginning to seem like a problem with OrangePi3 and less of a problem with RF24.

My OS is Ubuntu
The RF24 is working on espcam32 and arduino UNO.

2bndy5, 2023-09-19:
> The RF24 is working on espcam32 and arduino UNO.

That just tells me the radio is in working condition.

I'm not sure what to tell you at this point. I don't have an OPi to play with, and I probably never will. Personally, I try to avoid using Chinese-cloned hardware.

TMRh20, 2023-09-20:
Well we use [PiGPIO](https://abyz.me.uk/rpi/pigpio/) for GPIO with SPIDEV so I would start looking at getting that working/testing that, as that seems to be where your problem is. As per @2bndy5 this is likely not an RF24 issue, but we don't have the hardware to confirm.

eduardoeberhardt15, 2023-09-20:
I think that i'll use arduino.
Thanks by trying help me.

closed 2023-09-20 by 2bndy5

mirhamza708, 2024-12-27:
Just posting here for future reference. I was testing this library with Orange Pi 3 LTS with Ubuntu 22.04.5 LTS, which is the official image provided by Orange Pi. The GPIO exception occurs because the GPIO pin numbering is not defined properly in there ubuntu image. Also in there user manual they say to use wiringOP to handle gpios while it can be used I was unable to use that with RF24. The below image shows the pin map of the OPi 3 LTS.
[image]
The only pins I was able to control using the libgpiod (used by SPIDEV) were the ones in PORT L. The following commands can be used to turn the gpio PL08 turn on or off.
```
gpioset gpiochip0 8=1
```
or 
```
gpioset gpiochip0 8=1
```
I don't know how the pin naming works here but I randomly set the pin high and then checked with multimeter the level on each pin in the 26 pin header on OPi 3 LTS. I found out that the pin numbers in the PORT L were directly settable through the gpioset command. for example you can write gpioset gpiochip0 X=1 (where X can be 2, 3, 8, 10 and there are other pins which can be set but I haven't actually tested) and other pins will just return this error code 517.
```
gpioset: error setting the GPIO line values: Unknown error 517
```
which is the same exception from the RF24 getting started program. 

when I ran my getting started program with CE_PIN=8 and CSN_PIN=10. Then the code works just fine. for other pins it will result in error. 

Remember you have add the following line to the /boot/orangepiEnv.txt file
to open the file use 
```
sudo nano /boot/orangepiEnv.txt
```
then at the end add this line 
```
overlays=spi-spidev1, gpio
```
ctrl+x then press y and then press Enter to save the file  
a reboot is required 
```
sudo reboot
```

then first test the gpio in bash and if all works good then you can run the gettingstarted program. I was logged in with root so all the permissions were enabled maybe you need to give permission for you user. It worked for me hopefully it will work for someone else out there.

Another method which can be used to control all the pins is deprecated but the orange pi still uses it. which is the 'sysfs' to use this method you have to use the following commands
First export the pin that you want to control so the sysfs will consume this pin and it may not be used by any other modules during this time with
```
echo 117 > /sys/class/gpio/export # this will export the pin
``` 
you can check the exported pins with 
```
ls /sys/class/gpio/
```
Then set the direction of the exported pin
```
echo "out" > /sys/class/gpio/gpio117/direction  # this will make the pin as OUTPUT pin you can use "in" for INPUT pin
```
Then set the pin level
```
echo 1 > /sys/class/gpio/gpio117/value # this will make it HIGH level 3.3V
echo 0 > /sys/class/gpio/gpio117/value # this will make it LOW level 0V
```
then you need to unexport so its free to use by other programs
```
echo 117 > /sys/class/gpio/unexport
```

This method can control all the pins. but for this the RF24 library needs to be updated so the SPIDEV uses the older sysfs for some of the orange pis. I may do it some time but currently I don't have time. I guess the gpio.cpp and gpio.h under the SPIDEV needs to be edited to use sysfs instead of the libgpiod. @2bndy5 @TMRh20

2bndy5, 2024-12-27:
RF24 used the old sysfs interface for years, but it had to switch to character device interface to support the RPi5.

The sysfs interface is deprecated in favor of the character device interface, anyway. If you need the old sysfs interface, just checkout v1.4.8 as we switched over in v1.4.9.

mirhamza708, 2024-12-27:
hmm will check that thanks.