nRF24/RF24 #920: cannot find -lboost_python3

DIY89, 2023-09-22:
Hi

i've tried to get the RF24 libary running on raspi3B+ with bullseye.
I cloned `https://github.com/TMRh20/RF24` and run through the next steps
```
cd ~/rf24libs/RF24
./configure
sudo make install
sudo apt-get install python3-dev libboost-python-dev
sudo apt-get install python3-setuptools
```
But when i try to build the code it fails
```
cd pyRF24/
python3 setup.py build
sudo python3 setup.py install
```
`arm-linux-gnueabihf-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -Wl,-z,relro -g -fwrapv -O2 -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -L/usr/local/lib -I/usr/local/include -DRF24_NO_INTERRUPT -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.9/pyRF24.o -lrf24 -lboost_python3 -o build/lib.linux-armv7l-3.9/RF24.cpython-39-arm-linux-gnueabihf.so
/usr/bin/ld: cannot find -lboost_python3
collect2: error: ld returned 1 exit status
error: command '/usr/bin/arm-linux-gnueabihf-g++' failed with exit code 1
`

I checked old issues but still can't fix it

- https://github.com/nRF24/RF24Network/issues/135
- https://github.com/nRF24/RF24/issues/615

Found this entries here but don't know if they are correct
```

-rw-r--r--  1 root root   497420 13. Mär 2021  libboost_python39.a
lrwxrwxrwx  1 root root       27 22. Sep 10:46 libboost_python39.so -> libboost_python39.so.1.74.0
```

Does anybody have an idea

TMRh20, 2023-09-22:
Did u follow all the instructions at https://nrf24.github.io/RF24/md_docs_python_wrapper.html  ?

`sudo apt-get install python3-dev libboost-python-dev python3-pip python3-rpi.gpio`

`python3 -m pip install --upgrade pip setuptools`

`sudo ln -s $(ls /usr/lib/$(ls /usr/lib/gcc | tail -1)/libboost_python3*.so | tail -1) /usr/lib/$(ls /usr/lib/gcc | tail -1)/libboost_python3.so`

DIY89, 2023-09-22:
Worked perfectly. Thank you

closed 2023-09-22 by DIY89

2bndy5, 2023-09-22:
The pyrf24 package available on pypi is meant to replace the individual library wrappers (and avoid build problems like this one). You could just
```
pip install pyrf24
```
and change the import statements as [shown in the pyRF24 README](https://github.com/nRF24/pyRF24/#migrating-to-pyrf24).

> **Warning**
> Using the individual library wrappers with the newer pyrf24 package is not advised and may not work. Either use (at runtime) the pyrf24 package or the individual library wrappers, not both.