nRF24/RF24 #921: Calling from a C++ class causes weird behaviour on Linux

LollosoSi, 2023-10-12:
Information about this issue is available in 2 threads on the arduino forum
https://forum.arduino.cc/t/rf24-inconsistencies-on-linux-needs-investigation/1177591
https://forum.arduino.cc/t/transferring-data-via-nrf24l01-between-arduino-nano-and-raspberry-pi/1107524

**Description**
Data sent or received when caller is not in the main file / is a class are inaccurate or random

Full description along with project files, pictures of the data and code walkthrough is included in the **first** link

**Additional context**
Using the same project but moving the radio related functions to the main file transmits data as expected as long as there aren't calls from classes instances (which breaks the project entirely)

2bndy5, 2023-10-12:
Was there a reason that this was closed? 

Have you considered byte alignment may be different between devices?

The getting started example uses floats to show that data types aren't always represented the same on each platform.

2bndy5, 2023-10-12:
Looking at what you posted in Arduino forums, I think you mis-used the pointer to the data being sent when calling `radio.write()` from a class function.

```cpp
radiopacket* radiohandler::send(radiopacket *rp) {

	// no need to reference a pointer when passing data to write()
  // &rp is like passing a pointer to a pointer
	if (radio->writeFast(rp, 32)) { // just pass the pointer
```
This is the difference in your code (that you posted in Arduino forums) when you "move" the code into man.cpp.

LollosoSi, 2023-10-13:
sorry I misclicked. anyways yes we spotted the same issue. not a bug

closed 2023-10-13 by LollosoSi