nRF24/RF24 #923: nRF24L01+ gets recognized as nRF24L01 after software reboot

CommanderRedYT, 2023-10-19:
**Describe the bug**
I have a ESP32 connected to a nRF24L01+ module. When cold-booting (so after complete power cycle), the module gets registered as such. After some kind of software reset of the esp32 (OTA Update, esp_restart(), ...), the module gets recognized as a nRF24L01 without the "+".

I do not think that this is a hardware issue tho.

**Additional context**
This library is used in setup with https://github.com/tbnobody/OpenDTU. I already opened a issue there, but they could not help me (https://github.com/tbnobody/OpenDTU/issues/1221). After looking into their code, it seems that its basically just calling the isPVariant() function.

2bndy5, 2023-10-19:
What version  of RF24 library are you using?

CommanderRedYT, 2023-10-19:
nrf24/RF24 @ ^1.4.8 (platformio)

CommanderRedYT, 2023-10-19:
[image]
This is the module. The begin function should be called correctly. The weird thing is that it only detects it wrong if its a software reboot of the esp, but **never** when I power-cycle it (unplug esp32, wait a few secs, and then plug it back in again)

CommanderRedYT, 2023-10-19:
Also, it still functions like normal when it gets detected as the "+" and without the "+"

2bndy5, 2023-10-19:
Looks like you likely bought a bad clone (probably a RFM7x). The blob of black goo on the board is a huge indicator (& users should avoid purchasing these modules at all cost). The fact that it still functions as expected is extremely lucky. This is indeed a hardware issue that we can't solve on our end.

> **Note**
> We don't specifically support the RFM7x clones because the proper setup process is wildly different from the original nRF24L01 chips (and would cause compile size to increase significantly).

CommanderRedYT, 2023-10-19:
Oh okay, yea I was also bit surprised why all SMD support-components were missing, but that is a explanation for that. A friend of mine gave me one of them, but I guess, as long as it works, it works. Thank you!

closed 2023-10-19 by CommanderRedYT