nRF24/RF24 #924: STM32F411 doesn't receive data

iGerchak, 2023-10-21:
Hi All,
My hardware:
- NRF24L01+PA+LNA
- Arduino Nano
- STM32F411CE (black pill)

I tried to communicate between Arduino Nano and STM32F411CE (black pill) and had an issue with the communication.
In general, Arduino should be TX and STM32 RX.
When I try to send data to the STM32, always get a fail and STM32 doesn't receive any data.
After that, I connected another Arduino Nano instead of STM32 with the same code to check if NRF24L01 was working well, and as a result, it worked well.
So, after debugging for a few days, I found that if switch STM32 to TX and Arduino to RX, Arduino gets data, but on the STM32 payload is not available.

I tried different configs, pin connections, examples from the lib, power supplier, SPI speed, etc. but can't fix it.

Here are the details:
STM32:

```
================ SPI Configuration ================
CSN Pin                 = 196
CE Pin                  = 200
SPI Frequency           = 10 Mhz
================ NRF Configuration ================
Channel                 = 96 (~ 2496 MHz)
RF Data Rate            = 1 MBPS
RF Power Amplifier      = PA_MAX
RF Low Noise Amplifier  = Enabled
CRC Length              = 16 bits
Address Length          = 5 bytes
Static Payload Length   = 32 bytes
Auto Retry Delay        = 250 microseconds
Auto Retry Attempts     = 15 maximum
Packets lost on
    current channel     = 0
Retry attempts made for
    last transmission   = 0
Multicast               = Disabled
Custom ACK Payload      = Enabled
Dynamic Payloads        = Enabled
Auto Acknowledgment     = Enabled
Primary Mode            = RX
TX address              = 0xE7E7E7E7E7
Pipe 0 (closed) bound   = 0xE7E7E7E7E7
Pipe 1 ( open ) bound   = 0x65646F4E31
Pipe 2 (closed) bound   = 0xC3
Pipe 3 (closed) bound   = 0xC4
Pipe 4 (closed) bound   = 0xC5
Pipe 5 (closed) bound   = 0xC6
```

Arduino:
```
SPI Frequency           = 10 Mhz
Channel                 = 96 (~ 2496 MHz)
Model                   = nRF24L01+
RF Data Rate            = 1 MBPS
RF Power Amplifier      = PA_MAX
RF Low Noise Amplifier  = Enabled
CRC Length              = 16 bits
Address Length          = 5 bytes
Static Payload Length   = 32 bytes
Auto Retry Delay        = 250 microseconds
Auto Retry Attempts     = 15 maximum
Packets lost on
    current channel     = 0
Retry attempts made for
    last transmission   = 15
Multicast               = Disabled
Custom ACK Payload      = Enabled
Dynamic Payloads        = Enabled
Auto Acknowledgment     = Enabled
Primary Mode            = TX
TX address              = 0x65646f4e31
pipe 0 ( open ) bound   = 0x65646f4e31
pipe 1 ( open ) bound   = 0xc2c2c2c2c2
pipe 2 (closed) bound   = 0xc3
pipe 3 (closed) bound   = 0xc4
pipe 4 (closed) bound   = 0xc5
pipe 5 (closed) bound   = 0xc6
```

Code of STM32:
```
#include <Arduino.h>
#include <Wire.h>
#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"

//------------------ Radio ----------------------------
#define MOSI PA7
#define MISO PA6
#define SCK PA5
RF24 radio(PB0, PA4); // CE, CSN
byte address[][6] = { "1Node", "2Node", "3Node", "4Node", "5Node", "6Node" };
//------------------ END Radio ------------------------

void setup() {
    SPI.begin();
    SPI.setMOSI(MOSI);
    SPI.setMISO(MISO);
    SPI.setSCLK(SCK);
    Serial.begin(115200);
    Wire.begin();
    while (!Serial);

    if (!radio.begin()) {
        Serial.println(F("radio hardware is not responding!!"));
        while (1) {}  // hold in infinite loop
    }

    radio.begin();
    radio.setAutoAck(1);
    radio.setRetries(0, 15);
    radio.enableAckPayload();
    radio.setPayloadSize(32);
    radio.openReadingPipe(1, address[0]);
    radio.setChannel(0x60);
    radio.setPALevel(RF24_PA_MAX);
    radio.setDataRate(RF24_1MBPS);
    radio.powerUp();
    radio.startListening();
    char buffer[870] = { '\0' };
    uint16_t used_chars = radio.sprintfPrettyDetails(buffer);
    Serial.println(buffer);
    Serial.print(F("strlen = "));
    Serial.println(used_chars + 1); // +1 for c-strings' null terminating byte
}

void radioCommunication() {
    byte pipeNo, gotByte;

    while (radio.available(&pipeNo)) {
        radio.read(&gotByte, 1);
        radio.writeAckPayload(pipeNo, &gotByte, 1);
        Serial.print("Recieved: ");
        Serial.println(gotByte);
    }
}

void loop() {
    radioCommunication();
}
```

2bndy5, 2023-10-21:
> ```cpp
>     radio.begin();
>     radio.setAutoAck(1);
>     radio.setRetries(0, 15);
>     radio.enableAckPayload();
>     radio.setPayloadSize(32);
> ```

A few problems here:
1. `radio.begin()` doesn't need to be called twice.
2. Care should be taken when setting the auto-retry delay (`0` means 250 microseconds). The delay specified here is the amount of time that the radio waits for auto-ACK packets. The datasheet recommends using 500 microseconds (`1` means `250 + (250 * 1)` = 500 ) or more when using ACK payloads (of at least 5 bytes).
3. Using ACK payloads requires enabling the dynamic payloads feature. Internally this is done for pipes 0 and 1 when `enableAckPayload()` is called, but just beware other pipes won't be usable unless the `enableDynamicPayloads()` is invoked.
4. Building off of point 3, `setPayloadSize()` (for statically sized payloads) is useless when using dynamic payloads feature.

> I tried different configs, pin connections, examples from the lib, power supplier, SPI speed, etc. but can't fix it.

I highly suspect this is a power supply problem due to the use of PA/LNA modules. If you tried the library examples (specifically the ACK payloads example) and it still failed on your STM32, then I think this is a hardware problem (not a software problem). Did you read through our [COMMON_ISSUES.md](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md)?

iGerchak, 2023-10-22:
@2bndy5 Thank you for the quick answer.
Some bugs in the code can be as I modified it a lot of times during debugging.

Yes, I tried the ACK payloads example, here are the results:
[image]

As you can see in the screenshot, the payload on the STM32 isn't available.
And after switching a role, Arduino can't send data.

> Did you read through our [COMMON_ISSUES.md](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md)?
Yes, a few times) 
But I haven't tried to use a capacitor before, because I use a power adapter like this:
[image]

So, will try to add a capacitor.

2bndy5, 2023-10-22:
Just make sure the adapter board's VCC pin is getting at least a _stable_ 150 mA (at 5 V of course).

iGerchak, 2023-10-22:
I connected an additional power supply to the STM32 board, and now it works!!!!
So, power from USB isn't enough for NRF24L01+PA+LNA.
@2bndy5  thank you so much!

closed 2023-10-22 by iGerchak