nRF24/RF24 #930: 'make' errors, and RF24 on RPi, aarch64-linux-gnu

Givy, 2023-12-02:
I can't get around this issue on RPi 4 and RPi 3, both ARM64 OS. Connecting NRF24L01+ to the GPIO pins using any of the packages and libraries, /rf24libs/RF24, Raspberry-Pi-NRF24-master, RF24Network-master, etc.

#Linux RaXXXX 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux
#Linux MiXXXX 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux

/rf24libs/RF24# make
arm-linux-gnueabihf-g++ -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread
arm-linux-gnueabihf-g++ -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread
arm-linux-gnueabihf-gcc -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread
arm-linux-gnueabihf-gcc -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthrea
arm-linux-gnueabihf-g++ -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread
[Linking]
arm-linux-gnueabihf-gcc -shared -Wl,-soname,librf24.so.1 -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi.o spi.o bcm2835.o compatibility.o interrupt.o -lpigpio
**_/lib/libpigpio.so: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
make: *** [Makefile:46: librf24.so.1.4.8] Error 1_**

Thank you in advance.
Givi

More info::

g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/aarch64-linux-gnu/10/lto-wrapper
Target: aarch64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 10.2.1-6' --with-bugurl=file:///usr/share/doc/gcc-10/REAjc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-10 --program-prefix=aarch64-linux-gnu- --enr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocime=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libquadmath --disable-libquadmath-supportlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --enablcking=release --build=aarch64-linux-gnu --host=aarch64-linux-gnu --target=aarch64-linux-gnu --with-build-config=bootstr
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.1 20210110 (Debian 10.2.1-6)
g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/aarch64-linux-gnu/12/lto-wrapper
Target: aarch64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 12.2.0-14' --with-bugurl=file:///usr/share/obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-12 --program-prefix=aarch64-linux-without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enaxx-abi=new --enable-gnu-unique-object --disable-libquadmath --disable-libquadmath-support --enable-plugin lease --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --enable-fix-cortex-a53-8434u --host=aarch64-linux-gnu --target=aarch64-linux-gnu
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.2.0 (Debian 12.2.0-14)

2bndy5, 2023-12-03:
The `configure` script was really designed with the 32 bit armhf compiler in mind. We moved to cmake for several reasons, one reason was to support 64 bit aarch64 compiler better. Please use the [newer cmake instructions](https://nrf24.github.io/RF24/md_docs_using_cmake.html).

2bndy5, 2023-12-03:
I'm curious why pigpio was being linked to if the driver was not found. This shouldn't be a problem using cmake, but there may be something else we haven't coded for in the `configure` script.

2bndy5, 2023-12-04:
I was able to reproduce the build error for pyRF24 when using the `RPi` driver. I should have a fix committed soon.

@Givy Please specify the `SPIDEV` driver when install.sh asks. Using `RPi` driver on a 64 bit OS may not work as expected. Also the SPIDEV driver is much faster and doesn't require `sudo` permission to run examples.

Givy, 2023-12-05:
I sure will,  At the moment I am doing much better than before, and getting to know what's what.  Thank you again.  ;-]

2bndy5, 2023-12-05:
Ok, I'm closing this as we've worked through the problems presented. Just to summarize:
- CMake install instructions _must_ be used on 64-bit OS (also recommended for cross-compiling)
- Found and fixed a problem when building pyRF24 package from source code.

@Givy If you want to build the pyRF24 package from source then you must either
- update the local git clone:
  ```
  cd ~/rf24libs/pyRF24
  git pull
  ```
- just delete the pyRF24 folder
 
and then run the install.sh again.

closed 2023-12-05 by 2bndy5

Givy, 2023-12-05:
RPi-3 is working and "scanner" runs OK.  However, on RP-4 I have problem with the "pip install pyrf24", which I am sure I did that earlier at the same time with the RP-3.  I am gettinig the below issues on the RPi-4::
--------- 
:~/rf24libs# pip install pyrf24
error: externally-managed-environment

This environment is externally managed

To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
-------------------------------
[image]

2bndy5, 2023-12-05:
> If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.
>
> For more information visit http://rptl.io/venv

This is the solution you seek. It is generally good practice to use a python virtual environment ("venv" for short). You should not use `sudo` when working in a python (& pip) because it invokes the system python install instead of the venv python install.

----
The problem you are experiencing is actually not specific to pyRF24 at all. But I will try to clarify as simply as possible.

# 🚫 DO NOT DO THIS!!!
```shell
sudo pip install pyrf24
sudo python ~/rf24libs/pyRF24/examples/scanner.py
```
The above commands will only use python (interpreter and packages) installed for the entire system. 👎🏼 

# ✔️ DO THIS!
```shell
pip install pyrf24
python ~/rf24libs/pyRF24/examples/scanner.py
``` 
The above commands will use the packages specifically installed for the current user.

# ⭐ Most recommended!
1. Create a venv in the current directory
    ```
    python -m venv env
    ```
2. Activate the python venv
    ```
    source env/bin/activate
    ```
    If you want to deactivate a python venv (typically done for using different packages for different projects), then you can use the command `deactivate`. 
3. Install pyrf24 (while venv is active)
    ```
    pip install pyrf24
    ```
4. Run the examples (while venv is active)
    ```
    python ~/rf24libs/pyRF24/examples/scanner.py
    ```

> [!IMPORTANT]
> Using the `RPi` driver requires sudo permission to run the examples (or any code using RF24 with RPi driver employed). If you need to use RPi driver, then you must specify the venv install of python (while venv is active):
> ```
> sudo env/bin/python ~/rf24libs/pyRF24/examples/scanner.py
> ```

2bndy5, 2023-12-05:
Installing pyrf24 with pip uses the SPIDEV driver and does not require the C++ RF24* libraries installed. Again, SPIDEV is recommended for speed, and it does not require sudo permission to execute examples (or any code using RF24 with SPIDEV driver employed).

Givy, 2023-12-05:
I also tried the following and got these errors::

# apt install python pyRF24
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python-is-python3 2to3

E: Package 'python' has no installation candidate
E: Unable to locate package pyRF24
-------------

2bndy5, 2023-12-05:
I'm sorry, but you clearly are not very familiar with using Debian's `apt` program or python CLI. I cannot help you there because it is not my job (& I'm not getting paid for this). No offense intended.

Givy, 2023-12-05:
I appreciate your help.  I am not sure about the "venv ", but I guess I have to try it. Thank you so much.

2bndy5, 2023-12-07:
I recently ran into a similar problem with using `pip install` as a non-root user... Apparently, python's pip package recently implemented a "feature" proposed by PEP668. PEP668 basically states that pip should be run _only_ from within a venv by end-users (see [the full proposal](https://peps.python.org/pep-0668/) for justified rationality). This is a new restriction rolled out sometime in 2023 and most notably has affected Linux workflows.

This is what happens when python has been greatly adopted over 20+ years despite the many difficulties that python package management had from the beginning.

Givy, 2023-12-25:
Just FYI:
After more reading and additional pkg/s and re-install.  Your automated shell script worked on the RPi-3.  Will let you know on the RPi-4::

[100%] Built target rf24
/usr/local/lib/python3.9/dist-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
installing to build/bdist.linux-aarch64/wheel
running install
running install_lib
creating build/bdist.linux-aarch64
creating build/bdist.linux-aarch64/wheel
creating build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-39/pyrf24/__init__.py -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-39/pyrf24/rf24_network.cpython-39-aarch64-linux-gnu.so -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-39/pyrf24/rf24_network.pyi -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-39/pyrf24/rf24_mesh.pyi -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-39/pyrf24/py.typed -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-39/pyrf24/rf24_mesh.cpython-39-aarch64-linux-gnu.so -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-39/pyrf24/fake_ble.py -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-39/pyrf24/rf24.pyi -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-39/pyrf24/rf24.cpython-39-aarch64-linux-gnu.so -> build/bdist.linux-aarch64/wheel/pyrf24
running install_egg_info
Copying pyrf24.egg-info to build/bdist.linux-aarch64/wheel/pyrf24-0.2.4.post1.dev3-py3.9.egg-info
running install_scripts
adding license file "LICENSE" (matched pattern "LICEN[CS]E*")
creating build/bdist.linux-aarch64/wheel/pyrf24-0.2.4.post1.dev3.dist-info/WHEEL
creating 'dist/pyrf24-0.2.4.post1.dev3-cp39-cp39-linux_aarch64.whl' and adding 'build/bdist.linux-aarch64/wheel' to it
adding 'pyrf24/__init__.py'
adding 'pyrf24/fake_ble.py'
adding 'pyrf24/py.typed'
adding 'pyrf24/rf24.cpython-39-aarch64-linux-gnu.so'
adding 'pyrf24/rf24.pyi'
adding 'pyrf24/rf24_mesh.cpython-39-aarch64-linux-gnu.so'
adding 'pyrf24/rf24_mesh.pyi'
adding 'pyrf24/rf24_network.cpython-39-aarch64-linux-gnu.so'
adding 'pyrf24/rf24_network.pyi'
adding 'pyrf24-0.2.4.post1.dev3.dist-info/LICENSE'
adding 'pyrf24-0.2.4.post1.dev3.dist-info/METADATA'
adding 'pyrf24-0.2.4.post1.dev3.dist-info/WHEEL'
adding 'pyrf24-0.2.4.post1.dev3.dist-info/top_level.txt'
adding 'pyrf24-0.2.4.post1.dev3.dist-info/RECORD'
removing build/bdist.linux-aarch64/wheel
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./dist/pyrf24-0.2.4.post1.dev3-cp39-cp39-linux_aarch64.whl
Installing collected packages: pyrf24
  Attempting uninstall: pyrf24
    Found existing installation: pyrf24 0.2.4
    Uninstalling pyrf24-0.2.4:
      Successfully uninstalled pyrf24-0.2.4
Successfully installed pyrf24-0.2.4.post1.dev3

*** Installer Complete ***
See http://tmrh20.github.io for documentation
See http://tmrh20.blogspot.com for info

Listing repositories in ~/rf24libs
pyRF24  RF24  RF24Gateway  RF24Mesh  RF24Network

Thank you.

Givy, 2023-12-25:
Hi,
Your automated shell script also worked on the RPi-4 with some complaining::

-- Configuring done
-- Generating done
-- Build files have been written to: /root/rf24libs/pyRF24/build/temp.linux-aarch64-cpython-311/pyrf24
[  4%] Building CXX object CMakeFiles/rf24.dir/src/pyRF24.cpp.o
[  9%] Building CXX object CMakeFiles/rf24.dir/RF24/RF24.cpp.o
[ 14%] Building CXX object CMakeFiles/rf24.dir/RF24/utility/SPIDEV/gpio.cpp.o
[ 19%] Building CXX object CMakeFiles/rf24.dir/RF24/utility/SPIDEV/spi.cpp.o
[ 23%] Building CXX object CMakeFiles/rf24.dir/RF24/utility/SPIDEV/compatibility.cpp.o
[ 28%] Linking CXX shared module /root/rf24libs/pyRF24/build/lib.linux-aarch64-cpython-311/pyrf24/rf24.cpython-311-aarch64-linux-gnu.so
lto-wrapper: warning: using serial compilation of 6 LTRANS jobs
lto-wrapper: note: see the '-flto' option documentation for more information
[ 28%] Built target rf24
[ 33%] Building CXX object CMakeFiles/rf24_network.dir/src/pyRF24Network.cpp.o
[ 38%] Building CXX object CMakeFiles/rf24_network.dir/RF24/RF24.cpp.o
[ 42%] Building CXX object CMakeFiles/rf24_network.dir/RF24/utility/SPIDEV/gpio.cpp.o
[ 47%] Building CXX object CMakeFiles/rf24_network.dir/RF24/utility/SPIDEV/spi.cpp.o
[ 52%] Building CXX object CMakeFiles/rf24_network.dir/RF24/utility/SPIDEV/compatibility.cpp.o
[ 57%] Building CXX object CMakeFiles/rf24_network.dir/RF24Network/RF24Network.cpp.o
[ 61%] Linking CXX shared module /root/rf24libs/pyRF24/build/lib.linux-aarch64-cpython-311/pyrf24/rf24_network.cpython-311-aarch64-linux-gnu.so
lto-wrapper: warning: using serial compilation of 7 LTRANS jobs
lto-wrapper: note: see the '-flto' option documentation for more information
[ 61%] Built target rf24_network
[ 66%] Building CXX object CMakeFiles/rf24_mesh.dir/src/pyRF24Mesh.cpp.o
[ 71%] Building CXX object CMakeFiles/rf24_mesh.dir/RF24/RF24.cpp.o
[ 76%] Building CXX object CMakeFiles/rf24_mesh.dir/RF24/utility/SPIDEV/gpio.cpp.o
[ 80%] Building CXX object CMakeFiles/rf24_mesh.dir/RF24/utility/SPIDEV/spi.cpp.o
[ 85%] Building CXX object CMakeFiles/rf24_mesh.dir/RF24/utility/SPIDEV/compatibility.cpp.o
[ 90%] Building CXX object CMakeFiles/rf24_mesh.dir/RF24Network/RF24Network.cpp.o
[ 95%] Building CXX object CMakeFiles/rf24_mesh.dir/RF24Mesh/RF24Mesh.cpp.o
[100%] Linking CXX shared module /root/rf24libs/pyRF24/build/lib.linux-aarch64-cpython-311/pyrf24/rf24_mesh.cpython-311-aarch64-linux-gnu.so
lto-wrapper: warning: using serial compilation of 9 LTRANS jobs
lto-wrapper: note: see the '-flto' option documentation for more information
[100%] Built target rf24_mesh
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
installing to build/bdist.linux-aarch64/wheel
running install
running install_lib
creating build/bdist.linux-aarch64
creating build/bdist.linux-aarch64/wheel
creating build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-311/pyrf24/rf24_network.cpython-311-aarch64-linux-gnu.so -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-311/pyrf24/rf24.cpython-311-aarch64-linux-gnu.so -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-311/pyrf24/rf24_network.pyi -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-311/pyrf24/fake_ble.py -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-311/pyrf24/rf24_mesh.cpython-311-aarch64-linux-gnu.so -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-311/pyrf24/rf24_mesh.pyi -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-311/pyrf24/py.typed -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-311/pyrf24/rf24.pyi -> build/bdist.linux-aarch64/wheel/pyrf24
copying build/lib.linux-aarch64-cpython-311/pyrf24/__init__.py -> build/bdist.linux-aarch64/wheel/pyrf24
running install_egg_info
Copying pyrf24.egg-info to build/bdist.linux-aarch64/wheel/pyrf24-0.0.0.egg-info
running install_scripts
creating build/bdist.linux-aarch64/wheel/pyrf24-0.0.0.dist-info/WHEEL
creating 'dist/pyrf24-0.0.0-cp311-cp311-linux_aarch64.whl' and adding 'build/bdist.linux-aarch64/wheel' to it
adding 'pyrf24/__init__.py'
adding 'pyrf24/fake_ble.py'
adding 'pyrf24/py.typed'
adding 'pyrf24/rf24.cpython-311-aarch64-linux-gnu.so'
adding 'pyrf24/rf24.pyi'
adding 'pyrf24/rf24_mesh.cpython-311-aarch64-linux-gnu.so'
adding 'pyrf24/rf24_mesh.pyi'
adding 'pyrf24/rf24_network.cpython-311-aarch64-linux-gnu.so'
adding 'pyrf24/rf24_network.pyi'
adding 'pyrf24-0.0.0.dist-info/LICENSE'
adding 'pyrf24-0.0.0.dist-info/METADATA'
adding 'pyrf24-0.0.0.dist-info/WHEEL'
adding 'pyrf24-0.0.0.dist-info/top_level.txt'
adding 'pyrf24-0.0.0.dist-info/RECORD'
removing build/bdist.linux-aarch64/wheel
error: externally-managed-environment

This environment is externally managed

To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

*** Installer Complete ***
See http://tmrh20.github.io for documentation
See http://tmrh20.blogspot.com for info

Listing repositories in ~/rf24libs
env  pyRF24  RF24  RF24Gateway  RF24Mesh  RF24Network

Thanks.

2bndy5, 2023-12-26:
> This environment is externally managed

Again, that entire complaint is a user problem. Our install script should not manage a python venv. It is better to run the install script in an activated python venv (which would silence that warning).