nRF24/RF24 #931: [Question] NRF24 Transmitter get "stuck" when powered from wall power outlet

mittlc, 2023-12-27:
Hello everybody,
i have a question regarding a NRF24 transmitter which seems to get “stuck” when unplugged from PC and powered from wall power until another nrf24 "blink" sketch is uploaded first and the formerly working sketch is re-uploaded.

I started experimenting 2 nrf24l01+pa+lna modules each connected to a wemos d1 mini esp32.
One acting as a receiver and one acting as the transmitter. The goal is to build a DIY RC Controll.
The transmitter is sending analog readings from a 2 axis joystick to the receiver, which currently is just printing out the received messages to the serial monitor. Everything works fine.

The issue occures when i unplug the transmitter from the pc(on which also the receiver is connected to access the serial monitor) and power it from a wall power supply instead.
My expectation is that the transmitter is still reading the joystick values and sending it to the receiver - but no messages are received by the receiver.

This stays the same even when i am plugging the esp32 transmitter back into the pc and is only reverted by first upload another sketch (e.g. a sample nrf24 sketch wich is just sendig a fixed messages) followed by another re-upload of the exact same sketch formerly working to read and send the joystick analog readings - without any changes to the sketch - then now the receiver starts picking up messages from the transmitter again.

I cannot figure out what is causing the transmitter to getting “stuck” (in need of better characterization).

Any ideas what i can further try to narrow in on the root cause or did anybody experience the same issue before?

Thanks

Working transmitter code until unplugged from the pc:

`#include <Arduino.h>
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

const uint8_t CE_PIN = 12;
const uint8_t CS_PIN = 32;
const int analogPinX = 39;
const int analogPinY = 35;

RF24 radio(CE_PIN, CS_PIN); // Create a Radio

const byte address[6] = "0q0z1";
char text[20] = "";
char start[] = "START";

void setup() {
    //Serial.begin(115200);
    radio.begin();
    radio.openReadingPipe(0, address);
    radio.setPALevel(RF24_PA_MIN);
    radio.setDataRate(RF24_250KBPS);
    radio.stopListening();
    radio.powerUp();
    pinMode(analogPinX, INPUT);
    pinMode(analogPinY, INPUT);
    
    for (int i = 1; i <= 100; i++) {
      radio.write(&start, sizeof(start)); 
      delay(5); 
  }
}

void loop() {
  int sensorValueX = analogRead(analogPinX); 
  int sensorValueY = analogRead(analogPinY); 
  snprintf(text, sizeof(text), "%d,%d", sensorValueX, sensorValueY);
  radio.write(&text, sizeof(text));
  //Serial.print("Message sent: ");
  delay(5);
}
`

Working receiver code until unplugged from the pc:
`#include <Arduino.h>
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
// NRF24L01 pin configuration
const uint8_t CE_PIN = 12;
const uint8_t CS_PIN = 32;

RF24 radio(CE_PIN, CS_PIN); // Create a Radio
const byte address[6] = "0q0z1";
char text[32] = ""; // Buffer to store incoming messages

void setup() {
    Serial.begin(115200);
    radio.begin();
    radio.openReadingPipe(0, address);
    radio.setPALevel(RF24_PA_MIN);
    radio.setDataRate(RF24_250KBPS);
    radio.startListening();
    Serial.println("Setup finished");
    }

void loop() {
    if (radio.available()) {
        radio.read(&text, sizeof(text));
        Serial.print("Message received: ");
        Serial.println(text);
    }
    //delay(5);
}
`

2bndy5, 2023-12-27:
You can use an LED to indicate things without using a serial monitor. This might help you narrow down what part of the code is problematic when using the wall power adapter.

Looking at the code, I see you're sending a c-str instead of 2 `int` values. I would prefer beginners not transmit c-str because it could lead to segmentation faults if the c-str is not null-terminated or mishandled in some way.

The only other thing I can recommend is checking the power supply used at the wall. 5v should be fine, but ESP32 chips by themselves are rather amperage hungry. And nRF24+PA+LNA modules are also well known to _require_ more than 150 mA. So, just make sure the wall adapter is rated for 1500 mA or better; even very old wall adapters can be less efficient than their labels say.

mittlc, 2023-12-28:
Hello [2bndy5],
thanks for your prompt response.

The issue is fixed now and although i cannot offer a satisfying explanation, i want to provide the solution here just in case and to close the issue.

It turned out, that i made a copy&paste error initialising the transmitter radio with
radio.openReadingPipe(0, address);
instead of
radio.openWritingPipe(address);

The sketch to transfer the joystick readings with radio.openReadingPipe(0, address) couldn´t tolerate a restart by unplugging the esp32 and plugging it back in - the receiver then stopps to receive any message. This is the same behavior also on the PC so the issue had nothing to do with the power supply in the first place - sorry for that.

The bare transmitter sketch sending a fixed value used radio.openWritingPipe(address); correctly and does not suffer from this issue.

The reason behind that (and also how i could send messages from the transmitter using radio.openReadingPipe(0, address) in the first places) goes beyond my understanding of the nrf24 so far - but as of now i only want to build a transmitter and receiver and no transceiver module and will not digg into it further.

Again thanks a lot for the prompt response.

closed 2023-12-28 by mittlc