nRF24/RF24 #933: Radio.printDetails(); does nothing for MKR1010 but Radio.sprintfPrettyDetails(buffer); works fine????

bvernham, 2024-01-05:
Please read [about common issues](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md) first. It addresses the most common problems that people have (whether they know it or not).
printf_begin(); is call after Serial is started.  This code worked flawlessly with and RF-Nano but not for the MKR?

I tried all of the debug steps (lower spi clock speed), add cap and so on.

Ideas?

2bndy5, 2024-01-05:
> printf_begin(); is call after Serial is started. This code worked flawlessly with and RF-Nano but not for the MKR?

Our library's `printf_begin()` is just a helper function to integrate the RF24 library's `printf()` calls with the default `Serial` object, but it only does this for boards that can use `printf()`

> Radio.sprintfPrettyDetails(buffer); works fine????

Of course it does, because it doesn't use `printf()`. It uses `sprintf()` (which uses buffers instead of a stream like `printf()` does). The functions [`sprintfPrettyDetails()`](https://nrf24.github.io/RF24/classRF24.html#a03b7e739eba98e47fc92cf74bc6d8be4) and [`encodeRadioDetails()`](https://nrf24.github.io/RF24/classRF24.html#abe54fe7f4b776ca4927dcf8435ca1133) were designed as an alternative way to get debugging information on boards that don't support `printf()`.

I think your confusion is really just from not understanding the difference between [`printf()`](https://cplusplus.com/reference/cstdio/printf/) and [`sprintf()`](https://cplusplus.com/reference/cstdio/snprintf/) functions from the C standard library (`cstdio.h`).

## MKR1010 uses a ATSAMD21 chip
That board uses the [Arduino SAMD core](https://github.com/arduino/ArduinoCore-samd), which does not support `printf()`.

> [!TIP]
> An Adafruit board that uses the ATSAMD21 _can_ use `printf()` because it is supported by [Adafruit's fork of the Arduino SAMD core](https://github.com/adafruit/ArduinoCore-samd).

## RF-Nano uses ATMega328p
I'm guessing that board just uses a fork of the [Arduino AVR core](https://github.com/arduino/ArduinoCore-avr) which does support `printf()`.

bvernham, 2024-01-05:
Has it been considered to use LibPrintf library.

Work fine with MKR1010.

2bndy5, 2024-01-05:
No, we haven't considered using another library _just_ for debugging output on _some_ boards.

LibPrintf has been updated in years. If there were problems, then we (and our users) would have to hope that the LibPrintf authors are still responsive... 👎

If you're using PlatformIO, then it would be easy to make RF24 use LibPrintf. You just need to add a `#define` to the [`build_flags`](https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_flags.html#build-flags).
```ini
build_flags =
  -Dprintf_P=printf
```
Otherwise, if you're using Arduino IDE with a board that doesn't support `printf()`, then you can use `sprintfPrettyDetails()`.

2bndy5, 2024-01-05:
> No, we haven't considered using another library _just_ for debugging output on _some_ boards.

This isn't entirely true. We had started developing [RF24Log](https://github.com/nRF24/RF24Log) but stopped when we couldn't decide on how to control verbosity (logging level) per origin of logs.

closed 2024-01-06 by bvernham