nRF24/RF24 #934: Pipe is sometimes returned at "255".

bvernham, 2024-01-06:
Please read [about common issues](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md) first. It addresses the most common problems that people have (whether they know it or not).

Same as before this only happens with MKR1010, not for either the LGT or AVR based RF-Nano boards.

void ReadData() {
  uint8_t bytes;
  ......  
  if (Radio.available(&pipe)) {                     // is there a payload? get the pipe number that recieved it

Pipe should be 1 but sometimes returns 255 radomly with the MKR1010.

All of the other data look s fine and consistent with what is expected.

Any ideas?

2bndy5, 2024-01-06:
What version of RF24 are you using?

This seems similar to #913 which was supposedly fixed in #914 and released in [v1.4.8](https://github.com/nRF24/RF24/releases/tag/v1.4.8). Versions of RF24 affected were v1.3.10 - v1.4.7

bvernham, 2024-01-06:
Arduino library manager says 2.0.0.
Does anyone have any confirmed "good" boads?  There are so many out there an none of the usual suspects like SparkFun make a break out board so you are left with a ton of differnt boards on ebay/amazon/Alibaba and so on.

2bndy5, 2024-01-06:
> Arduino library manager says 2.0.0

I think you're confusing RF24 library with either RF24Mesh or RF24Network library. There is no v2.0.0 for RF24 library.

> Does anyone have any confirmed "good" boads?

Its pretty much a gamble now. Officially per Nordic Semiconductors, this radio is not recommended for new designs. The module PCB was open sourced in 2014 and the market has been flooded with knock-offs ever since.

bvernham, 2024-01-06:
Sorry 1.4.8.
Just looking for someone whom has had positive experinece from a particular supplier.

TMRh20, 2024-01-06:
All the RF24 modules I've gotten off Amazon have proven functional and are stamped as NRF24L01 which is something you always want to check.
Ebay has been hit and miss, but I've had good luck with the high powered E-Byte modules https://www.ebay.ca/str/cdebyte

This is in no way an endorsement, just my experiences so far.

bvernham, 2024-01-06:
I looked at mainline suppliers like Digikey and so on and I found MIKROE boards and TANGO-24PA modules on digikey.  They are <$20 but obvioulsy more than the usual Amazon/ebay/Alibaba.

bvernham, 2024-01-06:
Do the ebyte modules have the same issue with the SPI pull up resistors which the shielded version do on Amazon does?  The say the shielded ones on Amazon have to slow the spi speed becuase the resistors cause longer RC on the SPI lines.

TMRh20, 2024-01-06:
I'm not sure, I've only gotten the ebyte modules off ebay so far, but I've had no problems running them near full power off a RPi.

Regarding the Pipe issue, are you able to post some working code that produces this issue?

bvernham, 2024-01-06:
`void ReadData() {
  uint8_t bytes;
  uint8_t pipe;
  uint16_t payload = 0;
  if (Radio.available(&pipe)) {                     // is there a payload? get the pipe number that recieved it
    uint8_t bytes = Radio.getDynamicPayloadSize();  // get the size of the payload
    Radio.read(&MQData, bytes);                     // fetch payload from FIFO
    //if (Serial) {
      Serial.print(F("Bytes Recieved = "));
      Serial.print(bytes, DEC);
      Serial.print(F("\tPipe Recieved = "));
      Serial.println(pipe, DEC);
      uint8_t numsensor = (uint8_t)(MQData.encoded & numsensor_mask);
      Serial.print(F("Sensor Count = "));
      Serial.print(numsensor, DEC);
      Serial.print(F("\tMask = "));
      Serial.println((uint16_t)((MQData.encoded & Mask_mask)>>mask_shift) , BIN);
      Serial.print(F("Location = "));
      Serial.print(((MQData.encoded & Location_Mask)>>location_shift), DEC);
      Serial.print(F("\tStatus = "));
      Serial.println((uint8_t)((MQData.encoded & status_mask) >> status_shift), BIN);
      for (uint8_t i = 0; i < numsensor; i++) {
        Serial.print(i, DEC);
        Serial.print(F("\t:\t"));
        Serial.println(MQData.Data[i], 8);
      }
  }
}`

bvernham, 2024-01-06:
PS How do I know if the chips are legit?

I got the old Opti-tek scope out and all the chips look pretty much the same with slight variation in the lowest set of the #'s.
[image]
[image]
[image]
[image]

bvernham, 2024-01-06:
The  XIAO Sense 52840 is a stand alone micro?

I just the need a communciation module like the NRF24...

TMRh20, 2024-01-06:
Yup https://wiki.seeedstudio.com/XIAO_BLE/
Its a SOC with a radio n other stuff. Also other boards like the Feather 52840 Express if you need more pins.
I wrote https://github.com/TMRh20/nrf_to_nrf to allow communication with RF24 or between NRF52 devices using the RF24 API.

2bndy5, 2024-01-06:
The nRF52840 can also be used as a networking coprocessor if programmed correctly. Valve did exactly that with the nRF51822 in their Steam Controller which would pair with the bundled receiver dongle that used an nRF24LU1.

Ebyte modules have their quirks too. They don't allow changing the PA level.

The lowest set of numbers written on the chip is usually specific to the manufacturer. It indicates the manufactured batch number, and some form of date.

----
Is your code using interrupts? It might help to know how interrupts are used (if at all).

TMRh20, 2024-01-20:
@bvernham Hey any progress or still having issues with this? Thinking I can probably close this issue soon if we can't replicate it?

closed 2024-02-16 by TMRh20